Arduino-IRremote/Arduino-IRremote #103: __vector_7 Compatibility Issue

novasdream, 2015-01-24:
```
/hardware/arduino/avr/cores/arduino/Tone.cpp:536: multiple definition of `__vector_7'
Arduino-IRremote/IRremote.cpp.o:/home/novasdream/arduino-1.5.8/libraries/Arduino-IRremote/IRremote.cpp:339: first defined here
collect2: error: ld returned 1 exit status
```

That is error message..
A few libraries have some common pins in timers.

joshuajnoble, 2015-01-24:
This actually means that Tone is trying to use the same interrupt as IRremote. There's a lot of discussion around this on the Arduino forums [here](http://forum.arduino.cc/index.php/topic,106043.msg959970.html) for instance.

z3t0, 2015-03-07:
Take a look here, replacing the Tone Library may solve your problem.
https://code.google.com/p/arduino-new-tone/

This issue is with Arduino Software and not specifically this library

z3t0, 2015-03-07:
@novasdream This issue is being closed as it is not related to this library specifically but rather the Arduino Library Tone (maybe others too). Please follow [here](https://code.google.com/p/arduino-new-ping/wiki/HELP_Error_Vector_7_When_Compiling) instead.

It's not the same library but outlines the problem in more detail and what causes it

closed 2015-03-07 by z3t0