Arduino-IRremote/Arduino-IRremote #104: Library incompatibility with Atmega 8

varshu, 2015-01-24:
I was compiling the code on 'arduino NG or older w/ atmega8' and I got the following error. 
[image]

The code: 

#include <IRremote.h>
#include <IRremoteInt.h>

int RECV_PIN = 11;
IRrecv irrecv(RECV_PIN);
decode_results results;

void setup()
{
  Serial.begin(9600);
  irrecv.enableIRIn(); // Start the receiver
}

void loop()
{
  if (irrecv.decode(&results))
    {
     Serial.println(results.value, HEX);
     irrecv.resume(); // Receive the next value
    }
}

The code runs in Arduino UNO. How to fix this? How to make the library work for Atmega 8?

joshuajnoble, 2015-01-28:
This seems to indicate that the program is too large to fit in the program space afforded by the Atmega 8. I wonder if it would fit if you modify the library to remove all the defines and info for remotes other than the one that you're using.

z3t0, 2015-03-07:
@varshu Issue will be closed as this is a board dependent issue, and the board does not have enough memory space to load this library. 
**Possible Solution**: Remove defines in the library that are unneeded to try and make the library small enough to fit on the board.

closed 2015-03-07 by z3t0

christophe94700, 2016-03-09:
it can be used with a Atmega8. I deleted the protocols that I do not use the IRremote.h file.
example:
I use only:

#define DECODE_NEC 1
#define SEND_NEC 1

#define DECODE_SAMSUNG 1
#define SEND_SAMSUNG 1

z3t0, 2016-03-27:
awesome glad it worked