Arduino-IRremote/Arduino-IRremote #108: Recreate Wiki on Pages

z3t0, 2015-03-10:
I am working on totally re-doing the whole wiki because there are many things that have not been documented and also many new features have been added. Also we now have a GitHub [page](http://shirriff.github.io/Arduino-IRremote) so we might as well use it. 

If anyone can help in any way just post here. Mainly what I need is:
- Writers who can re-write parts of the wiki that need rewriting 
- HTML Programmers (Any Level, Beginners too) that can do simple things like make lists or create paragraphs
- Organizers who can work on deciding what needs to be added to the wiki.

Lauszus, 2015-03-10:
@z3t0 could you and me to the contributors on the webpage as well? Please see my commits for reference: https://github.com/shirriff/Arduino-IRremote/commits?author=Lauszus.

z3t0, 2015-03-10:
@Lauszus Sorry about that I thought I had added you already. Just added you to the webpage, I'll add you to the contributors.md now

Semphora, 2015-08-10:
the "Check here for tutorials and more information." still seems to link to the github page and is a dead end.

z3t0, 2015-08-10:
@Semphora great timing, I just committed the update 2 seconds before you commented lol

z3t0, 2015-08-10:
Update: I have not gotten much done in this (besides an api reference) since there are major rewrites being worked upon and many new additions, PRs etc.

Semphora, 2015-08-10:
:) that api reference seems to be exactly what I was looking for.

Semphora, 2015-08-10:
but wouldn't it make more sense for the "Check here for tutorials and more information." link to link to the wiki?

z3t0, 2015-08-11:
@Semphora 
>but wouldn't it make more sense for the "Check here for tutorials and more information." link to link to the wiki?

You are right.. but I feel reluctant doing so until the wiki itself is up to date. So the best solution would probably be to remove the link for now. Thoughts?

z3t0, 2015-08-11:
> :) that api reference seems to be exactly what I was looking for.

I'll put it up on the wiki tomorrow

z3t0, 2015-08-12:
@Semphora I just realized that I had deleted the api reference when I reinstalled my system (it wasn't committed to the repo since it was unpolished at the time). I will rewrite it soon but in the meantime you can take a look at the source code, its very well-commented and you can ask me if you have any questions. So sorry for the inconvenience

Semphora, 2015-08-12:
@z3t0 thanks. I'll do that.

Semphora, 2015-08-13:
@z3t0 and btw.: Thanks for doing this! I really appreciate you writing and supporting this library!

z3t0, 2015-08-13:
@Semphora No problem! Also @shirriff was the one to actually write the library and alot of other people have been contributing too, basically all I have been doing is trying to manage things like pull requests and issues.

z3t0, 2015-08-13:
Will definitely have the API up tomorrow :smile:

z3t0, 2015-08-14:
@Semphora What kind of things do you think I should put in the API? there isn't much point documenting the source code itself... I was thinking the IRrecv and IRsend classes since that would help other people add protocols and such.

Semphora, 2015-08-14:
@z3t0 seems like a plan :)

z3t0, 2015-08-15:
@Semphora I have started on the documentation, please check https://github.com/z3t0/Arduino-IRremote/wiki/api

I am not sure if its good/bad so any feedback is appreciated.

Semphora, 2015-08-15:
@z3t0 the link gets me a " create new page" :(

z3t0, 2015-08-15:
@Semphora Sorry about that I forgot that I renamed the page.
https://github.com/z3t0/Arduino-IRremote/wiki/API-Reference

closed 2020-10-19 by ArminJo