Arduino-IRremote/Arduino-IRremote #111: add decodeSHARP method

jichen3000, 2015-03-12:
Hi,

I just found that the IRremote.h and IRremote.ccp codes didn't have a method called decodeSHARP, which meant your code didn't support decode SHARP.

But actually, I found in the previouse codes( I found in http://ledstripes.googlecode.com/svn/!svn/bc/32/trunk/Software/SharpIR/IRremote.cpp) that it had decodeSHARP in there.

So I just copied the  decodeSHARP to IRremote.cpp, and added a defination in IRremote.h, then your codes support decoding SHARP remoter.

Could you change the your code, in case the other guys want to use it.

jichen3000, 2015-03-12:
I have changed codes in my forked project. https://github.com/jichen3000/Arduino-IRremote

z3t0, 2015-03-12:
Hi. Sorry one of the merges must have deleted it.

z3t0, 2015-03-12:
Sorry I wrote my previous comment in a state of confusion because I was quite certain that we did support sharp Decoding, I did not know what to say because I only recently became the maintainer of this project and did not know what features were available before.

However, my guess is that at some point in the past sharp decoding was removed for some reason. A few days a go I merged a PR for sharp decoding into #107 the experimental branch. Please test the experimental branch and let me know if it works for you. That is the sharp decoding in experimental. The reason I have not merged this into master as of yet is that I do not have any sharp equipment to test on, I am waiting to hear results of success/failure from others so that I know if its stable.

Please test #107 and let me know if Sharp Decoding works for you. I'll be closing this PR because sharp support is available just not in the master branch or the official release. I will release a beta soon for the features in Experimental, likely sometime today

closed 2015-03-12 by z3t0