Arduino-IRremote/Arduino-IRremote #112: IRrecvDemo won't complie

TheMabb, 2015-03-13:
I try to compile the demo and I get this:

Arduino: 1.6.1 (Windows 8.1), Board: "Arduino Uno"

C:\Program Files (x86)\Arduino\libraries\IRremote\IRremote.cpp: In member function 'long int IRrecv::decodeWhynter(decode_results*)':

C:\Program Files (x86)\Arduino\libraries\IRremote\IRremote.cpp:753:51: error: a function-definition is not allowed here before '{' token

 long IRrecv::decodeSharp(decode_results *results) {

                                                   ^

C:\Program Files (x86)\Arduino\libraries\IRremote\IRremote.cpp:1357:1: error: expected '}' at end of input

 }

 ^

C:\Program Files (x86)\Arduino\libraries\IRremote\IRremote.cpp:1357:1: error: expected '}' at end of input

Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

z3t0, 2015-03-13:
Thank you for the detailed report. I'll get on it :smile:

z3t0, 2015-03-13:
What version of the library are you using? 1.0 or 1.0.1 It seems as though you are using 1.0.1 which is a beta release. Please make sure you are using the latest [stable](https://github.com/shirriff/Arduino-IRremote/releases/tag/MAJOR) release.

Thank you for reporting that error. It really helps for further development.

TheMabb, 2015-03-13:
You are absolutely right I was using 1.0.1. My bad. Thanks!

z3t0, 2015-03-13:
No problem, Thank YOU for reporting the bug. We are now one step closer to adding new features to the main repo. :+1:

closed 2015-03-13 by z3t0