Arduino-IRremote/Arduino-IRremote #114: NEC code for LG sends bits reversed

Tarlin, 2015-03-16:
There are published IR codes for an LG tv. They use the NEC protocol. I have succeeding in getting the library to work, but the bits are being sent from least significant to most significant in each byte, rather than from most to lease.

For LG TVs, the id for the tv is the first byte and the complement to the id is the second byte. The ID for LG TVs should be 0x04, but is 0x20 for this library. Similarly, the code and the complement of the code are send next.

If this was changed, the documentation from LG tvs could easily be used to get this library controlling an LG tv.

z3t0, 2015-03-16:
So you're suggesting that we recreate the NEC protocol named in LG and reversing the order that the bits are sent?

Tarlin, 2015-03-16:
I am not sure.

First, I wanted to document it someplace and I was thinking of trying to address it in some manner. I have not contributed to a project before or used GIT, so not completely sure how to do this.

My thinking was that this seems simple to either add a function as you suggested or to add a helper function that flips the bits for LG codes. I think ideally, it would be good to research the NEC protocol and figure out if all of the codes have this same issue, and then flip the order of recorded and transmitted NEC codes. This would break backwards compatibility, so probably not a good idea, even if it turns out to be true.

After typing this, I think the best solution would be to add another sender function with a comment above it that explains this is what to use, if you are not recording your own codes and are using documented codes from an LG manual. This function would just call a helper function which would reverse the bit order on the code and then call the regular send.

z3t0, 2015-03-16:
That definitely sounds like a good plan. Good you please paste your implementation that you used to get this library working for LG on a gist? I'll try and get this implemented but I will need your help in terms of testing because I do not have any LG Equipment to test on.

AnalysIR, 2015-03-19:
Workaround: Alternatively, you could just re-order the value/bits before calling the function.

shirriff, 2015-03-19:
I'd rather have this helper function optional or ifdef'd out so it doesn't enlarge the code for people who don't need it.

z3t0, 2015-03-25:
@Tarlin Can you please post your sketch for this as a gist so I can try and implement it into the library? Thanks

z3t0, 2015-03-25:
@shirriff Agreed.

Tarlin, 2015-03-25:
I will post it tonight. I got sidetracked for a few weeks with another
project. Didn't think it would be a problem, sorry.

Tarlin, 2015-03-25:
Ok, I posted the [GIST.](https://gist.github.com/Tarlin/2643f5b7c8c17edf185f#file-lgremote) Sorry, I hadn't actually written it until today. I did not think this would be acted on as quickly as it was. The commands listed all work great with my tv.

One other point...
The description of the NEC protocol on the project page says, "NEC: 32 bits are transmitted, most-significant bit first. (protocol details)", but clicking through to the protocol details shows, "The picture above shows a typical pulse train of the NEC protocol. With this protocol the LSB is transmitted first."

z3t0, 2015-03-26:
@Tarlin No problem, thanks for the info! I'll see what I can do.
'''
The description of the NEC protocol on the project page says, "NEC: 32 bits are transmitted, most-significant bit first. (protocol details)", but clicking through to the protocol details shows, "The picture above shows a typical pulse train of the NEC protocol. With this protocol the LSB is transmitted first."
'''
I am starting to see other similar mistakes in the Wiki which is why I am working on #108 and revamping the documentation. Please report any similar mistakes if you find any!

Tarlin, 2015-03-26:
That isn't actually a mistake. The library does send msb first, but the protocol expects lsb first.

z3t0, 2015-03-26:
I see what you mean.

z3t0, 2015-03-27:
@Tarlin This may take a while since it is my first time adding a new protocol. If any one else is interested feel free to try it out. I'm definitely going to do it just that it may take a while.

Tarlin, 2015-03-27:
@zt30 What did you want to do? Do you want to fix the NEC protocol to actually be LSB or add another protocol that is the NEC protocol with the MSB/LSB thing changed? Do you want a more general purpose "ReverseBitsInEachByte" function?

If you let me know, I can probably write the change fairly quickly.

z3t0, 2015-03-27:
@Tarlin I was planning on adding a separate protocol instead of creating a helper function. The main reason for doing this is that I wanted to document the process of creating a protocol for the tutorials i am adding to the website. If you are willing then please go ahead and also note some of the steps and I can create the documentation from there. Thanks!

z3t0, 2016-07-11:
#331

closed 2016-07-11 by z3t0