Arduino-IRremote/Arduino-IRremote #117: #define statement for different devices selection

MustafaFayez, 2015-03-22:
You said to keep the size of the library manageable you're moving to a model where different device types use a #define statement, for instance:

#define SHARP

where is that ? and how you will do it ?

z3t0, 2015-03-22:
@MustafaFayez I understand your confusion as we have not made this very clear yet, I am working on a set of tutorials to clear this up and will release the next version at the same time. For now please use the latest Release, which does not yet need #define statements.

MustafaFayez, 2015-03-22:
@z3t0 I searched for this issue online, some people say it won't work:
http://stackoverflow.com/questions/14418936/arduino-overriding-define-in-libraries

z3t0, 2015-03-22:
@MustafaFayez That's what I thought too, but apparently it did work and provide results...
Check #99 and #110

z3t0, 2015-03-22:
@MustafaFayez We did end up using #defines so I'm getting confused as to how they actually worked...
You'll see what I meant if you read #110 because we actually have it working there.

MustafaFayez, 2015-03-23:
@z3t0 I read #110, great work there,but did you use #define in arduino sketch or just in .h file as in #110 ?? In other words, can the user selects the library code to be compiled from sketch or not?

z3t0, 2015-03-25:
@MustafaFayez I see what you mean, I will do some testing my self and see what I can come up with. Thank you very much for pointing this out

MustafaFayez, 2015-03-29:
@z3t0 any updates?

z3t0, 2015-03-29:
Sorry got side tracked. I'll get back on it. 

https://github.com/shirriff/Arduino-IRremote/issues/117#issuecomment-87444810

z3t0, 2015-04-02:
Still working on it... Sorry got a lot of things distracting me.

AllAboutEE, 2015-05-14:
According to #99 it only works in sublime, and it's just bad practice over all. I'll separate the functions into different files and submit a pull request, I know it was said in #100 that it breaks the Arduino convention of including only one library but I'm sure people will understand.

z3t0, 2015-05-14:
Yep, this is quite a big priblem and we are yet to find an ideal solution.

----- Reply message -----

&mdash;

AllAboutEE, 2015-05-14:
I think I found one: I'm going to use namespaces instead of defines, that way we can keep the one file include statement and do "using namespace NEC" instead of "#define SEND_NEC" which does not work.

closed 2020-08-17 by ArminJo