Arduino-IRremote/Arduino-IRremote #118: Timer 3 for sending fails to compile for ATmega32U4

kopfkopfkopfaffe, 2015-03-30:
I want to use pin 5 on an Arduino Leonardo (=pin9 on a Teensy 2.0, same microcontroller) for sending. I configure the **IRremoteInt.h** to use Timer 3 (pin 9 on Teensy 2, pin 5 on Leonardo) like this:

    // Teensy 2.0
    #elif defined(__AVR_ATmega32U4__)
    //#define IR_USE_TIMER1   // tx = pin 14
    #define IR_USE_TIMER3   // tx = pin 9
    //#define IR_USE_TIMER4_HS  // tx = pin 10

I get the following error when compiling for the Arduino Leonardo:

    In file included from /path/IRremote.cpp:22:
    /path/IRremoteInt.h:363:2: error: #error "Please add OC3A pin number here\n"
    /path/IRremote.cpp:1295: warning: this decimal constant is unsigned only in ISO C90
    /path/IRremote.cpp: In member function ‘void IRsend::enableIROut(int)’:
    /path/IRremote.cpp:318: error: ‘TIMER_PWM_PIN’ was not declared in this scope
    /path/IRremote.cpp: In member function ‘void IRsend::sendSharp(long unsigned int, int)’:
    /path/IRremote.cpp:1331: warning: unused variable ‘invertdata’

Obviously the error in line 362 of IRremoteInt.h is tripped:

    #if defined(CORE_OC3A_PIN)
    #define TIMER_PWM_PIN        CORE_OC3A_PIN  /* Teensy */
    #elif defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
    #define TIMER_PWM_PIN        5  /* Arduino Mega */
    #else
    #error "Please add OC3A pin number here\n"
    #endif

If I use the default Timer4HS or Timer1, everything works fine.

AnalysIR, 2015-03-30:
OC3A is PC6 or digital pin 5.

It looks like you may have to add in the relevant defines for the leonardo.

AnalysIR, 2015-03-30:
try this
#if defined(CORE_OC3A_PIN)
#define TIMER_PWM_PIN        CORE_OC3A_PIN  /* Teensy */
#elif defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
#define TIMER_PWM_PIN        5  /* Arduino Mega */
#else
#define TIMER_PWM_PIN        5  /* Arduino Leonardo */
#endif

kopfkopfkopfaffe, 2015-03-30:
I tried that before. Here's what happens:

    core.a(Tone.cpp.o): In function `tone(unsigned char, unsigned int, unsigned long)':
    /arduinopath/arduino-1.0.5/hardware/arduino/cores/arduino/Tone.cpp:241: multiple definition of `__vector_32'
    IRremote.cpp.o:/path/IRremote.cpp:78: first defined here

AnalysIR, 2015-03-30:
OK....thats a pity. I will have to leave it to others to assist further. :(

z3t0, 2015-04-02:
@kopfkopfkopfaffe There are many conflicts with the tone library, so a possible suggestion would be to temporarily delete the Tone library and back it up somewhere else. Please report your results so we can assist you further.

kopfkopfkopfaffe, 2015-04-02:
I would favour a solution that allows tone and IRremote to co-exist (when not used in the same sketch of course, because there are only limited hardware resources). So Arduino seems to compile every core component although they might not be used in that particular sketch. 

So let's analyze this a little more. The error is

    core.a(Tone.cpp.o): In function `tone(unsigned char, unsigned int, unsigned long)':
    /homes/user/opt/arduino-1.0.5/hardware/arduino/cores/arduino/Tone.cpp:241: multiple definition of `__vector_32'
    /path/IRremote.cpp:78: first defined here

Line 78 of IRremote.cpp reads

    77 #ifdef F_CPU
    78 #define SYSCLOCK F_CPU     // main Arduino clock
    79 #else
    80 #define SYSCLOCK 16000000  // main Arduino clock
    81 #endif

F_CPU is read a couple of times in tone() of tone.cpp, but as far as I can see is never defined there.

Are there any imlementations in IRremote that replace implementations in the core?

z3t0, 2016-07-11:
This seems to be a conflict between Tone and IRremote for Leonardo, did you solve this? Have you tried the NewTone Library?

Will close this issue as it was a closed case and a solution cannot be provided without replicating the case and seeing as it has been over a year since the last response, that is not very likely.

If anyone has a similar issue feel free to create a new issue

closed 2016-07-11 by z3t0