Arduino-IRremote/Arduino-IRremote #120: Cannot find decodeSharp or sendSharpRaw

levsa, 2015-04-06:
I can't find the implementations for decodeSharp or sendSharpRaw. Errors?

Also, ifdefs for sending to SHARP looks strange:

```c++
#ifdef SHARP
  void sendSharp(unsigned int address, unsigned int command);
  void sendSharpRaw(unsigned long data, int nbits);
#endif
#ifdef IRsendSHARP
  void sendSharp(unsigned long data, int nbits);
#endif
```

z3t0, 2015-04-06:
Are you sure you are looking in the right place? I see this:
```
#ifdef IRsendSHARP
void IRsend::sendSharp(unsigned long data, int nbits) {
  unsigned long invertdata = data ^ SHARP_TOGGLE_MASK;
  enableIROut(38);

  // Sending codes in bursts of 3 (normal, inverted, normal) makes transmission
  // much more reliable. That's the exact behaviour of CD-S6470 remote control.
  for (int n = 0; n < 3; n++) {
    for (int i = 1 << (nbits-1); i > 0; i>>=1) {
      if (data & i) {
        mark(SHARP_BIT_MARK);
        space(SHARP_ONE_SPACE);
      }
      else {
        mark(SHARP_BIT_MARK);
        space(SHARP_ZERO_SPACE);
      }
    }
    
    mark(SHARP_BIT_MARK);
    space(SHARP_ZERO_SPACE);
    delay(40);

    data = data ^ SHARP_TOGGLE_MASK;
  }
}
```

z3t0, 2015-04-06:
Hmm. Where are you getting that from? Which file and from the Release or the master branch?

levsa, 2015-04-06:
Master branch. There is no sendSharpRaw defined anywhere I can see:

```
evons-MacBook-Pro:Arduino_IRremote levsa$ ag "sendSharpRaw"
IRremote.h
169:  void sendSharpRaw(unsigned long data, int nbits);

IRremote.cpp
1373:  sendSharpRaw((address << 10) | (command << 2) | 2, 15);

keywords.txt
31:sendSharpRaw KEYWORD2
Levons-MacBook-Pro:Arduino_IRremote levsa$
```

z3t0, 2015-04-06:
I'll take a look at the previous commits, one of them must have changed it.

z3t0, 2015-04-06:
sendSharpRaw seems to have been removed here https://github.com/shirriff/Arduino-IRremote/commit/117059a17f3769adf535469f0ae41ada53dcee8c

z3t0, 2015-04-06:
decodeSharp is also nowhere to be found... I'll take a look at previous commits

closed 2016-07-11 by z3t0