Arduino-IRremote/Arduino-IRremote #125: IRSend pin should be configurable

levsa, 2015-04-06:
Light Blue Bean can use pins 0-3 for PWM. Does that mean an IR LED can be connected to any of these and used to send IR? In that case, IRSend class should have the output pin configurable, similar to IRRecv.

levsa, 2015-04-06:
Light Blue Bean has an Atmega328p on it, I cannot find the configuration for 328p in IRRemoteInt.h though.

z3t0, 2015-04-06:
I believe both ATmega328P and ATmega8 have the same pinout.
http://avrprogrammers.com/articles/atmega8-vs-atmega328

levsa, 2015-04-06:
Hmm, but probably my configuration will define __AVR_ATmega328P__ and that will not work if that is missing in IRRemoteInt.h. Correct?

I'm not very good at reading pinouts. Which pin should I connect my IR-LED? The pins I have access to are shown here (I guess I'd need one of D0-D3): 

https://punchthrough.com/bean/arduino-users-guide/

z3t0, 2015-04-06:
I see what you mean. I do not know a lot about the core code of this project so I cannot answer your question. I agree that the send pin should be configurable. I will take a look and see if I can figure it out. 

If you take a look [here](https://github.com/PunchThrough/PunchThrough-BEAN-Arduino-Firmware/blob/master/hardware/bean/variants/bean/pins_arduino.h) it looks like D3 should work for you.

levsa, 2015-04-06:
Ok, thanks! How can you see that D3 should work btw?

z3t0, 2015-04-06:
From the file I linked above it seems that D3 has PWM, also from the sendDemo example it seems that pin 3 is the default pin for sending, so this should work without any changes for your board?

levsa, 2015-04-06:
Sounds good enough. I was just unsure about pin 3 in the sendDemo being the same as pin 3 on my board. But I'll try it!

levsa, 2015-04-06:
Nope, nothing is happening on pin 3 when I use irsend (tried a regular LED for debugging). Also, the program hangs after the first call to send (sendSAMSUNG).

z3t0, 2015-04-07:
I'm out at the moment but I'd recommend using the release version since as you have pointed out,  the master branch is filled with bugs.

----- Reply message -----

https://github.com/shirriff/Arduino-IRremote/issues/125#issuecomment-90250048

levsa, 2015-04-07:
Good point :) Thanks for the quick response!

z3t0, 2015-04-07:
@levsa Let me know if that works

levsa, 2015-04-12:
No, didn't work. Same result, hangs after one try.

z3t0, 2015-04-12:
Very odd, I'll take a look and see what I can do

z3t0, 2015-04-18:
Take a look here where the pins are being changed, https://github.com/khair-eddine/Arduino-IRremote/pull/1/files

khair-eddine, 2015-04-18:
well if you want to change the output on pin9 in the original IRremote you just have to go to the IRremoteint.h in the library directory and uncomment (#define IR_USE_TIMER1   // tx = pin 9)
 and make comment for(// #define IR_USE_TIMER2     // tx = pin 3) 
but anyway try pin9 first.

levsa, 2015-04-20:
This looks very bad. I think it is very unconvenient to duplicate that much
code, it will be hard to maintain. Is there a way to give the pin number as
an argument to the constructor?

Then you don't need to duplicate any code, just create instances with
different input:

IRsend  irsend0(PIN_3);
IRsend irsend1(PIN_9);
IRsend irsend2(PIN_10); // or some pins

/Levon

z3t0, 2015-04-20:
@levsa that is what I am thinking about, also this way we could make it possible to have many more than 3/4 irsend's for microcontrollers that allow for more (I think the Mega)

Zhenit-606, 2015-04-20:
Not so, there are many PWM pins, but few TIMERs. In 328p only 2 TIMERs for PWM control, TIMER 1 (16 bit) and TIMER 2 (8bits), the TIMER 0 controls the time of shetch. Of course, this can be tweaked, but will always have only 3 TIMERs, I have difficulty in playing sound PWM and send IR pulse, I have to make a stop in the audio before sending the PWM.

khair-eddine, 2015-04-20:
@levsa @z3t0 i thinked about that too but i need some time to get used to C language and finish some work, if someone want to improve it, it will be great. or i'll have to close this issue and back to it when i finish my project in juin
@Zhenit-606 yes bro every timer timer is responsible for behavior of two PWM pins in Atmega328(arduino uno) i'm using timer1 and timer2 at the same time instead of using it every one alone i'll see too if i can increase pins by using timer0

levsa, 2015-04-20:
I can take a look at it some time soon.

Levon

z3t0, 2015-04-26:
> if someone want to improve it, it will be great. or i'll have to close this issue and back to it when i finish my project in juin

I will give it a try but may take a while. FINALLY got an arduino again (although it only partially works)

z3t0, 2016-07-11:
Bump. (just doing this so it is shown at the top of the issues and is more convenient)

closed 2020-07-03 by ArminJo