Arduino-IRremote/Arduino-IRremote #126: PANASONIC POWER OFF AND ON

Norfeldt, 2015-04-07:
I tried to make the following [Smart Remote Control](http://makezine.com/projects/smart-remote-control/) and realized that I have a problem with the way I turn on and off my PANASONIC TV. I have to long press the power button..

Just to be sure that this was the issue, I tried the simple (modified) example on your page. 

````Arduino
#include <IRremote.h>
 
#define PanasonicAddress      0x4004     // Panasonic address (Pre data) 
#define PanasonicPower        0x100BCBD  // Panasonic Power button

IRsend irsend;

void setup()
{  
}

void loop() {
    irsend.sendPanasonic(PanasonicAddress,PanasonicPower); // This should turn your TV on and off
   delayMicroseconds(50000);
}
````

It don't work... Any idea of how to get around this?

z3t0, 2015-04-07:
Just to clarify, do you have to long press with your original remote?

Norfeldt, 2015-04-07:
Yes, I have to hold it for about 3 sec

z3t0, 2015-04-07:
Could you not simply record what is being sent in those three seconds and then replay that on the Arduino?

Norfeldt, 2015-04-07:
The decoding tells me `PANASONIC: 4004 100BCBD` - which matches the example

z3t0, 2015-04-07:
And that is the initial button press right?

z3t0, 2015-04-07:
What happens if you try and record the 3 second power off phase using 
https://github.com/shirriff/Arduino-IRremote/blob/master/examples/IRrecvDemo/IRrecvDemo.ino

Norfeldt, 2015-04-07:
I'll check.

Norfeldt, 2015-04-07:
```Arduino
Unable to connect: retrying (1)... 
Unable to connect: retrying (2)... 
Unable to connect: retrying (3)... 
Unable to connect: retrying (4)... 
Unable to connect: is the sketch using the bridge?
````

I'm no super arduino user..

z3t0, 2015-04-07:
Hmm, that's odd I think.. First step would be to make sure you are using the Release version of this library, the master branch is undergoing major development.

https://github.com/shirriff/Arduino-IRremote/releases/tag/MAJOR

Norfeldt, 2015-04-07:
I changed `int RECV_PIN = 11;` to `int RECV_PIN = 13;` so it matches the layout

http://cdn.makezine.com/uploads/2014/06/smartremotecontrol_bb_resized.png

z3t0, 2015-04-07:
Pin 13 has 16 bits of resolution, so that may make a difference. http://41.media.tumblr.com/0b5c4be0cefa771d5e2b2d1102e19ce2/tumblr_muujlsYypU1s5t695o1_1280.png

Let me know if the same problem arises when using the Release version

Norfeldt, 2015-04-07:
I am using the latest release. I just downloaded it to try the example. The lib I used during the tutorial on make I named `Arduino_IRremote`. So pretty sure I used the right lib for the example

Norfeldt, 2015-04-07:
I realized that it is pin 11 and not 13 I'm using.. But it gives me the same error (after I saw this first I changed it to 13 because I thought this was the problem)

Norfeldt, 2015-04-07:
Just opening the serial monitor gives the connection issue..

z3t0, 2015-04-07:
Oh wait.. is it possible that the error might have something to do with the Yun itself?
```
Unable to connect: retrying (1)... 
Unable to connect: retrying (2)... 
Unable to connect: retrying (3)... 
Unable to connect: retrying (4)... 
Unable to connect: is the sketch using the bridge?
```

z3t0, 2015-04-07:
As far as I can see, the sketch I linked above should not repeat those lines in any case.

Norfeldt, 2015-04-07:
I believe this this is a YUN message..

z3t0, 2015-04-07:
Yup you're right.

```
Hi IBE007, happy you've got it running. As you've correctly spotted, most of the examples require the serial port (the /dev/tty or COMx entry in the Port menu)

You've got that "unable to connect message" because that sketch does not use the Console. Console is just like Serial but works over network/wifi connections, not serial ones.

If you wish to see the Console working, check out ConsoleAsciiTable example and upload it via wifi
```
http://forum.arduino.cc/index.php?topic=188587.msg1395830#msg1395830

Norfeldt, 2015-04-07:
The Make tutorial has a Console that decodes

```Arduino
// Smart Remote Control Sketch
// Copyright 2014 Tony DiCola
// Released under an MIT license: http://opensource.org/licenses/MIT

// Based on examples from the IR remote library at: https://github.com/shirriff/Arduino-IRremote
// Depends on a fork of the library to fix a file name conflict in recent Arduino IDE releases.
// Download forked library from: https://github.com/tdicola/Arduino_IRremote

#include <ctype.h>
#include <Console.h>
#include "IRremote_library.h"
#include "IRremoteInt_library.h"

// Pin connected to the IR receiver.
#define RECV_PIN 11

// Length of time to delay between codes.
#define REPEAT_DELAY_MS 40

// Size of parsing buffer.
#define BUFFER_SIZE 100

// Parsing state.
char buffer[BUFFER_SIZE+1] = {0};
int index = 0;
decode_results command;
unsigned int rawbuf[RAWBUF] = {0};

// Remote code types.
#define TYPE_COUNT 9
char* type_names[TYPE_COUNT] = { "NEC", "SONY", "RC5", "RC6", "DISH", "SHARP", "PANASONIC", "JVC", "RAW" };
int type_values[TYPE_COUNT] = { 1, 2, 3, 4, 5, 6, 7, 8, -1 };

// IR send and receive class.
IRsend irsend;
IRrecv irrecv(RECV_PIN);

// Setup function called once at bootup.
void setup() {
  // Initialize bridge, console, and IR receiver.
  Bridge.begin();
  Console.begin();
  irrecv.enableIRIn();
}

// Loop function called continuously.
void loop() {
  // Check if any data is available and parse it.
  if (Console.available() > 0) {
    // Read a character at a time.
    char c = Console.read();
    if (c == ':') {
      // Parse remote code type as current buffer value when colon is found.
      char* type = current_word();
      for (int i = 0; i < TYPE_COUNT; ++i) {
        if (strcmp(type, type_names[i]) == 0) {
          command.decode_type = type_values[i];
        }
      }
    }
    else if (c == ' ' && command.decode_type == PANASONIC && command.panasonicAddress == 0 && index > 0) {
      // Parse panasonic address as first value before whitespace.
      command.panasonicAddress = (unsigned int)strtoul(current_word(), NULL, 16);
    }
    else if (c == ' ' && command.decode_type != UNKNOWN && command.decode_type != PANASONIC && command.bits == 0 && index > 0) {
      // Parse number of bits as first value before whitespace.
      command.bits = (int)strtol(current_word(), NULL, 16);
[... 147 lines cut ...]
      // Scale length to microseconds.
      unsigned int value = results->rawbuf[i]*USECPERTICK;
      // Adjust length based on error in IR receiver measurement time.
      if (i % 2) {
        value -= MARK_EXCESS;
      }
      else {
        value += MARK_EXCESS;
      }
      // Print mark/space length.
      Console.print(value, HEX);
      Console.print(" ");
    }
    Console.println("");
  }
}

// Grab the currently parsed word and clear the buffer.
char* current_word() {
  buffer[index] = 0;
  index = 0;
  return buffer;
}

````

z3t0, 2015-04-07:
Try this sketch with Serial on instead of Console.
https://github.com/shirriff/Arduino-IRremote/blob/master/examples/IRrecvDemo/IRrecvDemo.ino

Then use your remote as if you were to the tv off and you will hopefully see the signals on the Serial Monitor, for this sketch to work we need to use Serial. I have never used a Yun so I may be missing something here..

Norfeldt, 2015-04-07:
I'm a bit confused.. I can't use the Serial at the moment (don't have a cable). The error will be the same as last. I then tried the following (had to try something)

````Arduino
#include <IRremote.h>
#include <Console.h>

int RECV_PIN = 11;

IRrecv irrecv(RECV_PIN);

decode_results results;

void setup()
{
  Console.begin();
  irrecv.enableIRIn(); // Start the receiver
}

void loop() {
  if (irrecv.decode(&results)) {
    Console.println(results.value, HEX);
    irrecv.resume(); // Receive the next value
  }
  delay(100);
}
```

This gave me the same error

```
Unable to connect: retrying (1)... 
Unable to connect: retrying (2)... 
Unable to connect: retrying (3)... 
Unable to connect: retrying (4)... 
Unable to connect: is the sketch using the bridge?
```

z3t0, 2015-04-07:
Hmm.. but how do you upload the sketch without a cable? I'm gonna go do some research on the Yun

Norfeldt, 2015-04-07:
It uploads via wifi

z3t0, 2015-04-07:
Ah, then do you know if we have to use a different method to Serial.println()?

z3t0, 2015-04-07:
OK got an idea.
```
/*
 * IRremote: IRrecvDemo - demonstrates receiving IR codes with IRrecv
 * An IR detector/demodulator must be connected to the input RECV_PIN.
 * Version 0.1 July, 2009
 * Copyright 2009 Ken Shirriff
 * http://arcfn.com
 */

#include <IRremote.h>

int RECV_PIN = 11;

IRrecv irrecv(RECV_PIN);

decode_results results;

void setup()
{
Console.begin();
  Serial.begin(9600);
while (!Console){
    ; // wait for Console port to connect.
  }
  Console.println("You're connected to the Console!!!!");
  irrecv.enableIRIn(); // Start the receiver
}

void loop() {
  if (irrecv.decode(&results)) {
    Console.println(results.value, HEX);
    irrecv.resume(); // Receive the next value
  }
  delay(100);
}

Norfeldt, 2015-04-07:
I fount the issue

z3t0, 2015-04-07:
NIce!

Norfeldt, 2015-04-07:
Now the serial gives me the following
```
100BCBD
100BCBD
100BCBD
100BCBD
100BCBD
```

z3t0, 2015-04-07:
Even if you hold the button, hmm. In that case maybe you just need to replay that a few times for the TV to register it?

Norfeldt, 2015-04-07:
Hmm.. Tried the following..

```Arduino
#include <IRremote.h>

#define PanasonicAddress      0x4004     // Panasonic address (Pre data) 
#define PanasonicPower        0x100BCBD  // Panasonic Power button

IRsend irsend;

void setup()
{  
}

void loop() {
   irsend.sendPanasonic(PanasonicAddress,PanasonicPower); // This should turn your TV on and off
   delay(100);
}
```
But I noticed that the little LED on the TV don't blink like it should.. Perhaps my IR LED is not the right one...

z3t0, 2015-04-07:
Has your setup been able to replicate any other button from the remote?

z3t0, 2015-04-07:
What pin is your IR Led for the above sketch?

Norfeldt, 2015-04-07:
I'm not sure what wave length my LED sends - it just worked on some Bose stuff and another digital TV box thing..

Norfeldt, 2015-04-07:
Well.. it's worth a try with another IR LED and see what happens.

Norfeldt, 2015-04-07:
Thank you very much for your help!!!!

z3t0, 2015-04-07:
No problem. Happy to help :smile:

z3t0, 2015-04-07:
Is your IR led connected to pin 3?

z3t0, 2015-04-07:
```c++
/*
 * IRremote: IRsendDemo - demonstrates sending IR codes with IRsend
 * An IR LED must be connected to Arduino PWM pin 3.
 * Version 0.1 July, 2009
 * Copyright 2009 Ken Shirriff
 * http://arcfn.com
 * JVC and Panasonic protocol added by Kristian Lauszus (Thanks to zenwheel and other people at the original blog post)
 */
#include <IRremote.h>
 
#define PanasonicAddress      0x4004     // Panasonic address (Pre data) 
#define PanasonicPower        0x100BCBD  // Panasonic Power button

#define JVCPower              0xC5E8

IRsend irsend;

void setup()
{
}

void loop() {
  irsend.sendPanasonic(PanasonicAddress,PanasonicPower); // This should turn your TV on and off
  
  irsend.sendJVC(JVCPower, 16,0); // hex value, 16 bits, no repeat
  delayMicroseconds(50); // see http://www.sbprojects.com/knowledge/ir/jvc.php for information
  irsend.sendJVC(JVCPower, 16,1); // hex value, 16 bits, repeat
  delayMicroseconds(50);
}
```

z3t0, 2015-04-07:
Try that ^^

z3t0, 2015-04-07:
Playing around with that should make it work

Norfeldt, 2015-04-07:
It is connected to pin 13.. I know that it is hooked up correctly because I can see it flash when I use my mobile camera.

z3t0, 2015-04-07:
When using the above sketch make sure you are using a IR Led (to send) and not an IR receiver. Also, it needs to be connected onto pin 3, not 13. Is your TV flashing or your IR Led?

Norfeldt, 2015-04-07:
Connecting it to pin 3 don't blink the IR LED.. But the pin 13 do blink IR LED (and the small red LED on the YUN board)

z3t0, 2015-04-07:
That's odd because the Sketch above says to use pin 3... 
```An IR LED must be connected to Arduino PWM pin 3.
```

Norfeldt, 2015-04-07:
I know..

...but I have to check that this isn't just a hardware issue. Then it is best to fix this right away

z3t0, 2015-04-07:
Hmm. I'll take a look at the pinouts

z3t0, 2015-04-07:
Are you sure that you are using my latest sketch posted above, because pin 13 should not be doing anything.... As far as I know. I will take a look into the code as you have suggested it may be a hardware issue.

z3t0, 2015-04-07:
Actually, you're right pin 13 should be the right pin.
```c++
#elif defined(__AVR_ATmega32U4__)
#define TIMER_PWM_PIN        13  /* Leonardo */
```

z3t0, 2015-04-07:
Try the sketch above, I just modified it.

Norfeldt, 2015-04-10:
I'm sorry I don't have the possibility to test it right now. Forgot the item - it will take some time for me to get it back.. meanwhile I will prepare the testing.

z3t0, 2015-04-10:
No problem, let me know when you test it. 

https://github.com/shirriff/Arduino-IRremote/issues/126#issuecomment-91523990

polarathene, 2016-03-04:
I have no idea if this helps. I've been learning how to use IR with a GlobalCache device and node.js. While researching came across this github issue (Also learning how to control a Panasonic tv but not with this project :) ).

While I was using GlobalCaches IR learning ability, I've also come across the pronto format IR codes listed [here](ftp://ftp.panasonic.com/pub/Panasonic/Drivers/Monitors/Discrete-remote-control-codesProntoCCFformat.pdf) for panasonic.

I've just written a parser that converts the burst sequence of on/off pulses minus preamble/leadin/leadout bursts from those pronto format codes to hex. Power toggle is `40040100BCBD`. Which checks out with your 16 bit and 32 bit params that you listed. I just noticed you're passing in numbers for int/long instead of strings like I am using with GlobalCache, was going to point out you might have been missing a `0` at the start :P

I've noticed that when I turn the tv off, it won't respond to the first power on command, no matter how many repeats for some reason. Soon as I send the full command a 2nd time the tv would turn on, this issue doesn't consistently happen though :\

---

```
        // Data
	for (unsigned long  mask = 1UL << (32 - 1);  mask;  mask >>= 1) {
        mark(PANASONIC_BIT_MARK);
        if (data & mask)  space(PANASONIC_ONE_SPACE) ;
        else              space(PANASONIC_ZERO_SPACE) ;
    }
```

I'm not familiar with C++, that's the logic taking your `PanasonicPower` value as 2nd param(data). `long` is 64-bits right? And you're doing a bitshift to remove the first 32 bits? That first 4 bits of 0000 is getting sent out right? `0000 0001 0000 0000 1011 1100 1011 1101` instead of `0001 0000 0000 1011 1100 1011 1101`

If that's all good maybe it's these hardcoded values in `ir_Panasonic.cpp`:

```
#define PANASONIC_BITS          48
#define PANASONIC_HDR_MARK    3502
#define PANASONIC_HDR_SPACE   1750
#define PANASONIC_BIT_MARK     502
#define PANASONIC_ONE_SPACE   1244
#define PANASONIC_ZERO_SPACE   400

...

// Set IR carrier frequency
enableIROut(35);
```

Preamble(pronto format): `0000 0071 0000 0032 0080 0040`. 
 - `0x71`(Decimal 113) is for frequency calculated as `1000000÷(113×0.241246)` giving us about `36683` for the frequency, that's already a bit higher than the 35 passed in. 
 - From what I understand, we can take `1 / frequency`(0.000027261) to get time of one cycle, then multiply that by the burst pair(mark-space) units for timing in microseconds to apply on/off signals which give us the binary 0/1 communication.
 - `0x32`(Decimal 50) is how long the burst pair sequence is in bits. This checks out as after you remove the lead in/out burst pairs(header/footer) you get 48.
 - `0x80`/`0x40`(Decimals 128/64) are the lead in burst pair(header). These are being referred to as `HDR_MARK/SPACE` above.
 - `0.000027261 * 128` = ``, 3489μs vs 3502
 - `0.000027261 * 64` = ``, 1744μs vs 1750

Burst pairs sequence after the preamble(full sequence can be found [here](ftp://ftp.panasonic.com/pub/Panasonic/Drivers/Monitors/Discrete-remote-control-codesProntoCCFformat.pdf)):
 - `0010 0031`, `0x10`(Decimal 16) and `0x31`(Decimal 49).
 - `0.000027261 * 16` = `0.000436173`, 436μs vs 400
 - `0.000027261 * 49` = `0.001335779`, 1336μs vs 1244

Lead out burst pair is `0010 0A9E` where `0xA9E` is 2718 in Decimal. This is just to keep a long off signal to indicate the sequence is complete, I think the way it's being handled here is effectively the same.

Sooo what to make of all this? I'm not sure where these Panasonic values are sourced from, but I'd assume the document I linked to is a good source, so updating values to work with those might help :) 
`enableIROut(35);` This function only accepts an `Int`, GlobalCache was reporting a frequency of 37,091 over the 36,683 from pronto codes, both work. **So maybe using a frequency of 37 instead of 35 will solve this?** 

The mark/space values all seem within tolerance(I have no idea how much of a difference that'll have, I've read that you have about 10% tolerance for IR to work reliably.), I'm not aware of the need for a `mark bit` just the two values for producing ones and zeroes, perhaps set it to the same as `PANASONIC_ONE_SPACE`?

---

I realize this is almost a year ago and the OP might have moved on since, might be helpful info to someone else and the issue should probably be closed :)

Szellem, 2017-02-19:
I know this is an old thread but I've got the same problem as Norfeldt had.
`
void loop() {
  irsend.sendPanasonic(PanasonicAddress,PanasonicPower); // This should turn your TV on and off
  delayMicroseconds(500);
}
`
This code doesn't even make the LED blink on my TV. Other devices work fine.

Here's the output of IRrecvDumpV2 using the original remote:

Encoding  : PANASONIC
Code      : 4004:100BCBD (48 bits)
Timing[99]: 
     +3500, -1650     + 500, - 400     + 450, -1250     + 500, - 350
     + 500, - 400     + 450, - 400     + 450, - 400     + 500, - 400
     + 450, - 400     + 450, - 400     + 500, - 350     + 500, - 400
     + 450, - 400     + 450, - 400     + 500, -1200     + 500, - 400
     + 450, - 400     + 450, - 400     + 500, - 400     + 450, - 400
     + 450, - 400     + 500, - 350     + 500, - 400     + 450, - 400
     + 450, -1250     + 500, - 400     + 450, - 400     + 450, - 400
     + 500, - 350     + 500, - 400     + 450, - 400     + 450, - 400
     + 500, - 350     + 500, -1250     + 450, - 400     + 450, -1250
     + 500, -1250     + 450, -1300     + 450, -1250     + 450, - 400
     + 500, - 350     + 500, -1250     + 450, - 400     + 500, -1250
     + 450, -1250     + 500, -1200     + 500, -1250     + 500, - 350
     + 500, -1250     + 500
unsigned int  rawData[99] = {3500,1650, 500,400, 450,1250, 500,350, 500,400, 450,400, 450,400, 500,400, 450,400, 450,400, 500,350, 500,400, 450,400, 450,400, 500,1200, 500,400, 450,400, 450,400, 500,400, 450,400, 450,400, 500,350, 500,400, 450,400, 450,1250, 500,400, 450,400, 450,400, 500,350, 500,400, 450,400, 450,400, 500,350, 500,1250, 450,400, 450,1250, 500,1250, 450,1300, 450,1250, 450,400, 500,350, 500,1250, 450,400, 500,1250, 450,1250, 500,1200, 500,1250, 500,350, 500,1250, 500};  // PANASONIC 4004:100BCBD
unsigned int  addr = 0x4004;
unsigned int  data = 0x100BCBD;

Encoding  : UNKNOWN
Code      : F61E2A57 (32 bits)
Timing[99]: 
     +3450, -1700     + 450, - 400     + 500, -1250     + 450, - 400
     + 450, - 400     + 500, - 400     + 450, - 400     + 450, - 400
     + 450, - 400     + 500, - 400     + 450, - 400     + 450, - 400
     + 450, - 450     + 400, - 450     + 450, -1250     + 450, - 450
     + 400, - 450     + 450, - 400     + 450, - 400     + 450, - 450
     + 400, - 450     + 450, - 400     + 450, - 400     + 450, - 450
     + 400, -1300     + 450, - 400     + 450, - 450     + 400, - 450
     + 400, - 450     + 450, - 400     + 450, - 450     + 400, - 450
     + 400, - 450     + 450, -1300     + 400, - 450     + 400, -1350
     + 400, -1300     + 400, -1300     + 450, -1300     + 400, - 450
     + 450, - 400     + 450, -1300     + 400, - 450     + 450, -1300
     + 400, -1300     + 450, -1300     + 400, -1300     + 450, - 450
     + 400, -1300     + 400
unsigned int  rawData[99] = {3450,1700, 450,400, 500,1250, 450,400, 450,400, 500,400, 450,400, 450,400, 450,400, 500,400, 450,400, 450,400, 450,450, 400,450, 450,1250, 450,450, 400,450, 450,400, 450,400, 450,450, 400,450, 450,400, 450,400, 450,450, 400,1300, 450,400, 450,450, 400,450, 400,450, 450,400, 450,450, 400,450, 400,450, 450,1300, 400,450, 400,1350, 400,1300, 400,1300, 450,1300, 400,450, 450,400, 450,1300, 400,450, 450,1300, 400,1300, 450,1300, 400,1300, 450,450, 400,1300, 400};  // UNKNOWN F61E2A57

After sending 100BCBD it sends F61E2A57 constantly in a loop while the button is pressed.
The funny thing is that I have a Logitech remote and it sends only 100BCBD but it works and switches the TV on.

Any idea how to make this work?

AnalysIR, 2017-02-19:
try setting your delayMicroseconds to 5000 milliSecs instead of 500 uSecs
i.e. delay(5000);

it is likely you IR receiver is shutting down as you have no real gap betwwe each iteration of the loop

Szellem, 2017-02-19:
delay(100); did the trick, it works like a charm, thank you very much!

Egyrt, 2017-04-10:
It does not work.. Where did you add this delay? I can't power on my tv this way

AnalysIR, 2017-04-10:
> It does not work..

This could mean a lot of different things.

Providing more detailed information, would improve your chances of getting help.

Also, You should really open a new issue and refer back to this one by typing #126
When you open a new issue, you will be prompted to provide the additonal information which is required.

mrtheduke, 2017-05-31:
Adding my function for Panasonic in case it helps anyone.  It takes a hex input from serial, and sends it via irsend.sendPanasonic.  If it receives the Panasonic code for power( 0x100BCBD) it handles the need for long press with a short loop.  If it's not that code it sends it just once.  It's working perfectly for me:
`void sendPanasonic() {
    if (newData == true) {
        int repeat = 1;
        String check(receivedChars);
        if(check == "0x100BCBD"){
          repeat = 5;
        }
        Serial.print("Recived Panasonic Code: ");
        Serial.println(receivedChars);
        unsigned long int code = strtoul(receivedChars,0,16); 
        #define PanasonicAddress      0x4004     // Panasonic address (Pre data) 
        int count = 0;
        while(count < repeat){
        IRsend irsend;
        irsend.sendPanasonic(PanasonicAddress,code);
        newData = false;
        delay(100);
        count++;
        }
    }`

ahmadfaridaziz, 2017-08-19:
please anybody help me ?? i can't decode remote AC panaspnic. maybe there is can decode remote AC panasonic

AnalysIR, 2017-08-19:
Some comments: @ahmadfaridaziz 

- Might be a good idea to open a new issue....
- If you want assistance might be a good idea to provide more info, such as what steps you have taken so far, what you are using, example code you have already tried etc
- List the steps you have taken already to solve your issue. Have you already searched thru other issues to see what advice was offered there & did you follow any of the previous advice offered.

@z3t0 ...this issue should probably be closed as it seems to be resolved & I don't have the appetite to read thru it again :)

[1 more reply]

closed 2017-08-20 by z3t0