Arduino-IRremote/Arduino-IRremote #128: multiple IR transmitter support IRremote (up to 3 pins)

khair-eddine, 2015-04-15:
hello, 
i modified the lib to support up to three IR transmitter at once. 
i want to conttribute in librairies because it's useful to others to use multiple transmitters.
i'm asking how can i contribute ? or should i get any permission?

z3t0, 2015-04-15:
In what way did you make changes? To the library code or in how you wrote your sketch?

&mdash;

khair-eddine, 2015-04-15:
hii bro, 
i didn't modified any instruction, i added some instructions in the IRremoteint to enable and disable 2 other pwm pins (9 and 10) than i duplicated the irsend with irsend1 and irsend2 and all their functions  so when i use new objects in arduino i declare them with those types(irsend,irsend1 and irsend2) 
i tryed this code and it worked for me. 
#include <IRremote.h>

IRsend  irsend0;
IRsend1 irsend1;
IRsend2 irsend2;

void setup()
{
  Serial.begin(9600);
}

void loop() {
  if (Serial.read() != -1) {
    for (int i = 0; i < 3; i++) {
      irsend0.sendSony(0xa90, 12); // pin 3
      irsend1.sendSony(0xa90, 12); // pin 10
      irsend2.sendSony(0xa90, 12); // pin 9
    }
  }
}

z3t0, 2015-04-15:
@khair-eddine I recommend forking this repository (top right) and then committing your changes and creating a pull request so we can review your changes.

Check: https://help.github.com/articles/fork-a-repo/

khair-eddine, 2015-04-16:
hii @z3t0 i forked a repository and i commit 3 changes to three files (IRremote.h IRremote.cpp and IRremoteint.h) see the modification than test the lib if u want.

khair-eddine, 2015-04-16:
https://github.com/khair-eddine/Arduino-IRremote/pull/1/files here's the link bro

z3t0, 2015-04-16:
@khair-eddine, thanks bro I'll take a look and see what I can do

brahbassim, 2015-06-15:
Hi, @khair-eddine
It's been a long time that i'm looking how to customise this library, how to add more IR emetter, and how to add new brand protocol, for exemple Hisense. If have any issue i will be the happiest in the world. thank's

z3t0, 2015-06-15:
Hi, So is your question to add another protocol? If so I'll take a look and
make a guide soon.

brahbassim, 2015-06-15:
Hi @z3t0 , yes my first question is how to add new protocol. Thank's

ipekbocegi, 2015-06-16:
@khair-eddine Thank you for modifications for 3 transmitter at the same time.

I  replaced your .cpp, .h versions with the original library. I tried your code but It gives the following error:
S_demo.cpp.o: In function `loop':
C:\Program Files (x86)\Arduino/S_demo.ino:26: undefined reference to `IRsend2::sendJVC(unsigned long, int, int)'
C:\Program Files (x86)\Arduino/S_demo.ino:28: undefined reference to `IRsend1::sendJVC(unsigned long, int, int)'
collect2.exe: error: ld returned 1 exit status

This is the code:
#include <IRremote.h>
long command_Left = 44L;
long command_Right = 66L;

IRsend1 irsend1; // digital pin 9 (PWM)
IRsend2 irsend2; // digital pin 10 (PWM)

void setup()
{
    Serial.begin(9600);
}

void loop() {
    irsend2.sendJVC(command_Left, 16, 0);
    delay(50);
    irsend1.sendJVC(command_Right, 16, 0);
    delay(50);
}

Any suggestions? 
Best wishes

khair-eddine, 2015-06-20:
hey @ipekbocegi  sorry for late reply bro, 
can you tell me what version of ide you are  using ?

z3t0, 2015-06-20:
This probably has to do with the #define usage. If you want to use JVC then remove the lines like https://github.com/khair-eddine/Arduino-IRremote/blob/khair-eddine-patch-1/IRremote.cpp#L244
or if you want just do #define SEND_JVC at the top of the file IRremote.cpp and then try it

z3t0, 2015-06-20:
Ohh wait. I got it. You replaced his files with the originals... thats what broke it. The master does not yet have multi receiver support

z3t0, 2015-06-20:
@brahbassim Please create a new issue just to make things easier to maintain

z3t0, 2015-06-20:
@khair-eddine I am on the #arduino IRC if you wanna chat about this

khair-eddine, 2015-06-20:
hii bro @z3t0  okai :)

alexbloggt, 2015-08-01:
Hi,
i just tried to run multiple IR transmitter using the library of @khair-eddine . However when I try to compile using code like that one of @khair-eddine provided here (https://github.com/z3t0/Arduino-IRremote/issues/128#issuecomment-93577771) I get errors like ```'IRsend1' does not name a type```.
What could be the issue?
I am using Arduino IDE 1.6.5 with Arduino Leonardo (for using normal IRremote my LED has to be connected to pin d13; are the other Pins different too?).
Thanks a lot!

khair-eddine, 2015-08-01:
hii @alexbloggt try using ide 1.0.5

alexbloggt, 2015-08-02:
Hey, error above was my fault - used the wrong library, sorry!
However now I am getting the same error as @ipekbocegi:
```undefined reference to `IRsend1::sendNEC(unsigned long, int)'```
Also tried IDE 1.0.5 but it is the same error here.
Used this (https://github.com/khair-eddine/Arduino-IRremote/tree/khair-eddine-patch-1) version of the library.
Thanks for helping!

z3t0, 2015-08-04:
@alexbloggt Please provide your full sketch, create a new issue and refer to this issue in the new issue.

Jeins98, 2016-08-03:
Hey, does this multiple mod still work? Because I don't get that work for me

z3t0, 2016-08-04:
@Jeins98 feel free to provide some more information in a new issue so that we can help you

stasiukaitis, 2016-11-10:
Hi. I trying to use Arduino UNO pors 5, 6, 7 to infrared sender, but no success. As I uderstand - I need to modify IRremoteInt.h 263-267 and 311-313 lines to use different pins?
I tried to modify library like this: https://github.com/khair-eddine/Arduino-IRremote/pull/1/files and it wok's. 
But I need at least 3 infrared outputs (better - 4, 5)and I use Ethernet shield... Pin 10 is used by ethernet shield... 
Please help me to use different ports like 5, 6 , 7. Thank's

AnalysIR, 2016-11-10:
@sttomas 

Why do you need 3 outputs?....maybe there is an alternative solution.

Did you consider using a external circuitry to achieve this?

sttomas, 2016-11-10:
I have three different devices, which need to control by infrared simultaneously. Commands are different, so have use three (or more) outputs.
Library modification, that I tried - works well. But the pin 10 is used by ETHERNET SHIELD. Therefore, I need to use a different PIN.
Ideally I should be five infrared outputs. for example on pin's: 3, 4, 5, 6, 7 or same...

Can you help me pls?

AnalysIR, 2016-11-10:
No, you can just send the different signals out on the same pin to all devices.
Individual devices will only react to their own signals.

The only exception would be if you were trying to control identical devices - which doesnt seem to be your use case.

So you should be OK using just one output pin.

sttomas, 2016-11-11:
I have minimum three infrared-controlled RGB spotlight, needing to turn different colors at the same time. Therefore, I need several parallel operating infrared diodes. The more the better...

I CAN SEND three different signals in parallel. Please see the attached library file. 
[IRremote-multiSend.zip](https://github.com/z3t0/Arduino-IRremote/files/585328/IRremote-multiSend.zip)
The library works perfectly. The only problem is that it use pin 10, which is required for the Ethernet board.

I do not asking about how to solve the problem with a single IR output.
My question is about the Arduino pin mapping. There are some useful info: https://www.arduino.cc/en/Tutorial/SecretsOfArduinoPWM ... 
My specific question is: How do I change the pin 10 to another pin in this library, that I attached? Such as 8, 9 and or smf... 

If you can answer to this question - would you answer? If you can't - maybe someone else knows how to do pin change properly.

Thank's

AnalysIR, 2016-11-11:
> If you can answer to this question - would you answer? If you can't - maybe someone else knows how to do pin change properly.

- Yes I can answer, but unfortunately, you have not described your problem, in sufficient detail to get a clear response.
- Are the 3+ lights in the same location or in different rooms/areas? It is important to know this. Please elaborate in more detail about your setup.
- the ATMega238 has a finite number of PWM pins that are tied to timers. Timer0 is reserved for the Arduino internal clock, leaving 2 other timers available. 
- If you are tackling tasks like this you need to read/study the data sheet for the MCU.
- The Ethernet board/library which you have provided no details on, will also likely use up at least one timer, thus leaving you with just one timer available and most likely just 1 IR output. (using IRremote)

Thus your only likely solutions are to use use 
**a)** external circuitry, 
**b)** upgrade to an Arduino Mega or Teensy etc. 
**c)** Use a softPWM and/or uPWM approach (details available by searching our [website](https://www.AnalysIR.com/blog/)) You would also need to customise the code yourself. 
**d)** Use multiple Arduinos Nanos to solve the problem (available for $2 each incl shipping)

However, based on the information you have provided so far, you cannot reliably send 3+ different signals, **at the same time**, to 3 spotlights that are located in the same area. (Maybe they are in different rooms or shielded from each other?) The different signals will typically interfere with each other and fail. If there was more inforamtion available about the physical setup/layout it may be possible to use other approaches/solutions to achieve the desired results.

PS: Most of the likely solutions will require extra work which is outside the scope of this library.

Mattadlard, 2016-12-30:
OK appreciate Newbie Question, never used an Audiono

Mattadlard, 2016-12-30:
Ok, Newbie Query on Arduino and Led Transmitters (have asked on Hacker groups, someone suggested here for advice/potential help.)

Looking to set up a Arduino as a Wireless wifi Receiver with cabled Led Infra-red transmitters, however it needs to be multiple transmitter options for multiple rooms from one unit. (unless its possible to slave a wifi enabled minis/small arduino to be a room transmitter.?

The aim is to control Infra-red controlled candles (Potentially same supplier but suspect candles operate on slightly variant signal wavelengths to allow product control and separation), replacing the stock infra-red controllers with an Arduino and possible app/web interface, however one needs to control several batches of candles and this will mean the Arduino has to send multiple signals to different and along different cable lengths.

What one wants to know if anyone has an idea is is this possible with the Arduino and multiple Infra-red Controllers, 4+ and has anyone tried or know someone who has tried this, and are their any issues with cable length or signal clashing with multiple transmitters.﻿ The candles not all coming on at the exact same time, a second/seconds time delay is fine and the Arduino needs to transmit to several rooms

pasathos, 2017-05-02:
Dear Sirss,
I've tried to compile the code to send 3 IR signals but I keep geting errors:

C:\Users\Paulo\Documents\Arduino\IRSendMulti\IRSendMulti.ino: In function 'void loop()':
IRSendMulti:22: error: 'class IRsend' has no member named 'sendSony'
       irsend0.sendSony(0xa90, 12); // pin 3
               ^
exit status 1
'class IRsend' has no member named 'sendSony'

I don't know why because the class has te procedure!!!
Please Help
thank you,
Paulo

z3t0, 2017-05-02:
Hmm not too sure. I'd recommend delete the library entirely and just download the fork in the original post

pasathos, 2017-05-02:
you mean download Arduino-IRremote-khair-eddine-patch-1.zip? I deleted the original lib and tried just this one and it didn't work. There are things missing in the code. Do I copy the code from these libs in the patch and paste them  in the original ones? Thank you

z3t0, 2017-05-02:
@pasathos try https://github.com/khair-eddine/Arduino-IRremote/archive/master.zip

pasathos, 2017-05-02:
class IRsend1 and class IRsend2 are absent from the irremote.h file so this happens:
IRSendMulti:11: error: 'IRsend1' does not name a type

 IRsend1 irsend1;
 ^
IRSendMulti:12: error: 'IRsend2' does not name a type

 IRsend2 irsend2;
 ^
C:\Users\Paulo\Documents\Arduino\IRSendMulti\IRSendMulti.ino: In function 'void loop()':

IRSendMulti:22: error: 'class IRsend' has no member named 'sendSony'

       irsend0.sendSony(0xa90, 12); // pin 3
               ^
IRSendMulti:23: error: 'irsend1' was not declared in this scope

       irsend1.sendSony(0xa90, 12); // pin 9
       ^
IRSendMulti:24: error: 'irsend2' was not declared in this scope

       irsend2.sendSony(0xa90, 12); // pin 10
       ^
exit status 1
'IRsend1' does not name a type

z3t0, 2017-05-02:
can you upload a zip of the library and the sketch that is failing.

pasathos, 2017-05-02:
-the lib I used is the one from the link 
@pasathos try https://github.com/khair-eddine/Arduino-IRremote/archive/master.zip

the file is just:
/*
 * IRremote: IRsendDemo - demonstrates sending IR codes with IRsend
 * Created by Khaireddine Hlali to support multiple transmitters 
 */
#include <IRremote.h>
#define SEND_SONY

IRsend  irsend0;
IRsend1 irsend1;
IRsend2 irsend2;

void setup(){
  Serial.begin(9600);
}
void loop() {
  if (Serial.read() != -1) {
    for (int i = 0; i < 3; i++) {
      irsend0.sendSony(0xa90, 12); // pin 3
      irsend1.sendSony(0xa90, 12); // pin 9
      irsend2.sendSony(0xa90, 12); // pin 10
    }
  }
}

z3t0, 2017-05-02:
@pasathos My apologies. I sent you the wrong link.

Please delete all copies of the library so far and instead try [this](https://github.com/khair-eddine/Arduino-IRremote/archive/b92302b5cc7a2b910a5e699b6f03ddb095d4dba1.zip)

pasathos, 2017-05-02:
Now the lib has the classes IRsend but this appears. With or without #define SEND_SONY there are errors
C:\Users\Paulo\Documents\Arduino\IRSendMulti\IRSendMulti.ino: In function 'void loop()':

IRSendMulti:19: error: 'class IRsend' has no member named 'sendSony'
       irsend0.sendSony(0xa90, 12); // pin 3
               ^
exit status 1
'class IRsend' has no member named 'sendSony'

the code is:
#include <IRremote.h>
IRsend  irsend0;
IRsend1 irsend1;
IRsend2 irsend2;

void setup(){
  Serial.begin(9600);
}
void loop() {
  if (Serial.read() != -1) {
    for (int i = 0; i < 3; i++) {
      irsend0.sendSony(0xa90, 12); // pin 3
      irsend1.sendSony(0x5a5, 12); // pin 9
      irsend2.sendSony(0xa5a, 12); // pin 10
    }
  }
}

pasathos, 2017-05-02:
the library is the one you last sent.
[Arduino-IRremote-b92302b5cc7a2b910a5e699b6f03ddb095d4dba1.zip](https://github.com/z3t0/Arduino-IRremote/files/972144/Arduino-IRremote-b92302b5cc7a2b910a5e699b6f03ddb095d4dba1.zip)

z3t0, 2017-05-02:
Add 

`#define SEND_SONY`
as line 26 of IRremote.cpp and see if that works,

pasathos, 2017-05-03:
If I put the #define after the #include irremote.h this appears
C:\Users\Paulo\Documents\Arduino\IRSendMulti\IRSendMulti.ino: In function 'void loop()':
IRSendMulti:19: error: 'class IRsend' has no member named 'sendSony'
       irsend0.sendSony(0xa90, 12); // pin 3
               ^
exit status 1
'class IRsend' has no member named 'sendSony',

if I put before the include this appears:
C:\Users\Paulo\AppData\Local\Temp\ccMfjj8a.ltrans0.ltrans.o: In function `loop':

C:\Users\Paulo\Documents\Arduino\IRSendMulti/IRSendMulti.ino:21: undefined reference to `IRsend::sendSony(unsigned long, int)'
C:\Users\Paulo\Documents\Arduino\IRSendMulti/IRSendMulti.ino:22: undefined reference to `IRsend1::sendSony(unsigned long, int)'
C:\Users\Paulo\Documents\Arduino\IRSendMulti/IRSendMulti.ino:23: undefined reference to `IRsend2::sendSony(unsigned long, int)'
collect2.exe: error: ld returned 1 exit status
exit status 1

z3t0, 2017-05-03:
You have to put the define inside irremote cpp. Not your sketch

pasathos, 2017-05-04:
Now it gives this error
C:\Users\Paulo\AppData\Local\Temp\cc5AFS6L.ltrans0.ltrans.o: In function `loop':

C:\Users\Paulo\Documents\Arduino\IRSendMulti/IRSendMulti.ino:19: undefined reference to `IRsend1::sendSony(unsigned long, int)'

C:\Users\Paulo\Documents\Arduino\IRSendMulti/IRSendMulti.ino:20: undefined reference to `IRsend2::sendSony(unsigned long, int)'
collect2.exe: error: ld returned 1 exit status

exit status 1

pasathos, 2017-05-04:
Ok, finally I was able to compile it! I had to put
#define SEND_SONY
#define SEND_AIWA_RC_T501
in both files irremote.cpp and irremote.h
I'll see if I can test it later.
THANK YOU SO MUCH for your help.
Paulo

z3t0, 2017-05-05:
Awesome! Happy to help. Sorry it took so long haha. I had personally never dabbled with this experimental feature so was grasping in the dark for a while.

sttomas, 2017-05-05:
I decided some work around of this problem... 
I tested it with "standart" one IR LED library. All LED's are connected to D3 and the other pin - to digital outputs of Adruino. Then we manipulate digital outputs. You need to turn on / off these outputs before sending the IR signal to LED... You can use many IR LED's (limits only Arduino I/O quantity)
Schematics: [image]

pasathos, 2017-05-06:
This a great, simple idea! Yes, multiplexing in time we can achieve the same results. My question is, can we use 2 servos and a PWM for a standard motor while using the IRremote lib?
Thank you mate!

z3t0, 2017-05-07:
Depends on how the servos are controlled but generally no. There simply isn't enough independent timers on an Uno

pasathos, 2017-05-07:
hi,
does anybody know how long does it take to transmit a 12-bit code using an IR coding?
Thank you very much,
Paulo

bengtmartensson, 2017-05-07:
Depends on the protocol. For example RC5 24ms, not counting repetitions.

Waywardnl, 2018-10-27:
Does anyone have the answer? I also posted the question here:
http://forum.arduino.cc/index.php?topic=575815.0

------------------------------------
Good afternoon,

I cannot get my head around this! So i seek your help and knowledge! My setup:

I got one Arduino 2560, with an ethernet card on it. I build a web interface, where i can use Klik/aan - Klik/uit emitter at 433 Hertz AM. I also can use a 16 relais board with the web interface and i can send IR signals with the same web interface. 

I hoped that the IR signals for my LED string and HDMI switches where different, but unformtunally, some colors are the same for hdmi channel 1 and 5. Bummer. So i thought why not have sperate IR Leds and send the signals seperate. 

Since i do not got much IO, but a lot of digital outs on the board i thought, why not switch the ground on for 1 IR led and of for the other two. Send the signal, tie the IR led to the receiver. And the other IR led to another IR receiver. 

Then i found this topic:
https://github.com/z3t0/Arduino-IRremote/issues/128

Also see the attachment for the picture.

This is what is said about this option:

I tested it with "standart" one IR LED library. All LED's are connected to D3 and the other pin - to digital outputs of Adruino. Then we manipulate digital outputs. You need to turn on / off these outputs before sending the IR signal to LED... You can use many IR LED's (limits only Arduino I/O quantity)

I dont understand how i can manipulate the output as inputs and turn them of, i think i don't understand this concept.

Can anybody help me with some examples and what does he mean?

Thankyou!

simk0024, 2019-04-01:
I can't access this link, https://github.com/khair-eddine/Arduino-IRremote/archive/master.zip
think author has removed the library. 
Any method to control 2 IR Emitters using one Arduino?

z3t0, 2019-04-01:
Here's an old archived copy.

z3t0, 2019-04-01:
[

](url)

z3t0, 2019-04-01:
https://drive.google.com/file/d/0ByTZ9ZV0QjBzRUpwVGcyV0d4MVlmVnI5QTFaQWRMOHZpZUk4/view?usp=drivesdk

AnalysIR, 2019-04-01:
> Any method to control 2 IR Emitters using one Arduino?

You could also just wire the 2 IR Leds in series and drive from 1 pin

simk0024, 2019-04-02:
> Here's an old archived copy.
> […](#)

Thank you, I will try it 

> > Any method to control 2 IR Emitters using one Arduino?
> 
> You could also just wire the 2 IR Leds in series and drive from 1 pin

I am trying to get different signal from 2 IR Emitters.

AnalysIR, 2019-04-02:
> I am trying to get different signal from 2 IR Emitters.

If the target devices  are different, they will ignore any signal received for a different device.

varleygianini, 2019-06-02:
Well I followed the project of adding more pins to infrared but until now I could not connect more than 1 LED Infrared.

My project consists of controlling 4 Air Conditioners of the same model so I can not use a single pin to send the information to the 4 Ar because they all respond to the same code, would you have a functional option for this project?

I am using an Arduino Mega 2560.

[8 more replies]

closed 2020-08-17 by ArminJo