Arduino-IRremote/Arduino-IRremote #132: Turn into RAW

Zhenit-606, 2015-04-20:
Hello guys, before I want to thank immensely for the wonderful library that you develop and maintain. This tool helps many, many people, thank you.

I'm new in electronics and programming (30 days), and start a laser-tag project, the prototype is already up and running. To communicate i using a modified version of irRemote. I cleaned everything leaving only send.SONY, send.RAW and reception.
I'm currently using the transmitting send.SONY, but as the trigger occurs several times, I would send with send.RAW using the same standard of SONY. So I ask your help to do a small code to add in my sketch that turns the value of a variable long in RAW times, for me to send with send.RAW.

Example:
unsigned long turnraw = 123456;
unsigned int sendraw[];
I need a code to turn the turnraw in RAW times based on SONY and store in sendraw.

Excuse my english, I am using Google translator, I hope you understand.
Thanks for listening.

z3t0, 2015-04-26:
@Zhenit-606 Firstly I would like to apologize that no one has replied to your issue until now, I would have replied but have been immensely busy lately. I will take a look and see what I can do, I am no expert with the ir code in this library but I will do my best! @joshuajnoble Any ideas?

z3t0, 2016-07-11:
This may be beyond the scope of this library, but instead something for a helper application?

closed 2016-07-11 by z3t0