Arduino-IRremote/Arduino-IRremote #134: IRemote library on atmega1284p

renno-bih, 2015-04-24:
I'm currently working on project where I use this library in combination with wifi module and I'm running out of flash and SRAM memory. Atmega328 and Atmega32u4 microprocessors just can't handle memory requirements.
Of course, atmega2560 is easy solution but it is pretty expensive and atmega1284p looks perfect for my application.

What is required to use this library with this microprocessor?

khair-eddine, 2015-04-24:
you don't require anything special, you only need to change the IR Led on pin 9 for the atmega1284 i think it's similar to atmega1284p 
and if you want to change the output pin go to program files\Arduino\Librairies\IRremote\IRremoteint.h on line33 and uncomment the wanted IR Led pin and you must let only one uncommented.

z3t0, 2015-04-26:
@renno-bih The above should work, if it does not just let us know and we will help make the necessary modifications to the library to support the chip. Should not be too hard.

z3t0, 2015-04-29:
@renno-bih No problem! Any particular reason for that chip? I am working on my own arduino project and may even get that chip to make things easier.

renno-bih, 2015-04-29:
I'm using Wifi and it requires more SRAM and flash memory, that is why I want to switch to atmega1284p.

I tried prototyping with atmega2560 and CC3000 wireless module but I have additional problems.
Using IR remote library works but as soon as I initialize connected client it stops working properly.

When I initialize ```Adafruit_CC3000_ClientRef client = httpServer.available();``` IRremote library goes crazy.
Before initializing client I get correct IR code for 95% of time, but as soon as I initialize client there is no way to get correct code, type is not recognize and it just saves random raw code.
What could interfere with Adafruit_CC3000_Library ?

z3t0, 2015-07-24:
Timers maybe... Not too sure

z3t0, 2015-07-24:
Please re-open if you need this.

closed 2015-07-24 by z3t0