Arduino-IRremote/Arduino-IRremote #135: compiler error with IRSend Demo

redruff, 2015-04-28:
I know this has come up a few times, but...

I get this with the send demo, it is the stock code right from the Examples folder.

Arduino: 1.6.3 (Windows 7), Board: "Arduino Uno"
Build options changed, rebuilding all
IRrecord.ino: In function 'void sendCode(int)':
IRrecord.ino:103:14: error: 'class IRsend' has no member named 'sendNEC'
IRrecord.ino:107:14: error: 'class IRsend' has no member named 'sendNEC'
IRrecord.ino:113:12: error: 'class IRsend' has no member named 'sendSony'
IRrecord.ino:128:14: error: 'class IRsend' has no member named 'sendRC5'
IRrecord.ino:131:14: error: 'class IRsend' has no member named 'sendRC6'
Error compiling.

I have deleted all the RobotIR libraries, I only have the library "IRRemote" in the IDE library folder.
The REC demo works just fine. Is there a work around for this?

Here is the code throwing the error
/*
 * IRremote: IRsendDemo - demonstrates sending IR codes with IRsend
 * An IR LED must be connected to Arduino PWM pin 3.
 * Version 0.1 July, 2009
 * Copyright 2009 Ken Shirriff
 * http://arcfn.com
 */

#include <IRremote.h>

IRsend irsend;

void setup()
{
  Serial.begin(9600);
}

void loop() {
  if (Serial.read() != -1) {
    for (int i = 0; i < 3; i++) {
      irsend.sendSony(0xa90, 12); // Sony TV power code
      delay(40);
    }
  }
}

Thanks in advance,
Red

khair-eddine, 2015-04-28:
well if you write this code exactly, than you should write #include <IRremote.h> (on line1) instead of #include

redruff, 2015-04-28:
not sure what happened there copy/paste error??

The include is there and correct in the IDE example.

/*
 * IRremote: IRsendDemo - demonstrates sending IR codes with IRsend
 * An IR LED must be connected to Arduino PWM pin 3.
 * Version 0.1 July, 2009
 * Copyright 2009 Ken Shirriff
 * http://arcfn.com
 */

#include <IRremote.h>

IRsend irsend;

void setup()
{
  Serial.begin(9600);
}

void loop() {
  if (Serial.read() != -1) {
    for (int i = 0; i < 3; i++) {
      irsend.sendSony(0xa90, 12); // Sony TV power code
      delay(40);
    }
  }
}

redruff, 2015-04-28:
interesting..it whacks the "<IRremote.h>" probably thinks its HTML tags and gets crabby?

khair-eddine, 2015-04-28:
oh yes i see, have you tried to unplug your board or restart sketch because i see no reason to error 
check it your library is written IRremote too because you writes IRRemote up

redruff, 2015-04-28:
Thanks for all your help! Appreciate it!
I compared upper and lower case, they match. 
I cant get past the compiler error here, so it doesn't even go look for a board to download to.
Ya, I'm stumped on this one!

khair-eddine, 2015-04-28:
you still have compiling problem ?

z3t0, 2015-04-28:
please try the release version of the library, the repository is goin through refactoring as we work on multi ir support.

----- Reply message -----

&mdash;

redruff, 2015-04-28:
where would one find the release version?
Yep Khair-eddine...I'm dead in the water right now.

z3t0, 2015-04-28:
Sorry, would have linked you before but I was on my phone.
https://github.com/shirriff/Arduino-IRremote/releases

khair-eddine, 2015-04-28:
hii bro, @z3t0 working on multi reception or emission ?

redruff, 2015-04-28:
emmission, gonna give the release version a try in a minute.

redruff, 2015-04-28:
oops.sorry you meant that for Z3to!

redruff, 2015-04-28:
Well that did it! At least on my Mac at home! Will try the release version on Winblows machine at work tomorrow! Man that was painful! LOL..
Thanks you two!
Red

khair-eddine, 2015-04-28:
well good for you, i didn't know about the modification :)

redruff, 2015-04-28:
me either! LOL
thanks again for the replies!

z3t0, 2015-04-28:
@khair-eddine @redruff The development workflow we are using now is that we develop the code on the master branch and then release them when things are relatively stable. I am thinking of alternatives because this is not too effective since we do not have the equipment to test some of the changes (like adding new protocols) and so development is slowed.

z3t0, 2015-04-28:
Also, the guides on Sparkfun and other websites have not been updated... I have contacted @SparkFun but they dont seem to be responding

redruff, 2015-04-28:
yep, I went to a couple other websites looking for alternatives, it was all the same.
As a rookie, the examples tend to be pretty solid, so I was a little flummoxed with this one!

z3t0, 2015-04-29:
@redruff Yep, the main reason this one is different is that devs tend to create protocols for something like a LG Remote from documentation and submit the code and then the problem is that the people maintaining the project (like me) usually do not have the required equipment to test with (like lg tv) and so the development in this is not a straight line but one that is crooked.

z3t0, 2015-04-29:
@redruff  use a backtick for code. I'll edit your response above to show you what i mean.

z3t0, 2015-04-29:
@khair-eddine Both actually. As soon as I get my TV working with IR-Remote I am working on a guide to adding new protocols so we can make it easier for junior devs to help as well.

closed 2015-04-29 by z3t0

khair-eddine, 2015-04-29:
@z3t0 yes bro the problem is multiple transmission and multiple reception because multiple reception works alone well and multiple emission works like a charm (at least for me) but the problem is when we use multiple reception with irsend.