Arduino-IRremote/Arduino-IRremote #136: change parameter type for mark & space functions

AnalysIR, 2015-05-04:
Hi guys

The parameter passed to the **mark and space** functions is defined as `int time` (16 bit signed) with a max value of 32767 (dec) or 0x7FFF.

However, both functions make  use of `delayMicroseconds(uSecs)` which has some limitations:
- A value '0' should not be passed which is already handled by both functions
- The maximum valid value that can be passed to `delayMicroseconds(uSecs)` is 16383, which is not handled by either function.

**Problem**
Many IR protocols for media devices and particularly Air Conditioner systems have very long gaps or spaces between sequenetial bursts or packets. Sadly, users often give up on IRremote or implement a workaround by splitting the signal into two parts and placing a `delay(mSecs)` in between these parts

**Suggestion**
- change  the parameter definition for both functions to `unsigned long time`
- change the way the delay is implemented to something like:
```
unsigned long endTime = micros();
while (micros()-endTime)< time){
      ; //do nothing  but wait
}
```
I believe this enhancement has already been implemented in the IRLib fork of IRremote. Might be a good idea to use the same code, used in IRLib.

z3t0, 2015-08-10:
This is a great catch! I will implement this soon. I am currently spending some time learning C/C++ at a more professional level and plan on rewriting this library in a more object oriented style.

AnalysIR, 2015-08-24:
I believe this is now resolve & can be closed

closed 2015-08-24 by AnalysIR