Arduino-IRremote/Arduino-IRremote #137: RAWBUF could be larger

RossosHEX, 2015-05-10:
Not really an exact Issue, just a remark:
If you have Sony or some other R5/R6 protocol Infrared remote control, which apparently sends their signal 3 times instead of using checksum last bits, you may need to open up <IRremote.h> file manually and go all the way down and adjust the RAWBUF constant. By default it is 100, but my codes were of the size 182. I've set the constant to 192 (higher values were causing crashes on my Arduino Nano) and everything started working.

Personal experience with Airwell RC-5(RC) remote controller.

closed 2015-05-10 by RossosHEX

AnalysIR, 2015-05-10:
Yes many AC units fail in IRRemote because of the RAWBUF size limit, which can be changed at the expense of SRAM.

FYI: Don't confuse your Airwell RC-5(RC) with the RC5 IR protocol from Philps. I have seen on several AC units and some Camera controls the use of RC-5 or RC-6 which does not mean they use the RC5/RC6 protocol. (Just an obvious name selected for the model number of hte remote)

However, for your main point it would be useful if the size of RAWBUF could be set from within the ARDUINO sketch.

RossosHEX, 2015-05-10:
Thanks for your quick reply.

I've tried using 
#define RAWBUF 192
after #include <IRremote.h>
within the Arduino Sketch itself, but it didn't work.

AnalysIR, 2015-05-10:
Unfortuately, that is a restriction/limitation of the way it is designed. It shouldn't be too difficult for the developers to make it adjustable.