Arduino-IRremote/Arduino-IRremote #138: Unable to use analogWrite

BladeSlayer1000, 2015-05-14:
Dear,

I'm currently unable to use analogWrite with the current version. Is it a bug, or am I doing something wrong? The code I'm using is;

```
#include <IRremote.h>
#include <IRremoteInt.h>

#define SONY

int RECV_PIN = 3;

IRrecv irrecv(RECV_PIN);

decode_results results;

void setup() {
  Serial.begin(9600);
  pinMode(11, OUTPUT);
  irrecv.enableIRIn(); // Start the receiver
}

void loop() {
	if (irrecv.decode(&results)) {
		Serial.println(results.value, HEX);
		switch(results.value) {
			case 0x63E7335D:
				analogWrite(11, 111);
				Serial.println("Receiver packed");
				Serial.println(" ");
			break;
		}
		irrecv.resume(); // Receive the next value	
	}
}

BladeSlayer1000, 2015-05-14:
I'have fixed the issue. I was using pin 11 and 3, the both use the same timer which causes conflicts with eachother.

Maybe its something to write about in the wiki, so people dont have the same problem as me.

z3t0, 2015-05-14:
Thanks and i will.

&mdash;

closed 2015-08-10 by z3t0