Arduino-IRremote/Arduino-IRremote #139: Re-factoring Code

z3t0, 2015-05-16:
Hi All,

@zenwheel @levsa @Lauszus @PaulStoffregen @shirriff @joshuajnoble 

I am working on re-factoring the library since the community is growing and the library with it. We need to develop a new system for managing features and new additions, especially adding protocols. This is to ensure that the library remains efficient and minimal.

1. We could use #defines within the library that would specify the protocol to be used (e.g. SONY). I do not particularly like this idea because with the help of @khair-eddine and @Neco7 I am working on adding multi receiver/sender support.
2. We could have TWO releases of the library, one being minimized for people who know what they are doing, this will still support multi receiver support but will be a little bit different. This has the advantage of keeping the library minimal for those who know how to use it and need the space for other things or their board does not have enough space. The other release being the standard one which will be number 3
3. We will have a standard library that will allow you to choose your protocol when creating a new irrecv and/or irsend. The focus is once again on establishing multi-receiver/sender support

The main concern here is to think ahead and make sure nothing breaks for multi-support.

I would like to hear your ideas and suggestions. Thanks!

z3t0, 2015-05-16:
Hi, should we remove the blinking of pin 13? I don't really see this as too practical because people are likely to have something connected to 13 that may not be an LED. If we want to keep this we should probably use a #define at the top or something.

khair-eddine, 2015-05-16:
hi, i think it's a good idea to modify the lib or even to create two releases like you said but we first have to make it work without any error. i think blinking led 13 is useful at least for testing the ir led is working or not.. i'm busy with my project now so i need another month then i'll start contributing

z3t0, 2015-05-16:
@khair-eddine Sounds good, contribute when you're free but if you can just reply to my ideas so that I know what would be a good idea.

z3t0, 2015-05-16:
Anyone know the difference between these two sharp sends?
```c++
#ifdef SEND_SHARP
    void sendSharp(unsigned int address, unsigned int command);
    void sendSharpRaw(unsigned int data, int nbits);
  #endif
  #ifdef SEND_IRsendSHARP
    void sendSharp(unsigned int data, int nbits);
  #endif
```

joshuajnoble, 2015-05-17:
They're legacy. Not sure about the provenance.

z3t0, 2015-05-17:
> They're legacy. Not sure about the provenance.

By legacy do you mean they do not work? Or that only one of them is needed. It's kinda hard to find any documentation...

toddtreece, 2015-05-18:
I added one of the sendSharp's a long time ago before this project was on GitHub. I'm not sure if it still works because I haven't used this library in a long time, but I still have the same Sharp TV and can test at some point soon if you'd like.

z3t0, 2015-05-18:
@toddtreece Thanks. In that case I will keep both functions there and release a test version once I am done.

z3t0, 2015-05-18:
Should this library be made c only? I think it may bring in performance improvements, which may or may not be measurable...

AnalysIR, 2015-05-18:
Have you considered merging efforts with the Author of IRLib (also active on GitHub), which is a more recent/modern re-write of IRremote. It is actively supported and has very good documentation.

It seems a pity to dilute any scarce resource/effort across 2 streams!

z3t0, 2015-05-21:
@AnalysIR 
Sorry for the late reply.
I have definitely and still am considering that we collaborate or even form one merged modular library but right now the priority is to get this one stable first.

z3t0, 2015-06-14:
#145 #146 #147 @BlueChip Thanks for all the fixes, I'll incorporate it into this. I'm almost done and will commit to experimental soon.

closed 2015-07-24 by z3t0

pcoughlin, 2015-08-15:
@AnalysIR when you are discussing collaboration on IRlib, is this the one:
https://github.com/cyborg5/IRLib

Thanks.

z3t0, 2015-08-16:
@pcoughlin Yes that is the one. Now that we have this library fairly stable I will be contacting @cyborg5 about merging efforts and working together.

AnalysIR, 2015-08-16:
@pcoughlin Yes, IRLib is a more modern re-write of IRremote & after several iterations has added much more functionality & documentation. It would be great to see efforts merged.

z3t0, 2015-08-16:
@pcoughlin @AnalysIR https://github.com/cyborg5/IRLib/issues/13