Arduino-IRremote/Arduino-IRremote #140: Help with an LG remote

leoditommaso, 2015-05-17:
I'm trying to get my LG air conditioner remote working but I'm a little bit confused. I've tried IRrecvDemo sketch with some remote controls and it works OK but not with my LG remote, it shows nothing in the serial monitor when I press any button on that particular remote.

I've also tried the IRrecvDump with the same remote and I'm getting stuff like the following: 

```
FFFFFFFF
FFFFFFFF (0 bits)
Raw (58): -9450 700 -1350 650 -350 650 -350 650 -400 600 -1400 600 -450 550 -450 550 -450 550 -450 550 -500 500 -500 500 -500 500 -1500 500 -1550 500 -500 500 -550 450 -1550 500 -500 500 -500 500 -1550 500 -500 500 -1500 500 -500 500 -550 500 -1500 500 -550 450 -550 450 -1600 450 
FFFFFFFF
FFFFFFFF (0 bits)
Raw (58): -9450 700 -1350 650 -350 700 -350 650 -350 600 -1450 550 -450 600 -400 600 -450 550 -450 550 -500 500 -500 500 -500 500 -1550 450 -1600 450 -550 450 -550 450 -1550 500 -500 500 -550 500 -500 500 -550 450 -1550 450 -550 500 -500 500 -1550 500 -500 500 -500 500 -500 500 
FFFFFFFF
FFFFFFFF (0 bits)
Raw (58): -9600 550 -1450 550 -500 550 -500 500 -500 550 -1450 500 -500 500 -550 500 -500 550 -450 500 -500 450 -550 500 -600 450 -1550 550 -1450 500 -550 500 -500 500 -550 400 -1550 500 -1500 550 -1550 400 -600 500 -1500 550 -500 450 -600 400 -550 450 -1600 450 -1550 500 -1550 500 
```

What I'd like to know is if this remote is not supported and, in that case, if someone can give me some instructions on what I need to do to add support for it (or where should I read). I've been trying for some hours but I can't find a solution.

Thanks in advance,
Leandro

AnalysIR, 2015-05-18:
Hi
I had a look at your signals and it doesn't look like any typical AC signal I have seen recently. It may be that you are not capturing the complete signal, although your output does seem to be consistent.
(click for larger view)

[image]

The protocol will not be supported by IRremote, unless you add some customisation code to the library yourself. You just need to duplicate the way it is done for other protocols. If it is within your skillset capabilty, it should be obvious.

You should be able to send the signal using IRremote sendRAW...e.g. as follows
(note ignore any leadin spaces, as sendRAW must start with a mark)
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
/*
Automatically Generated by AnalysIR - Batch Export Utility
Registered to: xxxxxxxx
Session History
Type : Key : Value : Bits
0 : RAW :  :  : 0
1 : RAW :  :  : 0
2 : RAW :  :  : 0
*/

// NB: Not all protocols are supported by IRremote or IRLib. You may need to edit the code below manually
// Automatically Generated by AnalysIR for xxxxxxxx, visit http://www.AnalysIR.com or email info@....... for further details
int khz=38; //NB Change this default value as neccessary to the correct carrier frequency

unsigned int Signal_0_0[] = {700,1350,650,350,700,350,650,350,600,1450,550,450,600,400,600,450,550,450,550,500,500,500,500,500,500,1550,450,1600,450,550,450,550,450,1550,500,500,500,550,500,500,500,550,450,1550,450,550,500,500,500,1550,500,500,500,500,500,500,500}; //AnalysIR Batch Export (IRremote) - RAW

irsend.sendRaw(Signal_0_0, sizeof(Signal_0_0)/sizeof(int), khz); //AnalysIR Batch Export (IRremote) - RAW
 // AnalysIR IR Protocol: RAW, Key:  

unsigned int Signal_1_1[] = {550,1450,550,500,550,500,500,500,550,1450,500,500,500,550,500,500,550,450,500,500,450,550,500,600,450,1550,550,1450,500,550,500,500,500,550,400,1550,500,1500,550,1550,400,600,500,1500,550,500,450,600,400,550,450,1600,450,1550,500,1550,500}; //AnalysIR Batch Export (IRremote) - RAW

irsend.sendRaw(Signal_1_1, sizeof(Signal_1_1)/sizeof(int), khz); //AnalysIR Batch Export (IRremote) - RAW
 // AnalysIR IR Protocol: RAW, Key:  

unsigned int Signal_2_2[] = {700,1350,650,350,650,350,650,400,600,1400,600,450,550,450,550,450,550,450,550,500,500,500,500,500,500,1500,500,1550,500,500,500,550,450,1550,500,500,500,500,500,1550,500,500,500,1500,500,500,500,550,500,1500,500,550,450,550,450,1600,450}; //AnalysIR Batch Export (IRremote) - RAW

irsend.sendRaw(Signal_2_2, sizeof(Signal_2_2)/sizeof(int), khz); //AnalysIR Batch Export (IRremote) - RAW
 // AnalysIR IR Protocol: RAW, Key:  
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

If that does not work for you then try the method available on our blog: (this is a more accurate method & works for the longest AC signals)
http://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/
...instructions etc are in the blog post.

leoditommaso, 2015-05-18:
Thanks a lot for your detailed explanation. All of this it's clearly not within my skillset capability, but I'll give it a try, it'd be great for me (and probably for others) if I can get this working. I will let you know the result...

Thanks again dude!

z3t0, 2015-05-18:
@leoditommaso I will see if I can help simplify the process. Also if you have ANY problems just ask and we will do our best to reply, regardless of the problem. Remember, there are no stupid questions.

leoditommaso, 2015-05-18:
@z3to thank you!!

z3t0, 2015-05-18:
@leoditommaso No problem! I am in the same boat as you (no idea how ir actually works). 
I am going to be learning about this and how to implement my own protocol too so I will keep you posted on good resources, here's one now.

http://electronics.howstuffworks.com/remote-control.htm

AnalysIR, 2015-05-18:
Here is a very good site - all about IR remote control signals:

http://www.sbprojects.com/knowledge/ir/

MeantCoder, 2015-07-10:
Hi,

This my basic TV Prank codes.
Step 1 Add TV IR codes 
Step 2 Send IR codes (loop)
Have a Good Day

#include <IRremote.h>

IRsend irsend;

/* Samsung TV Power OFF */
unsigned int Samsung_OFF[68]={4700,4300,600,1650,550,1700,550,1650,650,400,600,500,650,450,550,500,550,550,550,1650,550,1700,550,1650,600,500,650,450,550,500,650,450,550,500,600,500,550,1650,600,500,550,500,650,450,650,400,650,450,650,450,650,1550,550,500,600,1650,650,1600,550,1650,600,1650,600,1650,600,1600,600};

/* Sony TV Power OFF */
unsigned int Sony1_OFF[26]={2350,650,1150,600,600,600,1150,600,600,600,1150,600,600,600,550,600,1200,600,550,600,600,600,550,600,600};
unsigned int Sony2_OFF[26]={2450,600,1200,600,550,600,1150,650,550,600,1200,600,550,600,600,600,1150,600,600,600,550,600,600,600,550};
unsigned int Sony3_OFF[26]={2400,650,1150,600,600,600,1150,600,600,600,1150,600,600,600,550,600,1200,600,550,600,600,600,550,600,600};

/* LG TV Power OFF */
unsigned int LG1_OFF[68]={9050,4400,650,550,550,550,550,1550,650,550,550,550,550,550,550,550,500,550,550,1650,600,1600,600,550,550,1600,600,1650,550,1600,600,1650,600,1600,600,550,550,550,550,550,550,1600,600,550,550,550,550,550,550,550,500,1650,600,1600,600,1600,600,600,500,1550,650,1600,600,1650,600,1600,600};
unsigned int LG2_OFF[4]={8900,2250,550};

/* Philips TV Power OFF */
unsigned int Philips_OFF[40]={2900,750,400,900,450,500,400,500,1400,1300,450,450,400,500,450,500,400,450,400,500,450,450,500,400,500,450,450,450,400,450,500,400,950,500,400,900,400,550,350};

/* Beko TV Power OFF */
unsigned int Beko1_OFF[24]={900,850,1800,900,850,900,850,900,900,850,900,900,850,900,850,900,850,1800,900,900,1750,900,900};
unsigned int Beko2_OFF[24]={850,900,1800,900,850,900,850,900,900,850,900,850,900,900,850,900,850,1750,950,900,1750,900,850};
unsigned int Beko3_OFF[24]={950,800,1800,900,850,900,850,900,900,850,900,900,850,900,1050,700,900,1750,900,900,1750,900,900};

/* Vestel TV Power OFF */
unsigned int Vestel_OFF[36]={8450,4300,500,1650,500,1650,500,600,450,600,500,550,500,550,500,600,450,550,500,1650,500,1650,500,1650,500,600,500,1600,500,600,500,550,500,550,500};

/* Panasonic TV Power OFF */
unsigned int Panasonic_OFF[76]={3650,1900,400,550,400,1450,450,500,450,500,400,550,400,550,400,550,400,500,450,500,400,550,400,550,400,550,400,550,400,1450,450,500,400,550,400,550,400,550,400,550,400,500,450,500,400,550,400,550,400,1500,400,500,450,500,400,550,400,550,400,550,400,550,400,550,400,500,450,1450,400,550,400,1500,400,1450,400};

/* Sharp TV Power OFF */
unsigned int Sharp1_OFF[32]={300,1850,250,800,250,800,250,800,250,800,250,800,300,1850,300,1850,250,800,300,1850,250,800,250,800,250,750,350,1850,250,800,250};
unsigned int Sharp2_OFF[32]={300,1850,250,800,250,800,250,800,250,800,250,800,300,1850,300,1850,250,800,300,1800,300,800,250,800,250,800,250,1850,250,800,300};
unsigned int Sharp3_OFF[32]={300,1850,250,800,250,800,250,800,250,800,250,800,300,1850,300,1800,300,800,250,1850,250,800,250,800,250,800,300,1850,250,800,250};

/* Sanyo TV Power OFF */
unsigned int Sanyo1_OFF[68]={9000,4450,550,1650,550,1650,600,500,600,500,550,1650,650,1550,600,550,500,550,550,600,600,450,550,600,500,550,550,600,500,550,550,550,550,550,550,550,550,550,500,550,650,450,600,500,600,550,600,450,550,550,650,1550,600,1600,600,1650,550,1650,550,1650,550,1650,600,1650,500,1700,550};
unsigned int Sanyo2_OFF[4]={8900,2250,500};
  
void setup()
{  
  Serial.begin(9600); 
}

void samsung_send_ir() {
  irsend.sendRaw(Samsung_OFF, 68, 38);
  delay(40);
}

void sony_send_ir() {
  irsend.sendRaw(Sony1_OFF, 26, 38);
  irsend.sendRaw(Sony2_OFF, 26, 38);
  irsend.sendRaw(Sony3_OFF, 26, 38);
  delay(40);
}

void lg_send_ir() {
  irsend.sendRaw(LG1_OFF, 68, 38);
  irsend.sendRaw(LG2_OFF, 4, 38);
  delay(40);
}

void philips_send_ir() {
  irsend.sendRaw(Philips_OFF, 40, 38);
  delay(40);
}

void beko_send_ir() {
  irsend.sendRaw(Beko1_OFF, 24, 38);
  irsend.sendRaw(Beko2_OFF, 24, 38);
  irsend.sendRaw(Beko3_OFF, 24, 38);
  delay(40);
}

void sharp_send_ir() {
  irsend.sendRaw(Sharp1_OFF, 32, 38);
  irsend.sendRaw(Sharp2_OFF, 32, 38);
  irsend.sendRaw(Sharp3_OFF, 32, 38);
  delay(40);
}

void panasonic_send_ir() {
  irsend.sendRaw(Panasonic_OFF, 76, 38);
  delay(40);
}

void vestel_send_ir() {
  irsend.sendRaw(Vestel_OFF, 36, 38);
  delay(40);
}

void sanyo_send_ir() {
  irsend.sendRaw(Sanyo1_OFF, 68, 38);
  irsend.sendRaw(Sanyo2_OFF, 4, 38);
  delay(40);
}

void loop() {
  samsung_send_ir();
  delay(40);
  sony_send_ir();
  delay(40); 
  lg_send_ir();
  delay(40);
  philips_send_ir();
  delay(40);
  vestel_send_ir();
  delay(40);
  beko_send_ir();
  delay(40);
  sharp_send_ir();
  delay(40);
  panasonic_send_ir();
  delay(40);
  sanyo_send_ir();
  delay(40);
}

z3t0, 2015-07-11:
This might help. https://github.com/shirriff/Arduino-IRremote/blob/experimental/ir_Template.cpp
(Make sure you use the experimental branch for it)

closed 2020-08-17 by ArminJo