Arduino-IRremote/Arduino-IRremote #142: Compiling issue with the experimental branch and problem sending the signal with the exemples.

e-dredon, 2015-05-30:
Hello,
I spent some time trying to get the library working on an arduino UNO R3 (USB controller CH340G), an 1.6 to 6v tolérent IR LED and a VS1838B IR receiver.
I'm using the arduino IDE 1.6.4
None of the demo will compile on the master branch even I solved the conflict with the stock robot IR library, so I finally downloaded the 1.01 experimental branch on github
there was also a fatal error while compiling. Line 750 of the IRremote.cpp, according to the IDE. I had a quick look at the code and indeed 
at the end of the "long IRrecv::decodeWhynter" function, a few lines seemed to be missing, just before  // trailing mark" to complete the "// Data bits" section
I just compared with the same file from the master branch.

the following lines were missing:

		      data <<= 1;
    } 
    else {
      return ERR;
    }
    offset++;
  }
  
So i've added the missing code and the IRrecord demo finally compiled properly.
It would be nice to fix the experimental branch on the git repository.  

Ok so now I'm still having a few issues. 
At first I got only raw garbage from the VS1838B, so I added the 4.7uf and 100Ohm/100KOhm voltage divider as mentioned on the datasheet
Now I can get a quite clean signal, producing IR codes like this :
Unexpected codeType 11
E0E0E01F
Unexpected codeType 6
FFFFFFFF
But I still get randomly raw data like this : 
Received unknown code, saving as raw
 m4350 s4650 m350 s1850 m400 s1800 m400 s1850 m400 s700 m400 s700 m400 s750 m400 s700 m400 s700 m400 s1850 m400 s1800 m400 s1850 m400 s700 m400 s700 m400 s700 m400 s750 m400 s700 m400 s1800 m400 s1850 m400 s700 m400 s1850 m350 s750 m400 s700 m400 s700 m400 s750 m350 s750 m400 s700 m400 s1850 m350 s750 m400 s1800 m400 s1850 m400 s1800 m400 s1850 m400
The LG remote of my aircon is also only showing raw data. 
Last but nbot least, i could not manage to get the IR LED (or a blue LED for the test) working as well. Whatever digital pin I will assign to the pushbutton I also can't get it working to get the arduino send the signal thru the PIN 3. 
IF I set the digital pin low connecting the pushbutton to the GND, I don't get any detection. 
Some help would be greatly appreciate.

I hope i gave all the information you need.

Best regards,

z3t0, 2015-05-30:
Please use the latest stable [release](https://github.com/shirriff/Arduino-IRremote/releases/tag/MAJOR) since the master and experimental branches are undergoing a re-write due to an overlook in regards to the compiling process. #139

AnalysIR, 2015-05-30:
VS1838B.... is possibly the worst IR receiver you could have chosen...see
http://www.analysir.com/blog/2014/12/08/infrared-receiver-showdown-tsop34438-vs-vs1838b-winner-revealed/
...no surprise you get issues

FYI: I have never heard of 6v tolerent IR Led. Typical forward voltage of  an IR LED is around 1.35V @100mA, increasing to 3V @1A.

e-dredon, 2015-05-31:
Wow, I didn't expect such an enthusiastic community. Thanks so much for the 2 lightspeed answers ;)
z3t0: I tried the stable release that the fails to compile with the same variable definition issue as i experienced with the master branch.

AnalysIR: I have a VISHAY TSMP58000 from radioshack on the way, but the transit will take a while. As I went to the electronic store last friday to get stuff for another project I just took the chance to buy a couple of receivers and an IR LEDs witch is 1.35V to 1.6V tolerant (oops, bad copy past editing my first post).

Well today I'll give a try to IRlib. I hope you will join your forces to merge the code and features.

AnalysIR, 2015-05-31:
Unfotunately, the TSMP58000 will not work with IRremote.

This is an IR Learner which provides the actual carrier as well.

You need something like TSOP34438 or another TSOPxx38.

FYI: They are not 1.6V 'tolerant' per-se. The correct terminology is forward voltage drop across the IR Led.

z3t0, 2015-08-10:
@e-dredon Happy that something worked out at the end. I am closing this issue as there does not seem to be any need to leave it open, feel free to comment or re-open if I am mistaken.

Also, all the compilation errors should be fixed, try the master branch.

closed 2015-08-10 by z3t0