Arduino-IRremote/Arduino-IRremote #143: irrecv.resume not working after irsend.sendNEC

ArjanDeVries, 2015-06-01:
Hi,

The following code sends only one time the NEC code. After the resume command nothing is received anymore. Changing the line irrecv.resume() in irrecv.enableIRIn() gives a working solution. So i think something in the send and receive library is conflicting.

My arduino code:

    #include <IRremote.h>
    
    int RECV_PIN = 11;
    
    IRrecv irrecv(RECV_PIN);
    IRsend irsend;
    
    decode_results results;
    
    void setup()
    {
      Serial.begin(9600);
      Serial.println("Program started.");
      irrecv.enableIRIn(); // Start the receiver
    }
    
    void loop() {
      if (irrecv.decode(&results)) {
          Serial.println("Signal recieved.");
          irsend.sendNEC(0xF90AF5,32);
          irrecv.resume(); // Receive the next value  
      }
    }

Regards,
Arjan de Vries

AnalysIR, 2015-06-01:
you must ebale the receiver everytime after sending

  irrecv.enableIRIn(); // Start the receiver

ArjanDeVries, 2015-06-02:
That is indeed what i also see, but I did not excepted that. But if this is the way to do it, i will do it like this.

Thank you very much for your quick answer. btw if this documented anywhere?

Regards,
Arjan de Vries

AnalysIR, 2015-06-02:
I believe it is mentioned in the Author's blog...somewhere.

z3t0, 2017-08-22:
Closing due to inactivity.

closed 2017-08-22 by z3t0