Arduino-IRremote/Arduino-IRremote #144: How to replay a raw code?

marclar, 2015-06-02:
I've used the `IRrecvDump` example to get the signal from my air conditioner's power button:

```
19F6D827
Decoded NEC: 19F6D827 (32 bits)
Raw (68): -16328 9000 -4450 600 -550 550 -550 650 -500 550 -1650 650 -1650 600 -500 600 -550 550 -1650 650 -1600 550 -1700 600 -1650 600 -1650 600 -500 600 -1650 550 -1700 600 -550 600 -1600 600 -1650 650 -500 550 -1700 600 -1650 600 -500 600 -550 550 -600 550 -500 650 -500 550 -1700 600 -500 600 -550 550 -1700 550 -1650 650 -1650 600 
```

Now I'm not sure how to proceed. How do I convert that to be sent using the `IRsendDemo`?

Thanks so much :)

z3t0, 2015-06-02:
Not certain but take a look here https://github.com/shirriff/Arduino-IRremote/blob/master/examples/IRrecord/IRrecord.ino

z3t0, 2015-06-02:
Not sure why you are using that though, try IRrecvDemo and then send with IRsendDemo. That should be better.

The way I understand it the IRrecvDump is just a broken up IRrecvDemo for debugging or adding new protocols.

AnalysIR, 2015-06-02:
Lucky you, that your AC unit uses standard NEC signals. Normally, it is much more complicated for AC signals.

Irsend.sendNEC(0x19F6D827, 32);

alternatively, you could send using sendRAW.

Best approach would be to work thru all of the examples provided with the library first, so you become familiar.

marclar, 2015-06-02:
Awesome -- thanks for the feedback @z3t0 & @AnalysIR -- I'll keep at it.

closed 2015-06-02 by marclar