Arduino-IRremote/Arduino-IRremote #148: sendSharp

csBlueChip, 2015-06-14:
In IRremote.cpp, line 1380 within #ifdef SEND_SHARP 
Two functions are declared, both called sendSharp(...)
The second of which simply calls sendSharpRaw(...)
The function sendSharpRaw(..) does not exist

Looking at the .h file it seems that the first (larger) occurence of sendSharp() should be renamed to sendSharpRaw()

z3t0, 2015-06-14:
Yeah I noticed that too when I was going through... apparently one is legacy or something. I don't really have any Sony products to test it with but someone has agreed to do so once we have a release ready.

csBlueChip, 2015-06-15:
Looking at the header file, I'm pretty convinced it's just a typo ...as it stands a C compiler would belch, I think a C++ compiler will overload the function - which, although quite cool, can be confusing when debugging

closed 2020-08-17 by ArminJo