Arduino-IRremote/Arduino-IRremote #149: Customise

brahbassim, 2015-06-15:
Please how to customize the library to handle other brands? i have a TV witch is Hisense , and this library don't handle Hisense brand  . So i want an issue for this problem, and in future every one could customize and add new brands.

csBlueChip, 2015-06-15:
Yes, I've noticed my DENON and my XBOX(original) remotes produce "UNKNOWN" ...A little article on this would be very useful and I will certainly submit my findings for inclusion :)

z3t0, 2015-06-16:
I will work on a small guide very soon.

csBlueChip, 2015-06-17:
Just stumbled across these:
"Old" Panasonic (22 bits) : http://users.telenet.be/davshomepage/panacode.htm
RC5 : http://users.telenet.be/davshomepage/rc5.htm
Daewoo : http://users.telenet.be/davshomepage/daewoo.htm
Sony : http://users.telenet.be/davshomepage/sony.htm
JVC : http://users.telenet.be/davshomepage/jvc.htm
I've not studied them all yet, but I imagine that doing so will lend an idea to decoding other formats too.

z3t0, 2015-08-10:
@brahbassim Check the ir_template.cpp

z3t0, 2015-08-10:
Feel free to re-open if you need particular help

closed 2015-08-10 by z3t0