Arduino-IRremote/Arduino-IRremote #151: Significant reduction in code size

csBlueChip, 2015-06-15:
This is a reduction in the size of the .cpp file, NOT a reduction in the size of the compiled program

I've noticed in the .cpp there are a lot of blocks such as:
`#ifdef DEBUG
  Serial.println("Attempting RC5 decode");
#endif`

A cleaner (more readable) way to achieve this is like this:

At the top of the .cpp file:
`#ifdef DEBUG
#  define DBG_SER_PRINT(...)    Serial.print(__VA_ARGS__)
#  define DBG_SER_PRINTLN(...)  Serial.println(__VA_ARGS__)
#else
#  define DBG_SER_PRINT(...)
#  define DBG_SER_PRINTLN(...)
#endif`

Now you can replace these three lines:
`#ifdef DEBUG
  Serial.println("Attempting RC5 decode");
#endif`

with the single line of code:
`DBG_SER_PRINTLN("Attempting RC5 decode");`

Also the block of code:
`#ifdef DEBUG
  if (val == MARK) {
    Serial.println("MARK");
  } 
  else {
    Serial.println("SPACE");
  }
#endif`

can be replaced with the single line:
`DBG_SER_PRINTLN( (val == MARK) ? "MARK" : "SPACE" );`

As I say at the top, this will make zero difference to the compiled program, it will just make the code a lot smaller and more readable

z3t0, 2015-06-15:
Anything related to code errors can be posted in #139 to keep things organized.
Thanks for all your contributions, I am almost done on the rewrite and will commit to experimental sometime in the coming weeks, for the next few days I will be busy with other things but will still reply to issues on Github.

z3t0, 2015-06-15:
Definitely a good idea

z3t0, 2015-06-18:
@csBlueChip I am away for a while and thought that it would be helpful to use the label "Write Code" to signify issues directly related to code changes that may or may not affect other things. Anything that mainly focuses on code can be labeled by that.

z3t0, 2015-06-18:
@csBlueChip Also please mark anything relying on the re write with the 2.0 milestone... just helps keep things organized :D

ArminJo, 2020-08-17:
Is implemented

closed 2020-08-17 by ArminJo