Arduino-IRremote/Arduino-IRremote #152: Faster+smaller send() loops

csBlueChip, 2015-06-15:
BTW. If the code wasn't undergoing a major overhaul right now I would fork+pullRequest these changes, I think I've just arrived at a bad time - anyway, I hope these comments are appreciated - if not, let me know and I will shut up ;)

I notice that the send loops have instructions which can be removed. I have picked one at random and corrected it here:

original:
```void IRsend::sendRC5(unsigned long data, int nbits)
{
  enableIROut(36);
  data = data << (32 - nbits);
  mark(RC5_T1); // First start bit
  space(RC5_T1); // Second start bit
  mark(RC5_T1); // Second start bit
  for (int i = 0; i < nbits; i++) {
    if (data & TOPBIT) {
      space(RC5_T1); // 1 is space, then mark
      mark(RC5_T1);
    } 
    else {
      mark(RC5_T1);
      space(RC5_T1);
    }
    data <<= 1;
  }
  space(0); // Turn off at end
}```

Update:
```void  IRsend::sendRC5 (unsigned long data,  int nbits)
{
  enableIROut(36);  // Set carrier frequency to 36KHz

  // Start marker
  mark(RC5_T1);
  space(RC5_T1);
  mark(RC5_T1);

  for (unsigned long mask = (1 << (nbits - 1));  mask;  mask >>= 1) {
    if (data & mask) {  // 1 is space-mark 
      space(RC5_T1);  
      mark(RC5_T1);                            
    } else {            // 0 is mark-space
      mark(RC5_T1);
      space(RC5_T1);
    }
  }
  space(0); // Ensure LED is off at the end
}```

Instead of shifting the data at the start and then shifting it again every time around the loop, I have created a mask which starts with the correct bit and shifted that instead ...this mask manipulation replaced the loop counter.

z3t0, 2015-08-10:
@csBlueChip Has this been implemented in your rewrite?

csBlueChip, 2015-08-10:
Yes :)
Feel free to close

z3t0, 2015-08-11:
> Yes :)
Feel free to close

Cool, thanks for the contribution!

closed 2015-08-11 by z3t0

pcoughlin, 2015-08-22:
Why did you comment out the mark,space,mark preamble?
Is the preamble part of the RC5 coding or not?
Does it still work after it being taken out?  I do not have any RC5 devices, so I cannot test it.
@csBlueChip what device did you use to test the removal of the preamble?

z3t0, 2015-08-23:
@pcoughlin Information on the protocol can be found here http://www.sbprojects.net/knowledge/ir/rc5.php
Also the harctoolboxbundle may have support for decoding RC5, so you can try using that to test. We can add this to the new library, if possible can you make a list of protocols to add?

pcoughlin, 2015-08-23:
@z3t0 @csBlueChip The description of the protocol on the link provided above clearly shows that there are Start-Bits(S1, S2, T).

Are you assuming that the start-bits are in the "data"?
What about the "address" in the protocol description?
Looks like the original "mark; space; mark" did not match the protocol either.
I read it to have the preamble of "space; mark; space; mark; mark; space" for S1; S2; T.

And if the preamble is in the data, then we should not need a RC5 special routine at all.  Just a generic one that defines 1=space+mark and 0=mark+space.

How do the two of you read that info?

z3t0, 2015-08-23:
@pcoughlin I did not take a look at the implementation before as I had no way to test it, but looking at the link I provided, there are start-bits. You're right, do you have any remotes with the same protocol so we can test?

pcoughlin, 2015-08-23:
I do not have any remotes for RC5 (that I know of).

@csBlueChip Do you have any RC5 devices or remotes that can be tested?

AnalysIR, 2015-08-23:
FYI: most universal remotes will have it under one of Philips codes

z3t0, 2015-08-24:
@AnalysIR Good point... Unfortunately I returned my only universal remote (along with a tv box) just a few days ago. I might get another one in a few days when my new tv box shows up. 

Could we not just use an Arduino and this library? Can anyone verify that the RC5 send protocol in this library works?

AnalysIR, 2015-08-24:
Perhaps the real question as mentioned before is what is the correct way to implement it. That should be confirmed first so people know what to test against.

Having the start bits etc hardwired (not part of the bits) or treated as part of the bits. I would suggest having the implementation behave the same as previous versions of the IRremote library, to maintain consistency. (No point introducing changes unless neccessary and likely breaking code of existing users).

pcoughlin, 2015-08-24:
I agree with @AnalysIR.  Making changes to the preamble bits will change the way it works for anyone that updates the library and was using that encoding method.