Arduino-IRremote/Arduino-IRremote #154: POssible bug in AIWA send() function

csBlueChip, 2015-06-18:
I cannot for the life of me work out how to add the "write code" tag - help!

See inline comment:
```
//-v- THIS CODE LOOKS LIKE IT MIGHT BE WRONG
//    it only sends 15bits and ignores the top bit
//    then uses TOPBIT which is bit-31 to check the bit code
//    ...because of how codes are built by the receiver
//    I suspect "TOPBIT" should be changed for "0x00008000"
//    Of course, if this code is known to work, then I am clearly wrong
//    but I have nothing with which to test it :/
  // Skip firts code bit     
  code <<= 1;
  // Send code
  for (i = 0;  i < 15;  i++) {
    mark(AIWA_RC_T501_BIT_MARK);
    if (code & TOPBIT)  space(AIWA_RC_T501_ONE_SPACE) ;
    else                space(AIWA_RC_T501_ZERO_SPACE) ;
    code <<= 1;
  }         
//-^- THIS CODE LOOKS LIKE IT MIGHT BE WRONG
```

z3t0, 2015-06-18:
screenshot shows how to add the tag.

[image]

Basically on the right side bar (left of the issues/pull requests) click labels and search for the write code label.

z3t0, 2015-06-18:
@csBlueChip Yes that seems to be one of the main problems, it becomes hard to maintain code when we do not have access to the equipment required to run tests...

csBlueChip, 2015-06-18:
@z3t0 Labels: "Labels\nNoneYet", "Milestone", "Asignee" are all just text for me (nothing clickable) maybe there is a github option to enable "label changing for all commenters" or such? :/
Testing: Yes, we need to educate the Arduino community "if it doesn't work for you, *come and talk to us* ...programs/projects are made by more minds than the code jockeys who type stuff in" :)

csBlueChip, 2015-06-19:
I have emailed @crash7 and asked him to review this issue ...hopefully he will be able to verify if I am right or wrong.

z3t0, 2015-06-19:
@csBlueChip Sounds good. The reason I wanted to work of the Release was because the master branch as of now is unstable and it would have been more reliable to start from a stable state

crash7, 2015-06-19:
Hi all, 

@csBlueChip I will check this, thanks for reporting it! Give me a couple of days :)

@z3t0 I believe that reporters can't assign labels to their issues (or pull requests), only collaborators can!

z3t0, 2015-06-19:
> @z3t0 I believe that reporters can't assign labels to their issues (or pull requests), only collaborators can!

Okay I had no idea. I'll just assign them from now on then.

z3t0, 2015-07-24:
Any update? Thanks

crash7, 2015-07-24:
@z3t0 sorry both, didn't have the time to check this yet. I will try to do it tomorrow. Thanks for the reminder

z3t0, 2015-07-24:
No problem, take your time

ArminJo, 2020-10-19:
@z3t0 @bengtmartensson
Hi, I have had a look at the AIWA protocol implementation.
First it is not a family like NEC, it is only for one remote and one device (audio system NSX-V929) see http://lirc.sourceforge.net/remotes/aiwa/RC-T501
Second the send function had a bug. Instead of if (code & 0x80000000) it must be if (code & 0x8000). The 15 bits are correct according to the LIRC documentation.
Maybe the decode function is correct, but nobody knows.

I would like to **remove the code completely** especially since it was never tested and is only for one device.

If I do not get feedback to keep the code, I will delete it for the version 2.8.0, where I also removed Mitsubishi protocol, since the implementation is in contradiction with all documentation I could found and therefore supposed to be wrong too.

bengtmartensson, 2020-10-19:
Ditch it. IMHO, a questionable protocol implementation causes more confusion than it helps the community.  And please do not consider Lirc as a reliable source on IR protocols.

FYI: IrpTransmogrifier/IrScrutinizer uses the definition

    {38.123k,550}<1,-1|1,-3>(16,-8,D:8,S:5,~D:8,~S:5,F:8,~F:8,1,-42,(16,-8,1,-165)*)[D:0..255,S:0..31,F:0..255]

i.e. 21 bits payload; I'd call that "a family".

ArminJo, 2020-10-19:
> i.e. 21 bits payload; I'd call that "a family".
Is the 26 bit of 0x0227EEC0 (or 0x1D8113F) preamble valid for all members of the family?

crash7, 2020-10-19:
Hi,
The code as-is worked for me in the past, but:
1) I'm not using that device anymore (and the arduino with the code wasn't reflashed again)
2) I can't test the proposed solution (I don't have tthe device with me)

So if your logic says that the code should be different, go ahead and remove it.

ArminJo, 2020-10-19:
Thanks for your feedback 👍

closed 2020-10-19 by ArminJo

z3t0, 2020-10-20:
Removing the code is fine by me