Arduino-IRremote/Arduino-IRremote #155: Help to disable Led 13 blink

EricSKennedy, 2015-06-19:
Thanks for a great and useful library!  This is more of a support request, because I think you've already got this feature included, I just can't seem to figure out how to use it.  The library seems to use Pin 13 to give visual feedback on the IR receive.  I've been contributing to a Blockly-based multi-language arduino code generating IDE.  My latest add was a series of blocks for IR remote control of simple cars.  In testing I found a conflict with a motor shield I had which used Pin 13 for directional control.  Once the  library was initialized, the motor stopped working.  On the other hand, using the library with a Pololu Zumo worked fine (No motor controls on Pin 13).

All that to say....
Can you help me with the switch to turn off any use by the library of Pin 13?  Am I missing something?

Shield link:
https://cdn.sparkfun.com/assets/b/8/6/c/2/527c0c05757b7fc33c8b4567.png
Example of the blocks in action
[image]
Main 
https://github.com/technologiescollege/Blockly-at-rduino
My fork
https://github.com/EricSKennedy/Blockly-at-rduino

AnalysIR, 2015-06-19:
In IRremoteInt.h you can change the pin from 13 to whatever pin you want. (in the local copy of the library on your system)

EricSKennedy, 2015-06-19:
Thanks!  Can I disable blinking entirely?  Is there somewhere I can set blinkflag to false?  Will that turn it off? or should I use  "#define BLINKLED       0"  Ideally I'd love to send in a parameter from base adruino code without modifying the library.  That way I can leverage your excellent work without having to modify it every time.

AnalysIR, 2015-06-19:
Try 

````
....
#define BLINKLED       99
#define BLINKLED_ON()  
#define BLINKLED_OFF()
#endif

````
Making the pin something invalid like 99, should stop the pin being set as output

the other 2 defines just serve to turn on/off defines to nothing.

There doesn't seem to be a way to select the led pin or disablle it - generically

EricSKennedy, 2015-06-19:
Thanks, I'll give that a try...

closed 2015-06-19 by EricSKennedy

fatcarko, 2020-02-13:
Hi AnalysIR, 
I can't make the code work with a **Controllino MAXI**
[](https://www.controllino.biz/controllino-maxi/maxi-and-maxi-pure/)

I think it is related to the Led_Pin defined there (but really not sure), I tried, your technique, but still crashes at compilation. 

Maybe it is something else. Do you know what to do ? 
On my system, the <IRremote.h> in declaration looks grey, it's weird. It works on my Arduino UNO, but not with the Controllino (requires its own library). 
The printed error is "exit Status 1".

Thank you very much !!!

Felix