Arduino-IRremote/Arduino-IRremote #156: Recorded as NEC, Sent back shows unknown?

MACscr, 2015-06-20:
Ok, so when I record a button from my LG remote using the IRRecord sketch, its showing the code as:

`Received NEC: 20DF40BF`

So then I enter that into the IRsendDemo on my other arduino that has the IR led connected to it:

`irsend.sendNEC(0x20DF40BF, 32);`

But back on receiving arduino (just doing this for testing), im getting:

`Received unknown code, saving as raw
 m8700 s4650 m400 s750 m350 s750 m400 s1750 m400 s700 m450 s650 m450 s650 m450 s700 m400 s700 m400 s1750 m450 s1700 m450 s700 m400 s1700 m450 s1700 m450 s1700 m450 s1700 m450 s1700 m450 s650 m450 s1750 m350 s750 m350 s800 m350 s750 m350 s750 m350 s750 m450 s700 m400 s1700 m400 s700 m400 s1750 m450 s1700 m450 s1700 m450 s1700 m450 s1700 m450 s1700 m450`

What can I do? Shouldn't it recognize it again as NEC just as if i was using the remote itself? Right now I am having zero luck with any of the methods I have tried in order to control my tv through an arduino. =/

AnalysIR, 2015-06-20:
Looking at the timings you are getting on the receiving Arduino - some are way off.
The timings are 9000,4500,560 & 1690. Normally +/- 100 or 150 uSecs is OK, but you are getting as low as 350 which looks like the problem.

I supect your IR receiver might be the cause. What is the model number / data sheet link?. 
How is your IR Led wired up for tx?

csBlueChip, 2015-06-20:
I originally purchased a VS1838B IR receiver module from China ...It produced, at best, garbage, and some remotes (such as my first-gen-Xbox remote) were not recognised at all ...I purchased a Vishay TSOP4838 (note the important "38" on the end of that, it means 38KHz) [from Rapid Electronics] and all was well ...Another thing that you might want to check is that you have an appropriate filtering capacitor+resistor over the receiver - precise details of this should be in the datasheet for the receiver.

AnalysIR, 2015-06-20:
+1
VS1838B is total rubbish.
see: http://www.analysir.com/blog/2014/12/08/infrared-receiver-showdown-tsop34438-vs-vs1838b-winner-revealed/

For the better quality Vishay IR receivers, the capacitor+resistor are usually optional, to reduce noise.
The TSOP34438 model would be my favourite, these days.

MACscr, 2015-06-20:
I find it so odd that the ones Receiver/transmitter pack I received from Amazon had such great reviews if they are so bad and they don't even sell any of the TSOP34438's. eBay doesn't sell them either. Are we sure the issue is the receiver and not the transmitter?

- Mark

AnalysIR, 2015-06-20:
Not sure of anything....I guess more than a few things sold by either of the sites you mentioned are rubbish. Try Digikey, Mouser, Newark/Farnell/Element14 or RS for electronic components.

....is there a reason you haven't posted the model nos/link?...or how you have wired up the IR Led.
Difficult to provide informed assistance....'in the dark' :)

MACscr, 2015-06-20:
These are what I bought: http://www.amazon.com/gp/product/B00EFOQEUM?psc=1&redirect=true&ref_=oh_aui_detailpage_o00_s00

This is the receiver: http://chioszrobots.com/2014/02/11/vs1838-tl1838-vs1838b-universal-infrared-receiving-head-remote-control/. I simply have it plugged into 11 for data, 5v for power, and then the ground. All on the Uno R3.

For the transmitter, I don’t have a data sheet (neither actually came with one), but I have it connected to a Nano with ground to ground and then a 200ohm resistor before the data pin and connected to pin 3 on the nano.

-Mark

https://github.com/shirriff/Arduino-IRremote/issues/156#issuecomment-113819024

csBlueChip, 2015-06-20:
@MACscr I'm afraid I only have bad news... This is what I bought originally: http://www.ebay.co.uk/itm/151332038832 ...The moment I replaced it with this: http://www.rapidonline.com/electronic-components/tsop4838-infrared-receiver-38khz-55-0905 all my problems instantly went away ...I've not tried the LEDs that came from ebay yet, so I can't comment on those
As for "positive comments" this was in the news this week: http://www.bbc.co.uk/news/business-33186836 ...who knows if it is also true of Chinese exporters <shrug>

AnalysIR, 2015-06-20:
Ok
90%+ probability the IR receiver is an issue.
The IR LEds are probably OK or at least usable.

Best to get an IR receiver by Vishay. Most TSOPxxx38 will be better than the one you have.

You could try a 75R or 100R resistor instead, which might improve things, which should get you 3-4m range & make sure the top of the IR Led is pointed at the receiver.

Better still do up a circuit using a transistor driver

MACscr, 2015-06-20:
Ordered. Though I just need it yo learn ygr codes properly. System will only transmit after.

MACscr, 2015-06-21:
So I took another receiver from a HP MCE receiver that I had. Though it was super sensitive to light, I did most of the time successfully get the NEC code for turning on my tv from my remote. I then put that code into the sketch for my Nano and it actually successfully showed the it receiving the NEC code on the Uno most of the time too. Unfortunately it still doesn't actually work on my tv to turn it on from the Nano. Range shouldnt be an issue as i tried it at an inch, a foot and even 10 feet.  =/

MACscr, 2015-06-21:
How do i know what the second value (32) should be in irsend.sendNEC(0x20DF40BF, 32)?

AnalysIR, 2015-06-21:
32 is correct as NEC is a 32 bit protocol with a carrier frequency of 38 kHz.
(You will notice as well that every second byte is the inverse of the one before.)

If the nano is running at 3V3 (vs 5V) then you could reduce the value resistor. I suspect your next step should be to build a transistor driver for sending the IR. (examples via google)

AnalysIR, 2015-06-21:
Maybe the IR LED is a 'cheapo' as well.

FYI: TSAL6x00 are good ones from Vishay. (TSAL6100,TSAL6200 or TSAL6300)

...read the data sheets to see the difference in angle vs range

MACscr, 2015-06-21:
I got it working! I accidentally had it as 38 instead of 32!

csBlueChip, 2015-06-21:
Fantastic news! Well done! ...Thanks for keeping us updated all the way :)

MACscr, 2015-06-21:
Thanks everyone for the help!

I think i got the idea of 32 from a tutorial and posted it here, and then AnalysIR confirmed it and then I realized I wasn't even using that value. Glad I asked! If each type (NEC, SONY, etc) have their own protocol that should be used and it's known, why isn't that in the readme so we don't have to guess? Or heck, why not have it built into the function instead of having it specified as one of the values? Also, it wasn't clearly even known to me when i first started all of this that I had to add 0x to all my codes. Thoughts?

csBlueChip, 2015-06-21:
If I read z3t0 correctly, he has (in principal) accepted my pull request and will perform the pull inside the next couple of weeks ...If so, take a look at https://github.com/csBlueChip/Arduino-IRremote/blob/master/ir_Denon.cpp ...A lot of what you have said is there; although I think you are also correct "why do I have to pass in 'nbits'"?? Maybe there are a couple of protocols that have variable length fields and the code just got replicated with consideration <shrug> ...As for "0x", that's a C programming thing :/

MACscr, 2015-06-21:
but couldn't 0x just be added dynamically to the output for each copy and paste or to the input before its processed?

AnalysIR, 2015-06-21:
I suspect, if you had read the blog provided by the library Author, many of these things would have been clearer to you. (linked at the top of the code & examples)

....adding 0x to hex values does sound like a good idea. However, as it is a library which has to run alongside other code, the choice **may** have been to prioritise smaller Flash/SRAM usage vs usability???

csBlueChip, 2015-06-21:
@AnalysIR are you *sure* that MACscr's questions about 'code structure and bit length passing' and 'hex prefixing' are made clear in Ken's blog?

AnalysIR, 2015-06-21:
I think that if you read the blog posts & comments and work thru all of the examples provided with the library, you will have a good understanding of how to effectively use the library.

It seems the 0x convention of using HEX values is used within the library examples. Adding it to the values printed out by the library is a godd suggestion.

A search on the Arduino forum will uncover lots of experience & resolutions - for any newbie.

Of course a dedicated user guide, which covered everything, in one place, would be a great asset.

As a contrast, there is an excellent set of documentation provided with the IRLib library, much of which is applicable for IRremote, because the author re-implemented IRremote & added lots more functionality.

csBlueChip, 2015-06-22:
@AnalysIR OK, So all he needed to do was read Ken's blog posts & all the comments people made about them; then (with so little knowledge of programming that "0x" was ellusive) read and comprehend a few hundred lines of poorly commented C/C++ spread over several examples; then go to the Arduino forums and read an unspecified number of the many tens-of-thousands of posts made by n00bs & experts alike before wishing there was an actual guide that helped newcomers; and finally track down an alternate library and comprehend that before translating the information procured back over to this library ...He could then have stabbed a guess as to the rationale of why a parameter is passed to a function which does not require it ...At which point he could come here and ask if his guess is accurate.
...You make a fair point.

AnalysIR, 2015-06-22:
Hmmm... :(

...All anyone needs to do is to carry out a 'reasonable' amount of work.

I think you missed an important bit => The OP did get the support here & the issue was resolved!

Time to close this thread.

csBlueChip, 2015-06-22:
I have addressed the "output something that can be cut'n'paste'd" issue in IRrecvDumpV2.ino
Here is a test output:
```
Encoding  : PANASONIC
Code      : 4004:D08BCB9 (48 bits)
Timing[100]: 
     -19714
     +3500, -1700     + 500, - 350     + 500, -1250     + 500, - 350
     + 500, - 400     + 500, - 350     + 500, - 400     + 500, - 350
     + 500, - 400     + 500, - 350     + 500, - 400     + 450, - 400
     + 500, - 350     + 500, - 400     + 500, -1250     + 500, - 350
     + 500, - 400     + 450, - 400     + 500, - 350     + 500, - 400
     + 500, - 350     + 500, -1250     + 500, -1250     + 500, - 350
     + 500, -1250     + 500, - 400     + 450, - 400     + 500, - 350
     + 500, - 400     + 500, -1250     + 500, - 350     + 500, - 400
     + 500, - 350     + 500, -1250     + 500, - 350     + 500, -1250
     + 500, -1250     + 500, -1200     + 500, -1250     + 500, - 400
     + 500, - 350     + 500, -1250     + 500, - 350     + 500, -1250
     + 500, -1250     + 500, -1250     + 500, - 350     + 500, - 400
     + 500, -1200     + 500, 
unsigned int  rawData[101] = {34473, 70,34, 10,7, 10,25, 10,7, 10,8, 10,7, 10,8, 10,7, 10,8, 10,7, 10,8, 9,8, 10,7, 10,8, 10,25, 10,7, 10,8, 9,8, 10,7, 10,8, 10,7, 10,25, 10,25, 10,7, 10,25, 10,8, 9,8, 10,7, 10,8, 10,25, 10,7, 10,8, 10,7, 10,25, 10,7, 10,25, 10,25, 10,24, 10,25, 10,8, 10,7, 10,25, 10,7, 10,25, 10,25, 10,25, 10,7, 10,8, 10,24, 10,0};  // PANASONIC 4004:D08BCB9
unsigned int  addr = 0x4004;
unsigned int  data = 0xD08BCB9;
```

closed 2020-10-12 by ArminJo