Arduino-IRremote/Arduino-IRremote #162: atmega 328p at 12MHz

jfardello, 2015-06-27:
atmega 328p at 12MHz (Adafruit Pro trinket 3v) is not working, at least for the NEC protocol, I've tried the same code on arduino uno and teensy and it is working fine. 

Could it be the scaling not generating the right freq? Any suggestion on how to debug without an oscilloscope?  

thanks,

AnalysIR, 2015-06-27:
You will need to manually adjust the timer definitions within the library to work with 12MHz instead of the more common 16MHz or 8MHz. 

try setting 
`#define F_CPU 12000000`
before you include the library.

If that doesnt work then just replace the following code in IRremotInt.h
`    #ifdef F_CPU
    #define SYSCLOCK F_CPU     // main Arduino clock
    #else
    #define SYSCLOCK 16000000  // main Arduino clock
#endif`

to
`#define SYSCLOCK 12000000  // main Arduino clock`

jfardello, 2015-06-27:
When setting the board to be the 3v trinket, F_CPU is already defined as 12000000. Also printing out SYSCLOCK also outputs 12000000, that's why I think the prescaler calculation might not be right in this case.

AnalysIR, 2015-06-27:
OK - can't think of what could be wrong and I don't have a 12MHz system to test, it should be a relatively straightforward calculation.

Have you confirmed the millis are correct/accurate? AVR Fuses set correctly? Are you using Timer2?

jfardello, 2015-06-27:
thanks for the quick response, I didn't messed with fuses if you mean that, first I thought it could be caused by using softwareserial at the same time (may be uses timer2 ?), but I can reproduce it by just calling sendNEC() on a button press.

The point is that I don't fully understand what's going on inside TIMER_CONFIG_KHZ, and why OCR2B is set to SYSCLOCK / 2000 / val / 3.

May be in the next days I can try debugging with xmode2 from the lirc project, because I can't see if the timer really gets configured near 38khz PWM. ( I'm planning to wire a tsop to an ftdi cable).

AnalysIR, 2015-06-27:
Why not try configuring the use of Timer1 instead of Timer2...in case there is a conflict.

jfardello, 2015-06-27:
Mostly because I can reproduce it without the softwareserial code.

v5k248, 2016-02-28:
I have a similar/related question: does anyone know why TIMER_CONFIG_KHZ macro sets OCR2B to 1/3 of OCR2A? That results in asymmetric waveform, right? I.e. instead of the pin being high for half of the period of the modulation frequency and low for another half, the pin is set to high for 1/3 of the period and to low for the other 2/3. So for 38KHz modulation the pin is high for about 9us and low for about 17us instead of being high for 13us and low for 13us.  

I checked with the scope and it looks like that's exactly what's happening. But a couple remotes that I am using generate a symmetric waveform for modulation. So what's the thinking behind making it asymmetric?

bengtmartensson, 2016-02-28:
I do not know why someone made that very decision. But some general thinking: 

Most IR-controlled devices use a demodulating receiver, TSOP* or something, as the first component of the receiving chain. It removes the carrier, and with that the duty cycle, so that the rest of the electronics will not see it anyhow. By lowering the duty cycle to somewhat less than 50% you consume less energy, and lessens the stress on the IR LED. (Since they are often overdriven (used with a current that would be too high if sent continuously), this can be a real issue.)  Most of the first "tone" is there already.

Excercises  :-) :
1. (theoretical) compute the Fourier series of a  square wave  with duty cycle d.
2. (practical) perform some experiments on how the usable range depends on the duty cycle.

AnalysIR, 2016-02-28:
Carrier Duty cycle can range for 50% to 5%, with 50% & 33% being the most common I have seen in consumer devices. Vishay refer to it in the context of battery power consumption being more efficient when the duty cycle is lower.

The AGC of different  IR receiver models is designed to react to specific combinations of 'Envelope Duty Cycle' & Burst lengths, which will vary for different IR Protocols. 

As mentioned above, lower duty cycle will be kinder on driver circuits @ higher current levels (IR LED, transistor, current limiting resistor etc). I suspect that for 90%+ of applications the default setting in the library is fine.

In some lasertag systems they adjust range by varying the duty cycle.

v5k248, 2016-02-28:
Thank you for the responses! So basically that's indeed setting carrier duty cycle to 33% to conserve power and reduce stress on the circuit. I didn't know that was a common practice in consumer devices, thanks!

Would be nice to have a comment about that in the code somewhere around [here](https://github.com/z3t0/Arduino-IRremote/blob/3aebf42ca8daa865ce8ae572245328e7443e438e/irSend.cpp#L68). That would have saved me some confusion. I can add it but not sure it's worth creating a new pull request.

v5k248, 2016-02-28:
@bengtmartensson out of curiosity - what was the point of the "theoretical" exercise of computing Fourier transform?

bengtmartensson, 2016-02-29:
Ok, First, I requested the Fourier series, not the -transform. 

it is well known that a (symmetric) square wave of frequency f can be "written" as the infinite series of the odd sines:

(1/1) * sin(omega*t) + (1/3) * sin(3*omega*t) + (1/5) * sin(5*omega*t) + ....

with omega = 2 * pi * modulation_frequency, (omitting the constant term). For duty cycle d, 0 < d < 1, this instead will be

c1(d) * sin(omega*t) + c2(d) * sin(2*omega*t) + c3(d) * sin(3*omega*t) + c4(d)*sin(4*omega*t) + c5(d)*sin(5*omega*t) + ...

for the c_i_(d) being functions of d (which can be relatively easily computed as a certain integral). From comparing with the above, we know that c1(0.5) = 1, c2(0.5) = 0, etc. Now, the band pass filter in a receiver is (or at least should be) centered as the modulation frequency, so it is reasonable to assume that it kills all the component except for the "root tone" sin(omega*t). So it is really a matter on how c1(d) depends on d; it has a maximum at d = 0.5 (symmetric square) and away from that maximum, it decreases "a little". The size of c1(d) is really the quantitative measure of what we loose with  d < 0.5.

Math is fun...

v5k248, 2016-03-07:
Got it, interesting exercise. So I followed an example [here](http://lpsa.swarthmore.edu/Fourier/Series/ExFS.html), turns out that for 50% duty cycle c1=0.6366 and for 33% c1=0.5513 or about 87% of what it it was for 50% duty cycle. So you're right, the difference is not that great. Btw, this means that by decreasing duty cycle from 50% to 33% we are decreasing by (100-87)=13% the power of the signal that passes the filter in the receiver, right?

bengtmartensson, 2016-03-09:
yes, that is basically it, assuming that "power" depends linearly on the amplitude. If e.g. "power" (like the normal electrical engineeering usage) depends quadratically on the amplitude  it is 0.87^2. (I do not know which is the most realistic model.)

ArminJo, 2020-07-03:
The  duty cycle can now be changed by adjusting the DUTY_CYCLE in IRremoteBoardDefs.h

closed 2020-07-03 by ArminJo