Arduino-IRremote/Arduino-IRremote #163: IR remote sends two codes in quick succession, only first one picked up.

jmwohl, 2015-06-27:
I have an off-brand remote that sends two signals in quick succession when the power button is pressed. The library picks up the first, but the second is ignored... seems like the gap is big enough that it's recognized as and "end of code" but small enough that it isn't ready to handle the following signal.

Is there a way to increase the duration of time that needs to occur before a signal is deemed 'complete' and ready for decoding? I've been playing around with adjusting the _GAP value, but can't get it to do what I'm hoping.

Any tips would be appreciated.

AnalysIR, 2015-06-27:
You could try the sketch in this blog post:
http://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/

It wont decode the signal, but will record all the timings.

Is teh first part of the signal decoded by IRremote?

jmwohl, 2015-06-27:
That post looks like *exactly* my problem, thanks! 

Yes, the first part is decoded by IRremote, and decoded as a Samsung code.

jmwohl, 2015-06-27:
@AnalysIR - You, sir, are a gentleman and a scholar. Worked like a charm, thanks much!

closed 2015-06-27 by jmwohl

AnalysIR, 2015-06-27:
Great!