Arduino-IRremote/Arduino-IRremote #164: Compiler bug break library

csBlueChip, 2015-06-30:
Well, after a fair chunk of testing I have discovered that my "optimise the loops" patch has broken a bunch of code ...and it seems to be a bug in the compiler - someone please tell me I am wrong!
```
unsigned long  mask = 1 << (32 - 1);
Serial.println(mask, HEX);
```
Expected result: `0x80000000`
Actual result: `0`

Despite:
```
unsigned long  mask = 0x80000000;
Serial.println(mask, HEX);
```
Result: `0x80000000`

It's too late to work out what the heck is going on tonight, but be confident that now I know where it is, I will fix it ASAP

AnalysIR, 2015-06-30:
unsigned long  mask = 1UL << (32 - 1);

PureAbstract, 2015-07-01:
As @AnalysIR points out, this isn't a compiler bug - your code relies on undefined behaviour (see 5.8, [expr.shift] para 1) which says (in part):
> The behavior is undefined if the right operand is negative, or greater than or equal to the length in bits of the promoted left operand.

csBlueChip, 2015-07-01:
Thanks guys. "You live and learn". I fully understand what went wrong now. Patch submitted.

closed 2015-07-01 by csBlueChip