Arduino-IRremote/Arduino-IRremote #166: Random values with Ethernet Shield and RF receiver.

georgerappel, 2015-07-03:
I'm having an issue when i plug my IR Receiver to a board with an Ethernet Shield. I'm using the following libraries:

#include <IRremote.h>
#include <OSCMessage.h> // Used to send messages over UDP to an OSC server
#include <SPI.h>
#include <Ethernet.h>
#include <EthernetUdp.h>
#include <RCSwitch.h> // Used with the Radio Frequency transmitter.

My guess is that its a problem with the Ethernet Shield.

 - I uploaded the example code to another Uno board with no modules and it worked perfectly.
 - Then i upload the same code with no other includes and, although the led blinked, no text was shown on the serial. 
 - When using all the mentioned includes with the IR receiver code inside, instead of nothing i get lots of random codes.

AnalysIR, 2015-07-03:
You have a lot of libraries included...possible issues include - Timer conflict and/or pin conflicts. Do you have enough free SRAM left?

You could try changing IRremote to use Timer1 instead of Timer2, in IRremoteInt.h (pretty obvious) and note the tx pin on timer1 is 9 (instead of 3 for Timer2).

As a general approach, adding one library at a time might help highlight the issues.

Hope this helps!

georgerappel, 2015-07-03:
Thanks for the help, AnalysIR.

I tried in many different ways but i couldn't make it work.

Even when i upload the same code that works on the solo Uno board to the board with the ethernet shield it doesn't work. No other libraries included, using only the example provided.

AnalysIR, 2015-07-03:
Unfortunately, I don't have that setup available to test with...sorry.

Might be worth an extended search on the Arduino forum.

z3t0, 2015-07-27:
Just to clarify, simply connecting the Ethernet Shield causes the IR to stop working? Does the example sketch work without the Ethernet Shield on the same board?

georgerappel, 2015-07-28:
I discovered two things, and i'm sorry about it: there was probably something wrong with my port 11 on the Ethernet Shield, when i changed it to 10 it worked with the demo.
Then i uploaded IRrecvDemo to the board with the shield and all the other includes and now it's kinda working.
The code runs and when i press the button on a Remote controller once or twice it shows the button number, but somehow the board simply stops.

[image] (stuck after pressing once the button for no reason)

The board is with low memory because i'm using too many stuff, this might be the problem. I'll see what i can do. Also, the hosted webpage doesn't load with this code when i acess my ip as it should.

z3t0, 2015-07-28:
Can you share the full sketch, and if possible a diagram of the wiring? Thanks

georgerappel, 2015-07-29:
The full sketch is here: http://pastebin.com/GcRs6hWx
Some lines are commented because i was trying to fix the issue, and my code may be in portuguese some times... sorry about that.

The diagram is complicated. I have:
  - 1 LDR sensor
  - 1 RF emitter
  - 1 Relay module
  - Ethernet Shield
  - Trying to use an IR receiver.

z3t0, 2015-07-29:
@georgerappel Thanks for the sketch, I will take a look and get back to you

> Some lines are commented because i was trying to fix the issue, and my code may be in portuguese some times... sorry about that.

No problem.

The diagram is complicated.

> In that case maybe a few detailed photos

z3t0, 2016-02-22:
Do you still need help, sorry for not getting back earlier...

closed 2020-10-17 by ArminJo