Arduino-IRremote/Arduino-IRremote #167: IRrecvdump - bug & solution

AnalysIR, 2015-07-03:
HI Guys

There is an annoying set of bugs in this IRrecvdump example provided with the library, which I know causes a lot of hassle & confusion for beginners etc. 

1. The dump output always starts with a space which is generally not required and confusing to many users, particularly as sendRAW must always start with a mark.
2. The line `Serial.print(-(int)results->rawbuf[i]*USECPERTICK, DEC);` for printing out spaces as negative values has a bug in it (It handles large values incorrectly and prints them out as positive)

I have updated the example as follows, which provides a more readable output and fixes the 'bugs' mentioned above. I have also adjusted the formatted output to be more readable for users.
Essentially, just starts outputting at the second value in rawbuf, as the first is always a space & confusing. Also added in a blank line between signals for improved readability.

With this new code the 'dump' will always start and end with a mark, which is preferrable.

Code below is much improved & is tested.
````
/*
 * IRremote: IRrecvDump - dump details of IR codes with IRrecv
 * An IR detector/demodulator must be connected to the input RECV_PIN.
 * Version 0.1 July, 2009
 * Copyright 2009 Ken Shirriff
 * http://arcfn.com
 * JVC and Panasonic protocol added by Kristian Lauszus (Thanks to zenwheel and other people at the original blog post)
 */

#include <IRremote.h>

int RECV_PIN = 2;

IRrecv irrecv(RECV_PIN);

decode_results results;

void setup()
{
  Serial.begin(9600);
  irrecv.enableIRIn(); // Start the receiver
}

// Dumps out the decode_results structure.
// Call this after IRrecv::decode()
// void * to work around compiler issue
//void dump(void *v) {
//  decode_results *results = (decode_results *)v
void dump(decode_results *results) {
  int count = results->rawlen;
  if (results->decode_type == UNKNOWN) {
    Serial.print("Unknown encoding: ");
  } 
  else if (results->decode_type == NEC) {
    Serial.print("Decoded NEC: ");
  } 
  else if (results->decode_type == SONY) {
    Serial.print("Decoded SONY: ");
  } 
  else if (results->decode_type == RC5) {
    Serial.print("Decoded RC5: ");
  } 
  else if (results->decode_type == RC6) {
    Serial.print("Decoded RC6: ");
  }
  else if (results->decode_type == PANASONIC) {	
    Serial.print("Decoded PANASONIC - Address: ");
    Serial.print(results->panasonicAddress,HEX);
    Serial.print(" Value: ");
  }
  else if (results->decode_type == JVC) {
     Serial.print("Decoded JVC: ");
  }
  Serial.print(results->value, HEX);
  Serial.print(" (");
  Serial.print(results->bits, DEC);
  Serial.println(" bits)");
  Serial.print("Raw (");
  Serial.print(count, DEC);
  Serial.print("): ");

  for (int i = 1; i < count; i++) {
    if (i & 1) {
      Serial.print(results->rawbuf[i]*USECPERTICK, DEC);
    } 
    else {
      Serial.write('-');
      Serial.print((unsigned long) results->rawbuf[i]*USECPERTICK, DEC);
    }
    Serial.print(" ");
  }
  Serial.println();
}

void loop() {
  if (irrecv.decode(&results)) {
    Serial.println(results.value, HEX);
    dump(&results);
    irrecv.resume(); // Receive the next value
    Serial.println();
  }
}
````

z3t0, 2015-07-11:
@AnalysIR Is this one good?  https://github.com/shirriff/Arduino-IRremote/blob/experimental/ir_Template.cpp

AnalysIR, 2015-07-11:
@z3t0 
I had a quick look at the template code & it sounds like an interesting and good approach, to help users.

one thing.....
I don't think the name of the file is desciptive enough. maybe New_IR_Protocol_Template.cpp or similar.

Finally, I don't see the connection between this template file and this issue about IRrecvdump - maybe split them up.

z3t0, 2015-07-11:
>I had a quick look at the template code & it sounds like an interesting and good approach, to help users

@AnalysIR Yes, by the time I release 2.0 I will have written up an easy to follow guide in markdown.

> Finally, I don't see the connection between this template file and this issue about IRrecvdump - maybe split them up.

My bad... I linked the wrong file..
[here](https://github.com/shirriff/Arduino-IRremote/blob/experimental/examples/IRrecvDumpV2/IRrecvDumpV2.ino)

z3t0, 2015-07-11:
@AnalysIR Just a heads up, I believe code tags are 3 backticks (not four). Also you can add a syntax to use highlighting (I edited your OP so just check that to see what I mean)

AnalysIR, 2015-07-11:
@z3t0 
I had a  quick look at the code you linked & I would suggest the following for the same reason as points 1 & 2 in my first post here.

 ```
 // Print Raw data
  Serial.print("Timing[");
  Serial.print(results->rawlen, DEC);
  Serial.println("]: ");
  //Serial.print("     -"); //**** remove this and next line
  //Serial.println(results->rawbuf[0] * USECPERTICK, DEC);  ///****** no comma here, if you are leaving this in !!!!
  for (int i = 1;  i < results->rawlen;  i++) {
   // int  x = results->rawbuf[i] * USECPERTICK;//**** doesn't handle large values correctly (see my code above)
//replace with following line
   unsigned long x = (unsigned long) results->rawbuf[i] * USECPERTICK;
    if (!(i & 1)) {  // even
```
....sorry but untested.

Overall, I like this approach of providing more information for users.

z3t0, 2015-07-12:
@AnalysIR Thanks! I'll make the suggested changes and test.

AnalysIR, 2015-08-24:
Pull request #207 submitted to resolve this issue. Closing.

closed 2015-08-24 by AnalysIR