Arduino-IRremote/Arduino-IRremote #168: irrecv.decode not working after some time?

aalku, 2015-07-06:
This example code works for me with any arduino nano or bare atmega328p at 8Mhz but for both of them it freezes after some receptions and I have to reset the board/chip.

    void loop() {
      if (irrecv.decode(&results)) {
        Serial.println(results.value, HEX);
        irrecv.resume(); // Receive the next value
      }
      delay(100);
    }

Some times this happens fast and other times it is after 20 or more receptions.
I added a led blink on the loop and noticed the code is still running so it is not really frozen but it is the irrecv.decode that starts to return "false" forever.

Could it be that there is an unsuccessful decode and after that it is unable to decode a new reception? May it be caused by IR noise that drives the library to an unexpected state? May I reset it somehow as a workarround?

AnalysIR, 2015-07-06:
Might be a good idea to post the complete sketch.

...in the meantime try lowering the serial BAUD rate.

ReeceM, 2015-07-06:
AnalysIR, this question popped up on Arduin.SE 
http://arduino.stackexchange.com/questions/13189/arduino-irremote-stops-working
I think it is the same person

aalku, 2015-07-06:
Yes I am. I started using PWM and serial and then started removing things until I left only the loop in the example and still freezes sometimes.

z3t0, 2015-07-07:
Please test the new experimental branch as it is much improved!!! Will be releasing it within a few days. Credits go to @csBlueChip

aalku, 2015-07-07:
Maybe it has something to do with power load activated after reception. On my last test there is an arduino nano driving a relay, leds, and other stuff and a bareduino (powered by the ardunino nano) receiving IR and printing the read value to serial port so it is read by the arduino nano and do something. I noticed it usually hangs up after powering on the relay and the leds and not powering down. When they are off it usually does not hang up. But the power drain is not a permanent problem. If I reset the bareduino it works again no matter leds and relay are on or off. It seems that a power fluctuation affect the sensor or the library somehow. The arduino nano keeps working without problem.

csBlueChip, 2015-07-07:
If you suspect it's current draw, check this: http://pighixxx.com/nanopdf.pdf

AnalysIR, 2015-07-07:
Generally it is not a good idea to drive a relay directly from the arduino. You should use a transistor driving circuit in between & don't forget the protection diode (research this if unsure).

Driving a relay directly from any MCU will quickly damage the MCU permanently.

You can of course test all of this by replacing the relay with an LED&resistor.

aalku, 2015-07-07:
I plan to power leds from vin with a transistor activated by PWM and the relay is adapted to arduino with transistor or whatever. It has pins IN, VCC, GND, so I think it is OK.
I don't think the power drain is too high, the arduino program works fine, it only affects the infrared behaviour I think. Maybe there is a small power fluctiation when everything turns on.
Might it make the IR sensor emit somehting that makes the library fail?

aalku, 2015-07-07:
I'm gonna try this ASAP on the bareduino:

    #include <IRremote2.h>

    int RECV_PIN = 7;
    int RST_PIN = 3;
    int LED_PIN = 13;

    IRrecv irrecv(RECV_PIN);

    decode_results results;

    void setup()
    {
      Serial.begin(9600);
      irrecv.enableIRIn(); // Start the receiver
      pinMode(RST_PIN, INPUT_PULLUP);
      pinMode(LED_PIN, OUTPUT);
    }

    void loop() {
      if (digitalRead(RST_PIN) == LOW) {
        Serial.print("FIX");
        irrecv.resume(); // Fix
        while (digitalRead(RST_PIN) == LOW) {
          digitalWrite(LED_PIN, HIGH);
          delay(100);
          digitalWrite(LED_PIN, LOW);
          delay(100);
        }
      }
      if (irrecv.decode(&results)) {
        Serial.print("[");
        Serial.print(results.value, HEX);
        Serial.print("]");
        irrecv.resume(); // Receive the next value
        digitalWrite(LED_PIN, HIGH);
        delay(200);
        digitalWrite(LED_PIN, LOW);
      }
      delay(100);
    }

led and some kind of reset to see what happens when hang up

AnalysIR, 2015-07-07:
You need to drive the relay via transistor circuit with additional diode protection. 

Otherwise, broken CMU.

If you do some research on the topic, you will see it is not just about power-drain.

It seems this is an electronics design issue & not directly related to the  IRremote library!

aalku, 2015-07-07:
I thought the library was the only thing not working (under certain electronic circunstances, ok). I'll shut up until I got more info. Thank you all very much.

aalku, 2015-07-07:
It seems the problem was noise on VCC pin of the IR detector. It made decoding quite hard. Arduino tutorials suggest to connect just the pins to GND, VCC, Arduino pin but the IR detectors data sheets suggest to connect VCC through a resistor and to a capacitor with the other end to GND. I hope you like my paint schematic ;-)

[image]

It decodes much better now. I used 100 Ohm resistor and 6.8uF capacitor.

This question put me on the track http://electronics.stackexchange.com/questions/68310/arduino-ir-practice-vs-application-circuit-on-1838-datasheets

Thank you all!

aalku, 2015-07-12:
This is happening to me again. With the experimental branch today version.
Code: http://pastebin.com/L0YN3Psr
VCC is stabilized.
There is a RGB led driven with PWM. It's ok when the LED is off or it is white buy if I select random color mixes after some time it stop responding to IR. It's not decoding is wrong, it receives nothing.

If I reset IR works again.
If I send a command (vía USB/serial) to make the led white IR works again.
If I send a command (vía USB/serial) to make the led off IR works again.
If I send a command (vía USB/serial) to make the led pure RED, GREEN or BLUE then IR works again.
If I send a command (vía USB/serial) to change colour to a non pure one sometimes IR works for a moment but it usually hungs up again quite fast.

I have read analogWrite(0) and analogWrite(255) are equivalent to digitalWrite. Maybe that's why with pure colors everything's fine.

May it exploit something with PWM timers or something?
Is there any way I can debug the reception or decoding to know what happens when IR stops working?

Thank you.

z3t0, 2015-07-23:
Not sure if this will change anything but try using pin 11 instead of 10

aalku, 2015-07-24:
This is really weird for me. If I change PG = 10 for PG = 11 (pwm pin for green color) the ir decoding... changes. It recognize different codes for the same remote buttons.

z3t0, 2015-07-24:
Can you post the output?

aalku, 2015-07-24:
Sure, I put the dump function in the example.

Remote buttons I call 4, 5, 6 with green led on pin 10 and irrecv on pin 7:

	Decoded NEC: FF906F (32 bits)
	Raw (68): -7900 8750 -4450 550 -550 550 -550 550 -550 550 -550 550 -550 550 -550 550 -550 550 -550 550 -1700 500 -1700 500 -1700 500 -1700 550 -1650 500 -1700 550 -1700 500 -1700 500 -1700 550 -550 500 -600 550 -1650 550 -600 500 -550 550 -600 500 -600 500 -550 550 -1700 500 -1700 500 -600 500 -1700 550 -1650 550 -1700 500 -1700 500 

	Decoded NEC: FF10EF (32 bits)
	Raw (68): 32334 8950 -4250 750 -350 750 -350 700 -400 750 -350 750 -350 700 -400 700 -400 700 -450 600 -1600 600 -1600 600 -1650 550 -1650 600 -1600 600 -1650 500 -1700 550 -1650 550 -550 550 -550 550 -550 550 -1700 500 -600 500 -550 550 -600 500 -600 500 -1700 500 -1700 550 -1650 550 -550 550 -1700 500 -1700 500 -1700 550 -1650 550 

	Decoded NEC: FF50AF (32 bits)
	Raw (68): -19996 8800 -4450 550 -600 500 -550 550 -600 500 -600 500 -600 550 -550 500 -600 550 -550 550 -1650 550 -1700 500 -1700 550 -1650 550 -1700 500 -1700 550 -1650 550 -1700 500 -600 500 -1700 500 -600 550 -1650 550 -550 550 -550 550 -600 500 -600 500 -1700 500 -600 550 -1650 550 -550 550 -1700 500 -1700 550 -1650 550 -1700 500 

Remote buttons I call 4, 5, 6 with green led on pin 11 and irrecv on pin 7:

	Unknown encoding: E5CFBD7F (32 bits)
	Raw (68): 25408 9850 -4600 950 -250 950 -250 950 -250 950 -250 950 -300 900 -300 850 -350 850 -350 850 -1600 850 -1550 800 -1650 800 -1600 800 -1650 800 -1600 800 -1650 800 -1650 700 -1750 650 -550 650 -550 650 -1800 600 -600 600 -600 650 -600 550 -650 550 -650 600 -1800 600 -1850 550 -650 600 -1850 550 -1850 600 -1850 600 -1800 600 

	Unknown encoding: 8C22657B (32 bits)
	Raw (68): -27052 9600 -4900 600 -650 550 -650 600 -600 550 -650 600 -600 600 -600 600 -650 550 -650 600 -1800 600 -1850 550 -1850 600 -1850 600 -1800 600 -1850 600 -1800 600 -1850 550 -650 600 -600 600 -650 550 -1850 600 -600 600 -600 600 -600 600 -650 550 -1850 600 -1850 550 -1850 600 -650 550 -1850 600 -1800 600 -1850 600 -1850 550 

	Unknown encoding: 2A89195F (32 bits)
	Raw (68): 10934 9650 -4900 550 -650 600 -600 600 -650 550 -650 550 -650 600 -600 600 -650 550 -600 600 -1850 600 -1850 550 -1850 600 -1850 600 -1850 550 -1850 600 -1850 550 -1850 600 -650 550 -1850 600 -600 600 -1850 550 -650 600 -600 600 -600 600 -650 550 -1850 600 -650 550 -1850 550 -650 600 -1850 550 -1850 600 -1850 600 -1800 600 

Belive it or not they are the same three buttons in the same order.

When green on pin 12 if I turn off the LED vía serial command the IR decoding keeps wrong.

aalku, 2015-07-24:
After some tries using pin 11 the arduino got hunged up completely. It didn't even respond to serial port commands.

aalku, 2015-07-24:
If it has something to do with the timers I might try software pwm but if it's an electrical problem it would not help.

AnalysIR, 2015-07-24:
What model # of IR receiver are you using? 
Do you have a link to the data sheet or where it was purchased?

Looking at the signals you posted it 'could' be the receiver.

AnalysIR, 2015-07-24:
It could also be that you are pulsing the RGB LEDs and LEDs can emit IR as well. So shielding the IR receiver from the light of the RGB LEDs might help.

aalku, 2015-07-25:
This receiver. http://es.aliexpress.com/item/10pcs-Lot-Universal-IR-Infrared-Receiver-Sensor-Module-1838-TL1838-VS1838B-38Khz-Diode/32266398399.html
The light from the leds can't reach the receiver. I electrically isolated the water pump from the receiver, or tried to, and now it does not hung up. There are very few decoding errors (less than 10%, maybe less than 5%). I'm currently using hardware PWM. Green led on pin 10. IR Receiver is connected like the image.
[image]
I think the library is fine but I wish to know what happened to IR stuff when ir stopped working (because of the pump or any other interference).
To isolate the pump I used external power, not through the arduino usb. VCC_IN->PUMP+ARDUINO_IN. Capacitor between ARDUINO_V5 and GND.
I shared the full sketch: https://github.com/aalku/rgb_water_fountain/
Thank you very much guys.

closed 2015-07-25 by aalku

AnalysIR, 2015-07-25:
I believe your main issue is this IR receiver. Unfortunately, this VS1838B is the cheapest IR receiver available & also the worst. See our blog post for more details/insight:
http://www.analysir.com/blog/2014/12/08/infrared-receiver-showdown-tsop34438-vs-vs1838b-winner-revealed/

naurisgrinbergs0, 2021-08-13:
> It seems the problem was noise on VCC pin of the IR detector. It made decoding quite hard. Arduino tutorials suggest to connect just the pins to GND, VCC, Arduino pin but the IR detectors data sheets suggest to connect VCC through a resistor and to a capacitor with the other end to GND. I hope you like my paint schematic ;-)
> 
> [image]
> 
> It decodes much better now when the PWM leds are on. I think that was causing noise. I used 100 Ohm resistor and 6.8uF capacitor.
> 
> This question put me on the track http://electronics.stackexchange.com/questions/68310/arduino-ir-practice-vs-application-circuit-on-1838-datasheets
> 
> I hope that's all.
> 
> Thank you!

========================================= THIS HELPED ====================