Arduino-IRremote/Arduino-IRremote #171: Teensy 3.0/3.1 Support Broken

z3t0, 2015-07-11:
Somewhere along the rewrite we seem to have broken Teensy 3.0/3.1 support.
The version bundled with the teensyduino installer is still functional.

Log:
```
In file included from C:\Users\Rafi\Documents\Arduino\libraries\IRremote\IRremote.h:24:0,
from C:\Users\Rafi\Documents\Arduino\libraries\IRremote\irISR.cpp:3:
C:\Users\Rafi\Documents\Arduino\libraries\IRremote\irISR.cpp: In function 'void cmt_isr()':
C:\Users\Rafi\Documents\Arduino\libraries\IRremote\IRremoteInt.h:461:10: warning: unused variable 'tmp' [-Wunused-variable]
uint8_t tmp = CMT_MSC; \
^
C:\Users\Rafi\Documents\Arduino\libraries\IRremote\irISR.cpp:20:2: note: in expansion of macro 'TIMER_RESET'
TIMER_RESET;
^
In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/core_pins.h:34:0,
from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/wiring.h:33,
from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/WProgram.h:15,
from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/Arduino.h:1,
from C:\Users\Rafi\Documents\Arduino\libraries\IRremote\IRremoteInt.h:24,
from C:\Users\Rafi\Documents\Arduino\libraries\IRremote\IRremote.h:24,
from C:\Users\Rafi\Documents\Arduino\libraries\IRremote\irRecv.cpp:1:
C:\Users\Rafi\Documents\Arduino\libraries\IRremote\irRecv.cpp: In member function 'void IRrecv::enableIRIn()':
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/kinetis.h:1987:52: error: expression cannot be used as a function
#define CMT_CMD3  (*(volatile uint8_t  *)0x40062008) // CMT Modulator Data Register Space High
^
C:\Users\Rafi\Documents\Arduino\libraries\IRremote\IRremoteInt.h:512:2: note: in expansion of macro 'CMT_CMD3'
CMT_CMD3   = 0;               \
^
C:\Users\Rafi\Documents\Arduino\libraries\IRremote\irRecv.cpp:110:2: note: in expansion of macro 'TIMER_CONFIG_NORMAL'
TIMER_CONFIG_NORMAL();
^
In file included from C:\Users\Rafi\Documents\Arduino\libraries\IRremote\IRremote.h:24:0,
from C:\Users\Rafi\Documents\Arduino\libraries\IRremote\irRecv.cpp:1:
C:\Users\Rafi\Documents\Arduino\libraries\IRremote\IRremoteInt.h:461:10: warning: unused variable 'tmp' [-Wunused-variable]
uint8_t tmp = CMT_MSC; \
^
C:\Users\Rafi\Documents\Arduino\libraries\IRremote\irRecv.cpp:115:2: note: in expansion of macro 'TIMER_RESET'
TIMER_RESET;
^
Error compiling.
```

joshuajnoble, 2015-07-14:
I think #171 is the fix for this, or at least, after making the same change the library works with the Teensy 3.1. We might also want to add the Teensy LC setting

    // Teensy-LC
    #elif defined(__MKL26Z64__)
        #define IR_USE_TIMER_TPM1 // tx = pin 16

z3t0, 2015-07-15:
I have already added Teensy LC receiving support, is that for sending?

Also this issue is regarding the experimental branch... The stable release works with both 3 and 3.1

Will take a look at #171

z3t0, 2015-07-15:
Oh... This is #171

joshuajnoble, 2015-07-15:
Yeah, I meant to put [this](https://github.com/shirriff/Arduino-IRremote/commit/2e4538bac8d48eb395be7f6c055c3d089c25aaee) not #171

z3t0, 2015-07-15:
Sorry I should have been more clear. It is that commit which allows the board to be detected and then results in the error log above.

joshuajnoble, 2015-07-15:
Interesting. I made the same change and see the following warnings on compile:

````

Build options changed, rebuilding all
In file included from /Users/jnoble/code/arduino/libraries/Arduino-IRremote-master/IRremote.cpp:22:0:
/Users/jnoble/code/arduino/libraries/Arduino-IRremote-master/IRremote.cpp: In member function 'void IRrecv::enableIRIn()':
/Users/jnoble/code/arduino/libraries/Arduino-IRremote-master/IRremoteInt.h:470:10: warning: unused variable 'tmp' [-Wunused-variable]
  uint8_t tmp = CMT_MSC;  \
          ^
/Users/jnoble/code/arduino/libraries/Arduino-IRremote-master/IRremote.cpp:351:3: note: in expansion of macro 'TIMER_RESET'
   TIMER_RESET;
   ^
/Users/jnoble/code/arduino/libraries/Arduino-IRremote-master/IRremote.cpp: In function 'void cmt_isr()':
/Users/jnoble/code/arduino/libraries/Arduino-IRremote-master/IRremoteInt.h:470:10: warning: unused variable 'tmp' [-Wunused-variable]
  uint8_t tmp = CMT_MSC;  \
          ^
/Users/jnoble/code/arduino/libraries/Arduino-IRremote-master/IRremote.cpp:380:3: note: in expansion of macro 'TIMER_RESET'
   TIMER_RESET;
   ^

Sketch uses 18,660 bytes (7%) of program storage space. Maximum is 262,144 bytes.
Global variables use 5,244 bytes (8%) of dynamic memory, leaving 60,292 bytes for local variables. Maximum is 65,536 bytes.
````

PaulStoffregen, 2015-07-15:
Oh, that's the newer versions of the compiler being a bit silly about excessive warnings.

This same warning has come up with other code that needed to read a hardware register, so the hardware would clear some internal state, but then had no need to actually do anything with the data.

You can suppress the warning with this:

      uint8_t tmp  __attribute__((unused)) = CMT_MSC;  \

z3t0, 2015-07-15:
@PaulStoffregen Thanks! Will try that out... I was really confused at the error.

z3t0, 2015-07-15:
@PaulStoffregen Without success...

Before:
```

In file included from C:\Users\Rafi\Documents\Arduino\libraries\IRremote\IRremote.h:24:0,
from C:\Users\Rafi\Documents\Arduino\libraries\IRremote\irISR.cpp:3:
C:\Users\Rafi\Documents\Arduino\libraries\IRremote\irISR.cpp: In function 'void cmt_isr()':
C:\Users\Rafi\Documents\Arduino\libraries\IRremote\IRremoteInt.h:461:10: warning: unused variable 'tmp' [-Wunused-variable]
uint8_t tmp = CMT_MSC; \
^
C:\Users\Rafi\Documents\Arduino\libraries\IRremote\irISR.cpp:20:2: note: in expansion of macro 'TIMER_RESET'
TIMER_RESET;
^
In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/core_pins.h:34:0,
from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/wiring.h:33,
from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/WProgram.h:15,
from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/Arduino.h:1,
from C:\Users\Rafi\Documents\Arduino\libraries\IRremote\IRremoteInt.h:24,
from C:\Users\Rafi\Documents\Arduino\libraries\IRremote\IRremote.h:24,
from C:\Users\Rafi\Documents\Arduino\libraries\IRremote\irRecv.cpp:1:
C:\Users\Rafi\Documents\Arduino\libraries\IRremote\irRecv.cpp: In member function 'void IRrecv::enableIRIn()':
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/kinetis.h:1987:52: error: expression cannot be used as a function
#define CMT_CMD3  (*(volatile uint8_t  *)0x40062008) // CMT Modulator Data Register Space High
^
C:\Users\Rafi\Documents\Arduino\libraries\IRremote\IRremoteInt.h:512:2: note: in expansion of macro 'CMT_CMD3'
CMT_CMD3   = 0;               \
^
C:\Users\Rafi\Documents\Arduino\libraries\IRremote\irRecv.cpp:110:2: note: in expansion of macro 'TIMER_CONFIG_NORMAL'
TIMER_CONFIG_NORMAL();
^
In file included from C:\Users\Rafi\Documents\Arduino\libraries\IRremote\IRremote.h:24:0,
from C:\Users\Rafi\Documents\Arduino\libraries\IRremote\irRecv.cpp:1:
C:\Users\Rafi\Documents\Arduino\libraries\IRremote\IRremoteInt.h:461:10: warning: unused variable 'tmp' [-Wunused-variable]
uint8_t tmp = CMT_MSC; \
^
C:\Users\Rafi\Documents\Arduino\libraries\IRremote\irRecv.cpp:115:2: note: in expansion of macro 'TIMER_RESET'
TIMER_RESET;
^
Error compiling.
```

After:
```

In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/core_pins.h:34:0,
from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/wiring.h:33,
from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/WProgram.h:15,
from C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/Arduino.h:1,
from C:\Users\Rafi\Documents\Arduino\libraries\IRremote\IRremoteInt.h:24,
from C:\Users\Rafi\Documents\Arduino\libraries\IRremote\IRremote.h:24,
from C:\Users\Rafi\Documents\Arduino\libraries\IRremote\irRecv.cpp:1:
C:\Users\Rafi\Documents\Arduino\libraries\IRremote\irRecv.cpp: In member function 'void IRrecv::enableIRIn()':
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/kinetis.h:1987:52: error: expression cannot be used as a function
#define CMT_CMD3  (*(volatile uint8_t  *)0x40062008) // CMT Modulator Data Register Space High
^
C:\Users\Rafi\Documents\Arduino\libraries\IRremote\IRremoteInt.h:512:2: note: in expansion of macro 'CMT_CMD3'
CMT_CMD3   = 0;               \
^
C:\Users\Rafi\Documents\Arduino\libraries\IRremote\irRecv.cpp:110:2: note: in expansion of macro 'TIMER_CONFIG_NORMAL'
TIMER_CONFIG_NORMAL();
^
Error compiling.
```

bisnad, 2016-02-03:
Hi, I'm currently running into a problem when trying to use this library in combination with a Teensy 3.2 that looks suspiciously similar to what has been discussed in this thread. Therefore' Im wondering if this issue is indeed still open for the current version of the library or if the problem has a different cause. 

The error log is as follows:

Arduino: 1.6.5 (Mac OS X), TD: 1.27, Board: "Teensy 3.2 / 3.1, Serial, 96 MHz optimized (overclock), US English"

In file included from /CustomApplications/Engineering/Electronics/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3/core_pins.h:34:0,
from /CustomApplications/Engineering/Electronics/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3/wiring.h:33,
from /CustomApplications/Engineering/Electronics/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3/WProgram.h:15,
from /CustomApplications/Engineering/Electronics/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3/Arduino.h:1,
from /Users/dbisig/Documents/Arduino/libraries/IRremote/IRremoteInt.h:24,
from /Users/dbisig/Documents/Arduino/libraries/IRremote/IRremote.h:24,
from /Users/dbisig/Documents/Arduino/libraries/IRremote/irRecv.cpp:1:
/Users/dbisig/Documents/Arduino/libraries/IRremote/irRecv.cpp: In member function 'void IRrecv::enableIRIn()':
/CustomApplications/Engineering/Electronics/Arduino.app/Contents/Java/hardware/teensy/avr/cores/teensy3/kinetis.h:3115:52: error: expression cannot be used as a function
#define CMT_CMD3  (*(volatile uint8_t  *)0x40062008) // CMT Modulator Data Register Space High
^
/Users/dbisig/Documents/Arduino/libraries/IRremote/IRremoteInt.h:524:2: note: in expansion of macro 'CMT_CMD3'
CMT_CMD3   = 0;               \
^
/Users/dbisig/Documents/Arduino/libraries/IRremote/irRecv.cpp:120:2: note: in expansion of macro 'TIMER_CONFIG_NORMAL'
TIMER_CONFIG_NORMAL();
^
In file included from /Users/dbisig/Documents/Arduino/libraries/IRremote/IRremote.h:24:0,
from /Users/dbisig/Documents/Arduino/libraries/IRremote/irRecv.cpp:1:
/Users/dbisig/Documents/Arduino/libraries/IRremote/IRremoteInt.h:473:10: warning: unused variable 'tmp' [-Wunused-variable]
uint8_t tmp = CMT_MSC; \
^
/Users/dbisig/Documents/Arduino/libraries/IRremote/irRecv.cpp:125:2: note: in expansion of macro 'TIMER_RESET'
TIMER_RESET;
^
Multiple libraries were found for "IRremote.h"
 Used: /Users/dbisig/Documents/Arduino/libraries/IRremote
 Not used: /CustomApplications/Engineering/Electronics/Arduino.app/Contents/Java/libraries/RobotIRremote
 Not used: /CustomApplications/Engineering/Electronics/Arduino.app/Contents/Java/hardware/teensy/avr/libraries/IRremote
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

z3t0, 2016-02-07:
Have you tried using the Teensy version of the library?

PaulStoffregen, 2016-08-01:
I submitted a fix for this problem.  https://github.com/z3t0/Arduino-IRremote/pull/343

PaulStoffregen, 2016-08-01:
Fixed

closed 2016-08-01 by PaulStoffregen