Arduino-IRremote/Arduino-IRremote #173: IRsend "breaking" IRrecv

rumpelheinz, 2015-07-12:
Hello, 
I am working on a project where two arduinos have to communicate back and forth, using IR. In order to get to know this library, I tried the following code, which is supposed to let one arduino send x, and then the other one receives it and then sends back x+1, going on forever.
---------------------------------------------------------------------------------------------------
#include <IRremote.h>
#include <IRremoteInt.h>
IRsend irsend;
int RECV_PIN = 11;
IRrecv irrecv(RECV_PIN);
decode_results results;
int i=0;
void setup()
{
  Serial.begin(9600);
  irsend.sendSony(1, 12); // Sony TV power code
  irrecv.enableIRIn(); // Start the receiver
}

void loop() {
  delay(40);
  if (irrecv.decode(&results)) {
    Serial.println(results.value);
    irsend.sendSony(results.value+1,12);
    irrecv.resume(); // Receive the next value
  }
}
---------------------------------------------------------------------------------------------------
 The code does not work, and through some debugging I've come to the conclusion, that as soon as  irsend.sendSony() is used, the irrecv object stops working correctly. Is there a reason for this, and is there a proper way to do it?

I am using the "latest release" ( 1 ) branch.

Thanks in advance, Tobias Sander

AnalysIR, 2015-07-12:
Some comments:
1. Leave a longer gap between sends. many IR receivers will turn off if continuous signals are received. Try a 100ms gap & if that doesnt work up it towards1sec +, until it works.
2. I presume the other Arduino doesnt have the same sketch or it can never work. (remember that only one can send at the same time or there will be interference & failure)
3. once you get to the max of 12 bits, it will wrap round to 0.

z3t0, 2015-07-13:
@rumpelheinz By latest release do you mean the one from master or the one in the releases?

The correct one is: https://github.com/shirriff/Arduino-IRremote/releases/tag/MAJOR

rumpelheinz, 2015-07-13:
I have gotten it to work now , using the following code.

#include <IRremote.h>
#include <IRremoteInt.h>

IRsend irsend;
int RECV_PIN = 11;
IRrecv irrecv(RECV_PIN);
decode_results results;
void setup()
{
  Serial.begin(9600);
  irsend.sendSony(1, 12); // Sony TV power code
  delay(50);
  irrecv.enableIRIn(); // Start the receiver
}

void loop() {
  if (irrecv.decode(&results)) {
    Serial.println(results.value);
    delay(200);
    irsend.sendSony(results.value+1,12);
   // delay(100);
    irrecv.enableIRIn(); // Start the receiver
    delay(300);
}

The timings weren't the problem, as I had already played around with different settings. What did it was that I now use  irrecv.enableIRIn() after every irsend.sendSony(), insetead of irrecv.resume(). 

To me this seems strange, as I thought you should only use  enableIRIn() once...

z3t0, 2015-07-15:
I'll write up an Api reference and we can go on to rename things from there if we need to.

closed 2015-07-24 by z3t0