Arduino-IRremote/Arduino-IRremote #174: Arduino Micro issue

Longrange, 2015-07-13:
When I hook up a LED to pin 3 on the arduino micro from adafruit I am unable to send any codes, If i just use the standard blink example it works through that pin. Im not sure if there's another pin I should be using or if there is just no support for the micro yet. thanks for the help

AnalysIR, 2015-07-13:
The adafruit micro is more like a Leonardo, so try Pin 9 instead.

Longrange, 2015-07-14:
I tried pin 9 and had no output, so I went down the pins 13 to 1 and got no outputs from any of them. im going to try just blinking the led to make sure I don't have faulty hardware, i'll post back.

Longrange, 2015-07-15:
Okay i figured it out. the pin i needed to use is pin 13 which is less than ideal considering that if I am over powering my send led I need to make sure it is disconnected before I upload code. I missed this because I was testing using the IRRecord example program and that uses pin 13 for indication. Thanks for the help.

z3t0, 2015-07-15:
Definitely should not be pin 13...

This is the part for the Micro
```c
// Teensy 2.0
#elif defined(__AVR_ATmega32U4__)
  //#define IR_USE_TIMER1   // tx = pin 14
  //#define IR_USE_TIMER3   // tx = pin 9
  #define IR_USE_TIMER4_HS  // tx = pin 10
```

Comment the line with pin 10 and uncomment the line with pin 9. The file is IRremoteInt.h

z3t0, 2015-07-27:
@Longrange did the fix above do anything?

alexbloggt, 2015-08-07:
It is indeed Pin 13 on Micro, as it says in IR_USE_TIMER4_HS:
```c
#elif defined(__AVR_ATmega32U4__)
#	define TIMER_PWM_PIN  13             // Leonardo
```
Don´t know why comment says `// tx = pin 10` here - should rather be 13 in my opinion.
However by using a different timer (Timer1 or Timer3) pins #9 (using T1) or #5 (using T3 with replacing `error "Please add OC3A pin number here\n"` by `#define TIMER_PWM_PIN        5` in Timer3 section in IRremoteInt.h) can be used as well.

z3t0, 2015-08-10:
@alexbloggt I am not certain but I am assuming it is labelled as pin 10 because the Teensy 2.0 might break it out on that. 

Closing this issue as the problem @Longrange was facing either no longer seems to be apparent or relevant

@alexbloggt Feel free to continue the conversation here or another issue.

closed 2015-08-10 by z3t0