Arduino-IRremote/Arduino-IRremote #176: How can I use it on attiny84?

emilburzo, 2015-07-22:
According to #97  there is support for it, but I can't seem to get it to work.

``arduino/libraries/IRremote/irSend.cpp: In member function 'void IRsend::mark(int)':
arduino/libraries/IRremote/IRremoteInt.h:210:30: error: 'TCCR2A' was not declared in this scope
 #define TIMER_ENABLE_PWM    (TCCR2A |= _BV(COM2B1))``

Looking through the code, I can't see any references to *tiny*.

What am I missing?

AnalysIR, 2015-07-22:
There is no Timer2 on the ATTiny84....so change the internal library definitions to use another Timer  e.g. Timer1.

z3t0, 2015-07-24:
try the latest experimental release: https://github.com/shirriff/Arduino-IRremote/tree/experimental

emilburzo, 2015-07-24:
The code compiles fine now, but I'm not sure what pin should I use?

AnalysIR, 2015-07-24:
try ...you just need to make sure "__AVR_ATtinyX4__" is defined as true or just make your own new ones for __AVR_ATtiny84__

````
#elif defined( __AVR_ATtiny84__ )
  #define IR_USE_TIMER1   // tx = pin 6  (Means Arduino Pin 6 or PA6 or Physical IC pin 7)
.......

#elif defined(__AVR_ATtiny84__)
#   define TIMER_PWM_PIN  6             // ATtiny84

````
and use PA6 or physical pin 7 as the tx pin  
The Arduino pin 6 is PA6(physical pin 7)

Make sure these definitions are placed before the one for __AVR_ATtinyX4__

z3t0, 2015-07-24:
Updated experimental to include the fix mentioned above.

emilburzo, 2015-07-26:
I've just tried it and it works!

Thank you very much.

AnalysIR, 2015-07-26:
Could you clarify, for future reference, what actually worked?

emilburzo, 2015-07-26:
I've downloaded the experimental branch (with the latest changes from z3t0 included) and ran my project (which sends some IR commands) on an attiny84 (1MHz, internal oscillator).

Everything worked as expected.

z3t0, 2015-07-26:
Happy it worked for you

closed 2015-07-26 by z3t0