Arduino-IRremote/Arduino-IRremote #178: Disabling protocols not working

gaetanwalter, 2015-07-27:
When you disable protocols using
#define DECODE_RC5           0

the sketch can't compile because in irRecv:

#ifdef DECODE_RC5
	DBG_PRINTLN("Attempting RC5 decode");
	if (decodeRC5(results))  return true ;
#endif

should be:

#if DECODE_RC5
	DBG_PRINTLN("Attempting RC5 decode");
	if (decodeRC5(results))  return true ;
#endif

And ir_RC5_RC6.cpp is missing a:

#if (DECODE_RC5 || DECODE_RC6)

z3t0, 2015-07-28:
Thanks, will take care of this and commit upstream

z3t0, 2015-08-10:
@gaetanwalter Do you know if this is a problem for other protocols as well? Thanks

gaetanwalter, 2015-08-10:
Yeah, all ifdef schould be if

DECODE_XXX is defined (as 0 or 1) so ifdef will always return true.

z3t0, 2015-08-10:
@gaetanwalter Great, will fix this tomorrow.

z3t0, 2015-08-10:
@gaetanwalter Committed. 5c920bd31dead97ce9acfc6429b9c3e0162faf47

closed 2015-08-10 by z3t0