Arduino-IRremote/Arduino-IRremote #179: Recommended IR Modules

z3t0, 2015-07-28:
@AnalysIR

Hi All,

I am working a new project and am looking for a good pair of SMD IR Receiver and Emitter modules. I want to solder these onto a PCB, which will also contain an AVR chip and RS232 Chip.

Any recommendations for which IR receiver or emitter? 

Thanks!

AnalysIR, 2015-07-28:
....how long is a piece of string...  :)

it really depends on the application and PCB design/layout/enclosure/orientation/current/voltage..not to mention the application & carrier frequency etc.

What I would do is go to digikey and do a search based on your requirements and then study the data sheet to see that they match requirements.

For general purpose TH parts, I usually default to TSAL6x00 for emitters and TSOP34438 for receivers (obviously change the carrier frequency as required). So if your requirement is general, just restrict your search on digikey to similar specs & SMD.(for SMD dont forget orientation)

Most important: I always stick with Vishay.

SMD parts are usually more expensive than the TH equivalents, particularly in low quantities. In many cases SMD versions are not readily available in low quantities.

z3t0, 2015-07-28:
> it really depends on the application and PCB design/layout/enclosure/orientation/current/voltage..not to mention the application & carrier frequency etc.

I was looking at some new open source projects and stumbled on [this](https://www.crowdsupply.com/longan-lab/irduino). Unfortunately it never reached the goal, however I thought I might try and make one for fun. In terms of carrier frequency I'm just going to have to settle for 38KHz. 

> For general purpose TH parts, I usually default to TSAL6x00 for emitters and TSOP34438 for receivers (obviously change the carrier frequency as required). So if your requirement is general, just restrict your search on digikey to similar specs & SMD.(for SMD dont forget orientation)

In that case I guess I might just use TH parts, which would mean an encasing with the IR receiver being flush with the top. I would also need a transmitter

How about the [TSOP38238](https://www.sparkfun.com/products/10266)? The main idea for the application is a programmable Universal remote control which interfaces to its pairing software via RS232. I haven't figured out the specifics of how that will work just yet, I wanted to decide on a parts list so I could start prototyping.

I'll draw up some CAD drawings soon and post them here, one of my main concerns is the range in relation to the enclosure. Basically think of a USB with 2 parts flush with the enclosing (there will be cutouts for them) the two parts being emitter and receiver. The reason for the receiver is so that you can record certain button presses and save them using the software (onto the computer).

z3t0, 2015-07-28:
One of the main differences will be that although my version will allow you to control your computer, the  main usage will be to use it as a universal remote for other appliances.

AnalysIR, 2015-07-28:
The TSOP38238 should be fine but may have issues on some difficult signals(protocols). It is classified as legacy by Vishay. TSOP38438 is the more modern equivalent.

The benefit of TH is that you have lots of room to position within the enclosure. With SMD, enclosure & receiver positioning must be well designed & spot on. Range should just depend on the power of the remote. You can also hotglue the receiver / emitter to the enclosure and wire them back to the PCB. there are 5mm panel mounts available for LED that work with IR LEDs. Havent seen any affordable panel mounts for TH receivers recently.

You can always bend the TH components to suit orientation.

If you use a leonardo (clone), you get a HID/keyboard interface for free(?), which is what those guys were doing. So no need for special apps.

I would usually prototype on a breadboard first & afterwards think about PCB layout, SMD/TH & enclosures.

FYI, you can always get a media centre remote & receiver, which should do most of this PC control stuff (cheap on ebay/aliexpress)

PS: we will be launching a range of interesting IR boards/devices soon, probably via crowdfunding....stay tuned.

z3t0, 2015-07-28:
Looking forward to the IR boards, do you think they will be open source? Also are they currently in development?

AnalysIR, 2015-07-29:
Development is completed.

Some (but not all) will be open.

z3t0, 2015-07-31:
Cool. Can't wait to see the result.

Thanks for the replies. I'll close the issue and report my experiences here once I get something running.

closed 2015-07-31 by z3t0