Arduino-IRremote/Arduino-IRremote #180: IR Multiple receiver timing issue

domonation, 2015-07-29:
Has anyone else had an issue with the timers? I am currently implementing 4 receivers but when I attempt to use the millis() function it gives me a constant value. I am using an Arduino Mega 2560

z3t0, 2015-07-30:
@domonation Can you please post the full sketch into a gist and share it here. Also, are you doing anything else with the Arduino while trying to get the 4 receivers to work? I mean has the problem been isolated to the library/receiver or are there additional shields which are also attached? Thanks

z3t0, 2015-07-30:
@domonation Do you know which timer is being used for the different receivers? Which pins? (If it is realistic could you make a rough schematic or post a photo?)

z3t0, 2015-07-31:
Also I'm assuming it works fine when not using the IR library?

domonation, 2015-07-31:
I am using an Arduino Mega 2560 with 4 receivers while simultaneously emitting an IR signal. What I have done however is used an Attiny 85 to handle the emitting portion. What I am doing is essentially using the 4 receivers covering different quadrants in order to help the car navigate towards an IR beacon. I am also incorporating an ultrasonic sensor. On the mega the pins were connected to pins 42, 44-46.

As you can see from the sketch that the receiver initialization is commented out. Printing out millis() and obtaining an accurate distance works perfectly, but as soon as I remove the comment, the delays and printed millis() value are not what is desired which in turn affects the ultrasonic sensor. 

I am currently using a Nano that a friend had lying around to handle the receiving portion and I simply send over to the mega which receiver detects the signal. The problem with the nano however is that it can only handle 3 receivers stably.

z3t0, 2015-08-04:
@domonation Sorry for the delay, will get back to you ASAP (have been out for a few days_

z3t0, 2015-08-10:
@domonation Sorry for not being much help... I really don't know what is going on and also do not know too much about the Multiple Receiver Fork as it isn't very robust or production ready... I am working on a multiple receiver implementation of my own and hopefully that will work for you, in the meantime you could converse with @khair-eddine as he may know more about this particular topic.

z3t0, 2015-08-10:
@domonation Take a look at the patch in #184 as it may prove useful

z3t0, 2015-11-27:
Closing as there has been no response for over three weeks. Please re-open if your issue has not been solved

closed 2015-11-27 by z3t0