Arduino-IRremote/Arduino-IRremote #181: Travis CI Build Failing

z3t0, 2015-07-30:
I am new to using Travis CI and have not yet figured out all the different options.
@ivankravets has been kind enough to take care of all the setup and I have finally managed to get Travis to work... only now it seems to be reporting very odd errors. 

If anyone knows how to use it please take a look at https://travis-ci.org/shirriff/Arduino-IRremote
I will definitely dig in and solve it, although it may take me a while as I am a bit hyped on the Windows 10 Update

ivankravets, 2015-07-30:
@z3t0 Sorry, this is PlatformIO bug. I'm working on it and will release new version today. I'll report here.

ivankravets, 2015-07-30:
@PlatformIO 2.2.2 is out! However, the one example still doesn't work. See https://travis-ci.org/z3t0/Arduino-IRremote/builds/73409409

```
IRsendDemo.ino:11:8: error: no matching function for call to 'IRrecv::IRrecv()'
```

Need to fix this line https://github.com/z3t0/Arduino-IRremote/blob/master/examples/IRsendDemo/IRsendDemo.ino#L11 to
```c
int RECV_PIN = 11;

IRrecv irrecv(RECV_PIN);
```

z3t0, 2015-07-31:
Thanks, working great now. Do you know if there's any way to reject commits that fail.

closed 2015-07-31 by z3t0

ivankravets, 2015-07-31:
I hope this should help you http://stackoverflow.com/questions/448919/how-can-i-remove-a-commit-on-github

However, I don't recommend to use it :blush: