Arduino-IRremote/Arduino-IRremote #184: Problems using multiple IR Transmitter (@khair-eddine fork)

alexbloggt, 2015-08-04:
Hello,
when trying to use multiple IR Transmitter (as provided by @khair-eddine's fork, see https://github.com/z3t0/Arduino-IRremote/issues/128), I'm getting some errors from Arduino IDE when trying to compile.
Here is the sketch I am using:
```c
#include <IRremote.h>

IRsend irsend0;
IRsend1 irsend1;
IRsend2 irsend2;

void setup()
{
  Serial.begin(9600);
}

void loop() {
 
    for (int i = 0; i < 100; i++) {
      irsend0.sendNEC(0xFE50AF, 32);
      Serial.println("sending on pin 13");
      delay(10000);
      irsend1.sendNEC(0xFE50AF, 32);
      Serial.println("sending on pin 9");
      delay(10000);
      irsend2.sendNEC(0xFE50AF, 32);
      Serial.println("sending on pin 10");
      delay(10000);
    
  }
}
```
And this is the error:
```
Arduino: 1.6.5 (Linux), Platine: "Arduino Leonardo"

IRsendDemo_nec_test_multiple_irs.cpp.o: In function `loop':
/usr/sbin/arduino-1.6.5/IRsendDemo_nec_test_multiple_irs.ino:18: undefined reference to `IRsend1::sendNEC(unsigned long, int)'
/usr/sbin/arduino-1.6.5/IRsendDemo_nec_test_multiple_irs.ino:21: undefined reference to `IRsend2::sendNEC(unsigned long, int)'
collect2: error: ld returned 1 exit status
Fehler beim Kompilieren.
```
I am using this library (https://github.com/khair-eddine/Arduino-IRremote/tree/khair-eddine-patch-1) with the "ifdef"s removed to use NEC codes.

I am using IDE 1.6.5 (but also tried 1.0.5) and an Arduino Leonardo.

Thank you all for helping!

z3t0, 2015-08-04:
@alexbloggt Can you try compiling the same sketch for the Arduino Uno? Is the problem still present?

alexbloggt, 2015-08-05:
Hi @z3t0 
thanks for helping!
Just gave it a try using Arduino Uno - same here.

pcoughlin, 2015-08-05:
One of the potential problems with using this Library seems to be the need to modify the #defines in the IRremoteInt.h.

Did @alexbloggt make any modifications?  Was the Library re-compiled?

z3t0, 2015-08-05:
@alexbloggt Not too sure why it isn't working... Try using the examples see if they work...

alexbloggt, 2015-08-06:
Hi,
Thank you all for support!
@pcoughlin : No, I did not change anything in IRremoteInt.h yet, just in IRremote.cpp + IRremote.h to be able to use NEC-Codes.
But I think I have found the problem:
In IRremote.cpp it says
```c
#ifdef SEND_AIWA_RC_T501
void IRsend::sendAiwaRCT501(int code) {
[...]
}
void IRsend1::sendNEC(unsigned long data, int nbits)
{
[...]
}
[...]
#endif
```
So the endif of ```SEND_AIWA_RC_T501``` is after all the additions with IRsend1 and IRsend2. If I put the endif right after the AIWA-Block the sketch compiles - at least for Arduino Uno.
Using Arduino Leonardo it gives me a lot of errors regarding Timers:
```
Arduino: 1.6.5 (Linux), Platine: "Arduino Leonardo"

/home/alexander/Arduino/libraries/IRremote/IRremote.cpp: In member function 'void IRsend1::mark1(int)':
/home/alexander/Arduino/libraries/IRremote/IRremote.cpp:1848:4: error: 'TIMER_ENABLE_PWM3' was not declared in this scope
    TIMER_ENABLE_PWM3;
    ^
/home/alexander/Arduino/libraries/IRremote/IRremote.cpp: In member function 'void IRsend2::mark2(int)':
/home/alexander/Arduino/libraries/IRremote/IRremote.cpp:1857:3: error: 'TIMER_ENABLE_PWM4' was not declared in this scope
   TIMER_ENABLE_PWM4;
   ^
/home/alexander/Arduino/libraries/IRremote/IRremote.cpp: In member function 'void IRsend1::space1(int)':
/home/alexander/Arduino/libraries/IRremote/IRremote.cpp:1867:3: error: 'TIMER_DISABLE_PWM3' was not declared in this scope
   TIMER_DISABLE_PWM3;
   ^
/home/alexander/Arduino/libraries/IRremote/IRremote.cpp: In member function 'void IRsend2::space2(int)':
/home/alexander/Arduino/libraries/IRremote/IRremote.cpp:1874:3: error: 'TIMER_DISABLE_PWM4' was not declared in this scope
   TIMER_DISABLE_PWM4;
   ^
/home/alexander/Arduino/libraries/IRremote/IRremote.cpp: In member function 'void IRsend1::enableIROut1(int)':
/home/alexander/Arduino/libraries/IRremote/IRremote.cpp:1881:9: error: 'TIMER_PWM_PIN4' was not declared in this scope
 pinMode(TIMER_PWM_PIN4, OUTPUT);
         ^
/home/alexander/Arduino/libraries/IRremote/IRremote.cpp: In member function 'void IRsend2::enableIROut2(int)':
/home/alexander/Arduino/libraries/IRremote/IRremote.cpp:1888:11: error: 'TIMER_PWM_PIN3' was not declared in this scope
   pinMode(TIMER_PWM_PIN3, OUTPUT);
           ^
Fehler beim Kompilieren.
```
As I am not familiar with Timers any help is appreciated :)

z3t0, 2015-08-06:
@alexbloggt Thanks for all the details as it makes helping a lot easier. Please fork khair-eddines fork and add your changes there so we can see what changes you made, that would be the easiest way to figure out what the problem is. Thanks!

z3t0, 2015-08-06:
@pcoughlin 
>One of the potential problems with using this Library seems to be the need to modify the #defines in the IRremoteInt.h.

As it stands this is the most robust way to support several different platforms

alexbloggt, 2015-08-07:
Alright @z3t0 .
Just found a working solution for using multiple IR transmitter on 32u4 based Arduinos like Leonardo. Gonna check if it works reliable the next days and make a fork then.

pcoughlin, 2015-08-07:
What about using the method that other Libraries use like this one:
https://www.arduino.cc/en/Hacking/LibraryTutorial

It puts the _pin (in the IR library case, it would be iparam and maybe others) in the Class so that it is unique to each instance.

Later in the tutorial it shows using multiple instances of the library like this:
Morse morse(13);
Morse morse2(12);

Instead of like this:
IRsend irsend0;
IRsend1 irsend1;
IRsend2 irsend2;

Subtle difference, but it makes the code in the Library smaller and does not repeat code. Also, there is no limit to the number of instances like in the current 3 limit. The pins do not need to be hard coded inside the Library code either.

What do you think?

alexbloggt, 2015-08-08:
See fix for IRremote.cpp and support for multiple transmitters on 32u4 devices here: https://github.com/alexbloggt/Arduino-IRremote/tree/alexbloggt-patch-1

pcoughlin, 2015-08-08:
My question, as stated above, is that you RE-declare the IRsend as IRsend1 and IRsend2.

Why not just put the variables in the IRsend class as PRIVATE and then you can declare them like this:
``` C++
IRsend irsend0;
IRsend irsend1;
IRsend irsend2;
```

Also, if you make the "pin" a variable that is assigned at the time of creation, then it could look like this:
``` C++
IRsend irsend0(3); // pin 3
IRsend irsend1(9); // pin 9
IRsend irsend2(10); // pin 10
```

z3t0, 2015-08-08:
> It puts the _pin (in the IR library case, it would be irparams and maybe others) in the Class so that it is unique to each instance.

That's definitely a much better way to handle things and is how I plan on doing it eventually.

>My question, as stated above, is that you RE-declare the IRsend as IRsend1 and IRsend2.

That's the reason I did not merge the PR regarding this. The code is horrible for maintenance and not very well written. I will be gearing towards an approach very similar to what you suggested

z3t0, 2015-08-10:
@alexbloggt I am assuming this issue has been resolved and will be closing it, please re-open and/or comment if you still need help with this particular issue.

closed 2015-08-10 by z3t0