Arduino-IRremote/Arduino-IRremote #185: mitsubishi ir

franhotmail, 2015-08-07:
Hi Rafi,
my name is Francisco and I´m working in create my domotic home using Arduino and old electronic devices such routers, telephones, ...
I don´t like the inalambric tecnology as wifi, bluetooth, RF,.., I prefer to use cable or IR, (health issues relating to electromagnetic RF ) then I would like to controle my air conditioner using IR, that it is the reason because now I´m writing this e-mail.
I have checked your library with a Canon camera, and it is working fine but mi Air conditioner is Mitsubishi. I've always heard that Mitsubishi remote controls are very difficult to copy, and now  I´m  checking your library and I can´t to do that its working fine.

Now I´m running the IRrecord example and i have clicked the poweroff andgive me the following message in the serial port:

POWEROFF
Received unknown code, saving as raw
 m3250 s1450 m500 s300 m500 s1150 m400 s300 m500 s300 m500 s1150 m400 s350 m450 s1200 m350 s350 m450 s350 m450 s1200 m350 s1200 m400 s1200 m350 s400 m400 s1200 m350 s400 m400 s1200 m400 s1200 m350 s1250 m350 s350 m400 s400 m400 s400 m400 s400 m400 s1200 m350 s1250 m350 s350 m400 s1250 m350 s1250 m300 s400 m400 s400 m400 s1250 m300 s400 m400 s400 m400 s1250 m300 s400 m400 s400 m400 s1250 m300 s1250 m350 s400 m400 s1250 m300 s1250 m350 s1250 m300 s400 m400 s400 m400 s1250 m300 s1250 m350 s1250 m300 s1250 m350 s1250 m350 s400

POWERON
Received unknown code, saving as raw
 m3250 s1500 m450 s300 m500 s1250 m350 s300 m450 s300 m500 s1250 m300 s350 m450 s1250 m350 s300 m450 s350 m450 s1250 m350 s1250 m300 s1250 m350 s400 m350 s1250 m350 s400 m400 s1250 m300 s1250 m350 s1250 m300 s450 m350 s400 m400 s400 m350 s450 m350 s1250 m350 s1250 m300 s450 m350 s1250 m300 s1300 m300 s450 m350 s400 m350 s1300 m300 s450 m350 s400 m350 s1300 m300 s450 m350 s450 m300 s1300 m300 s1300 m250 s450 m350 s1300 m300 s1250 m300 s1300 m300 s450 m300 s500 m300 s1300 m300 s1300 m250 s1300 m300 s1300 m250 s1300 m300 s450

Could you help me with that or it is very difficult to use your library with Mitsubishi and it is prefeable to insert an (Arduino with IR receptor, Relay) in my air conditioner to have the option that power off, power on at least?

Thanks  you very much and I am anxious for your reply because I looked much the internet and I have not clear how to operate.

Thank you very much again

z3t0, 2015-08-10:
@franhotmail Thanks for your detailed question. Sorry for the late reply, I do not have a Mitsubishi AC to test with but with some trial and error I am sure we will be able to get to the end of this!

First please take a look at this: http://www.analysir.com/blog/2015/01/06/reverse-engineering-mitsubishi-ac-infrared-protocol/

raffaeler, 2015-08-13:
I suspect the issue is related to RAWBUF being limited to <=255
Using the DumpV2 with a mitsubishi, the overflow message is clear.
Hope that limitation will soon be resolved.

pcoughlin, 2015-08-13:
If you compare the two outputs and look at them as being LONG (>=m1150) or SHORT (<=500) pulses then they compare the same.  Also, the "error twitch" seems to be relatively cumulative, as in one is "m350 m450" and the other is "m400 m400".

I agree with @raffaeler that the RAWBUF needs to be increased to "see" the end of the signal so we will see some difference. As you can see, both outputs (ON and OFF) show 100 pulses (RAWBUF is 100).

Try going into IRremote.h and changing the value of RAWBUF to a bigger number (like from 100 to 200). Then repeat your test.
https://github.com/khair-eddine/Arduino-IRremote/blob/master/IRremote.h
Line 194 near the end of the file.

Please post your results.

Hope this helps.

z3t0, 2015-08-13:
I plan on working on increasing the RAWBUF but the main problem is that I dont have anything that I can test with since all my remotes use relatively "short" protocols, any ideas?

z3t0, 2015-08-13:
>  agree with @raffaeler that the RAWBUF needs to be increased to "see" the end of the signal so we will see some difference. As you can see, both outputs (ON and OFF) show 100 pulses (RAWBUF is 100).

#187 Should be useful

raffaeler, 2015-08-14:
My idea is the following: drop me the exact code you need to record the sequences from one or more remote controls.
I can send you the sequence so that you can test and adjust the code.

Side note:
I am already using the dumpv2 to see the codes. And I have also modified the code to accept larger sequences but the dump to the serial port is still wrong since the recognized code printed on the serial port is still only 32 bits. I believe this code won't help you, this is why I am suggesting you to send me the exact code to record longer sequences.

z3t0, 2015-08-14:
@raffaeler Do not worry about it, as of #190 I am able to send larger codes from the Arduino, so I can use that to test and figure out how to fix the RAWBUF.

> And I have also modified the code to accept larger sequences but the dump to the serial port is still wrong since the recognized code printed on the serial port is still only 32 bits. I believe this [...]

Please share this if possible. Also, just to be clear I have not yet written the code necessary to decode larger signals, I was hypothesizing that IF i did manage to fix it then I would have no way to test it as I do not have any AC Remotes or similar which use large signals. However this is no longer a problem as this library now has support for sending large signals (so I can use another arduino for testing).

raffaeler, 2015-08-14:
You probably already know the changes in IRremoteInt.h:
#define RAWBUF  400
struct {
//...
uint16_t      rawlen;
//...
}
irparams_t;

But the the decode_results (Value field) does not hold the longer decode code. Therefore, for example, temperature up and down are decoded with the same value.
I had no time to dig further. But let me know if I can help you in any way.
BTW Tomorrow I should be able to test two Daikin remotes and let you know.

raffaeler, 2015-08-15:
The following is just one key from the first Daikin remote control (I removed the timings).
If you are interested pls let me know.

Encoding  : UNKNOWN
Code      : EED01DBB (32 bits)
unsigned int  rawData[13] = {16681, 16,2, 14,3, 14,3, 14,3, 14,3, 13,0};  // UNKNOWN EED01DBB

Encoding  : SANYO
Code      : FFFFFFFF (0 bits)
unsigned int  rawData[133] = {494, 73,30, 14,21, 13,4, 13,4, 13,4, 13,22, 13,4, 13,4, 13,5, 12,5, 12,22, 13,4, 13,22, 12,22, 13,4, 13,22, 12,22, 13,21, 13,22, 13,21, 13,4, 13,4, 13,22, 12,5, 12,5, 12,5, 12,6, 11,6, 12,5, 12,5, 12,5, 12,6, 11,6, 11,23, 12,5, 12,23, 11,6, 12,5, 12,5, 12,23, 11,23, 12,5, 12,5, 12,5, 12,6, 11,23, 12,5, 12,5, 12,6, 11,6, 11,6, 11,6, 11,6, 12,5, 12,6, 11,6, 11,6, 11,23, 12,23, 11,23, 12,5, 12,5, 12,23, 11,23, 12,23, 11,0};  // SANYO FFFFFFFF
unsigned int  data = 0xFFFFFFFF;

Encoding  : SANYO
Code      : FFFFFFFF (0 bits)
unsigned int  rawData[309] = {701, 73,30, 14,21, 13,4, 13,4, 13,4, 13,22, 13,4, 13,4, 13,5, 12,5, 12,22, 12,5, 13,22, 12,22, 12,5, 12,23, 12,22, 12,22, 13,22, 12,22, 13,4, 13,4, 13,22, 12,5, 12,5, 12,5, 12,5, 12,6, 11,6, 12,5, 12,5, 12,5, 12,6, 11,6, 11,6, 11,6, 12,5, 12,5, 12,6, 11,6, 11,6, 11,6, 12,5, 12,6, 11,23, 11,23, 12,23, 11,6, 11,6, 11,6, 12,5, 12,23, 11,6, 11,23, 12,23, 11,6, 11,6, 12,5, 12,5, 12,6, 11,6, 11,6, 11,6, 11,6, 12,5, 12,23, 11,23, 12,23, 11,23, 12,22, 12,23, 12,22, 12,5, 12,5, 12,6, 11,6, 11,6, 11,6, 12,5, 12,5, 12,6, 11,6, 11,6, 11,6, 12,5, 12,6, 11,6, 11,6, 11,6, 11,6, 12,23, 11,23, 11,6, 12,5, 12,6, 11,6, 11,6, 11,6, 11,6, 12,5, 12,6, 11,6, 11,23, 11,24, 11,6, 11,6, 11,6, 11,6, 12,5, 12,6, 11,6, 11,6, 11,6, 11,6, 12,6, 11,6, 11,6, 11,6, 11,6, 11,6, 12,6, 11,23, 11,6, 11,6, 12,6, 11,6, 11,6, 11,23, 12,23, 11,6, 11,6, 11,6, 11,6, 12,6, 11,6, 11,6, 11,23, 12,5, 12,6, 11,6, 11,6, 11,6, 11,6, 12,6, 11,6, 11,6, 11,6, 11,23, 12,6, 11,6, 11,23, 11,6, 12,23, 11,0};  // SANYO FFFFFFFF
unsigned int  data = 0xFFFFFFFF;

z3t0, 2015-08-15:
Very interesting, this may take me a while to figure out but I will get to the bottom of it

franhotmail, 2015-08-19:
Hi again, sorry by the delay I just change the RAWBUF from 101 to 401 and here you can see the result:

POWEROFF
3E2ADECE
Unknown encoding: 3E2ADECE (32 bits)
Raw (184): 18646 3350 -1400 600 -200 600 -1200 350 -200 600 -200 600 -1200 350 -200 600 -1250 300 -250 550 -200 100 -50 450 -1250 300 -1250 350 -1250 300 -250 50 -50 450 -1250 350 -400 350 -1250 350 -1250 300 -1250 350 -400 400 -400 350 -450 350 -450 350 -1250 300 -1300 300 -400 400 -1250 300 -1250 350 -400 350 -450 350 -1300 300 -400 350 -450 350 -1300 300 -400 350 -450 350 -1300 300 -1250 300 -450 350 -1300 250 -1300 300 -1300 300 -400 350 -1300 300 -450 350 -1250 300 -1300 300 -1250 300 -1300 300 -450 350 -1250 300 -450 350 -1250 300 -450 350 -450 350 -450 350 -450 300 -1300 300 -1300 250 -1300 300 -450 350 -450 300 -1300 300 -450 350 -1300 250 -450 350 -450 350 -450 350 -1250 300 -1300 300 -450 300 -1300 300 -450 350 -450 300 -1300 300 -1300 250 -1300 300 -1300 300 -450 300 -1300 300 -450 350 -1300 250 -450 350 -450 350 -450 300 -500 300 -1300 300 -450 300 -1300 300 

POWERON

4523E597
Unknown encoding: 4523E597 (32 bits)
Raw (182): -20236 3350 -1350 600 -200 600 -1150 450 -200 550 -250 550 -1150 400 -250 550 -1200 400 -200 550 -250 550 -1200 400 -1150 400 -1200 400 -200 550 -1200 400 -250 550 -1200 350 -1200 400 -1200 350 -250 550 -250 50 -50 450 -350 400 -400 400 -1200 400 -1200 350 -350 450 -1200 350 -1250 350 -350 450 -350 450 -1200 350 -350 450 -350 400 -1250 350 -400 400 -350 400 -1250 350 -1200 400 -350 400 -1250 350 -1200 350 -1250 350 -400 400 -1200 350 -400 400 -1200 350 -1250 350 -1200 400 -1200 350 -400 400 -1200 350 -400 400 -1250 350 -350 400 -400 400 -400 400 -400 400 -1200 350 -1250 350 -1200 350 -400 400 -400 400 -1200 350 -400 400 -1250 350 -350 400 -400 400 -400 400 -1250 300 -1250 350 -400 400 -1200 350 -400 400 -400 400 -1200 350 -1250 350 -400 400 -1200 350 -400 400 -1250 300 -400 400 -1250 350 -400 350 -400 400 -1250 350 -400 400 -1200 350 -400 400 -1250 300

AnalysIR, 2015-08-19:
@franhotmail 
it look like your IR receiver is struggling with this AC signal. What is the model # of your IR receiver?
data sheet link?

raffaeler, 2017-03-06:
BTW, has the most recent version been updated with a fix to support large codes?
Thanks!

AnalysIR, 2017-03-06:
@raffaeler 
I believe that the library was never designed to handle such long codes - because it was designed to fit inside MCUs with relatively low levels of resources and also work as a library with other applications which also require some resources of the MCU to operate.

IRremote has been improved a lot in the last year or 2. It can send long codes using sendRAW, but the problems you raised above are firstly related to your IR receiver not the library.

minusdavid, 2017-07-28:
I have access to a Mitsubishi AC, so I'm keen to investigate this further. 

My smartphone has an IR transmitter, and I notice the Peel Remote on Android can control the Mitsubishi AC unit. (It has a wizard for configuring and the first couple sets of attempts didn't work but the third variation was successful.) 

Using my smartphone, an Arduino Uno, an IR receiver, and this library, I've tried to see what the app is sending, but I can't comprehend it at all. Despite reading through https://github.com/r45635/HVAC-IR-Control, I don't see any perfectly constant data coming through, although I occasionally see some patterns in the HEX that "irrecv.decode(&results)" puts into results.value. Usually it's "FFFFFFFF" or "64800004" or "64800000". But it seems to happen irregularly. 

I just tried a few consecutive tries on the power button and on the 4th I saw "64800000" amongst the strings in my serial monitor. Tried again and on the 3rd try there was "FFFFFFFF". Another attempt has "FFFFFFFF" in the 5th try. 

This is my first foray into IR, so I'm pretty clueless :).

minusdavid, 2017-07-28:
I notice this file https://github.com/z3t0/Arduino-IRremote/blob/master/ir_Mitsubishi.cpp, but I don't understand how it's used. (I mostly program in Python/Perl/PHP. Sometimes Java. My only experience with C++ has been with an occasional beginner level Arduino project.)

Within https://github.com/z3t0/Arduino-IRremote/blob/master/IRremote.h, it looks like there is some support for receiving Mitsubishi signals but not sending them?

In irRecv.cpp that is a line about attempting Mitsubishi decode but... I don't know if that ever runs? Maybe I should turn on the debug directives in IRremote.h...

minusdavid, 2017-07-28:
Attempting Mitsubishi decode
Testing mark (actual vs desired): 3450us vs 350us: 300 <= 3450 <= 600?; FAILED

minusdavid, 2017-07-28:
I'm guessing that it's falling through all the tests and gets to IRrecv::decodeHash.

Happy to troubleshoot further if someone could steer me in the right direction?

maxdd, 2018-08-03:
Maybe this could help

https://harizanov.com/2012/02/control-daikin-air-conditioner-over-the-internet/
https://github.com/mharizanov/Daikin-AC-remote-control-over-the-Internet

He reports

#define DAIKIN_HDR_MARK	    3650 //DAIKIN_ZERO_MARK*8
#define DAIKIN_HDR_SPACE	1623 //DAIKIN_ZERO_MARK*4
#define DAIKIN_ONE_SPACE	1280 
#define DAIKIN_ONE_MARK	    428
#define DAIKIN_ZERO_MARK	428
#define DAIKIN_ZERO_SPACE 428

ArminJo, 2020-10-12:
Try [IRMP ](https://github.com/ukw100/IRMP) for Mitsubishi protocol.

closed 2020-10-12 by ArminJo