Arduino-IRremote/Arduino-IRremote #190: IR RAW Code that works with IRLib but NOT with IRRemote

baget, 2015-08-10:
Hi

with the following IR RAW Data I'm able to control my A/C Using IRLib but not IRRemote

```
int khz=38; 

unsigned int Signal_0_0[] = {8341,3541,576,1600,576,491,555,1600,576,1600,555,491,555,491,555,1600,555,1600,555,491,555,491,555,1621,555,1600,555,491,555,1600,555,491,555,1600,576,1600,555,1600,555,491,555,1600,555,491,555,491,555,491,555,1600,555,491,555,1600,555,1600,555,491,555,512,555,491,555,1579,576,448,555,16383,8341,3541,555,491,555,1600,576,491,555,491,555,1600,555,1600,555,491,555,512,555,1600,576,1600,555,491,555,491,555,1600,576,491,555,1600,555,491,555,491,555,491,555,1600,555,491,555,1600,555,1600,555,1600,555,491,555,1579,512,555,555,491,555,1600,555,1600,555,1600,576,491,555,1557,576,16383,8341,3541,555,1600,576,491,555,1600,555,1600,576,491,555,491,555,1600,555,1600,555,512,555,491,555,1600,576,1600,555,512,555,1600,555,491,555,1600,576,1600,555,1600,555,491,555,1621,576,491,555,512,555,491,555,1600,555,491,533,1621,555,1600,555,491,555,491,576,491,576,1600,555,448,555}; 

irsend.sendRaw(Signal_0_0, sizeof(Signal_0_0)/sizeof(int), khz); 
```

z3t0, 2015-08-10:
@baget can you provide some more information regarding what setup you are using. Also, if that is not the full sketch please share the full sketch, also make sure you are using the latest code from the master branch as it is now stable and has newer features.

As far as I can see, the implementation of sendRaw is identical in both libraries so this may be a bug with this library or something else.

z3t0, 2015-08-10:
@baget I have just committed a change to the master branch which may have solved the problem, please check by downloading the latest code from the master branch.

z3t0, 2015-08-10:
My bad, the latest commit broke things. Will fix and recommit

z3t0, 2015-08-10:
I just recommitted waiting on Travis CI build, please test changes.

z3t0, 2015-08-10:
@baget The build was a success, please try the latest changes

baget, 2015-08-12:
@z3t0 
my setup is simple, Arduino Uno and IR Led.
I'm trying to control my A/C.

I recorded signals using AnalysIR and IR Toy and I have simple loop that send "ON" Command wait for 2sec then send "Off" command.

I took your latest code and it still don't not work... not sure why... I see the IR LED is transmitting, but my AC does "Accept" the command.
I built a library for my AC that works using IRLib but not IRRemote, I'm trying to make it universal.
https://github.com/baget/ChunlanAc

z3t0, 2015-08-12:
@baget Thanks for the information, I will look into it and see if I can figure out what the problem is

z3t0, 2015-08-12:
It may have something to do with the way mark and space are implemented. I think I will hold off on this as I plan on making this library more object oriented and so will polish things then. Once I am done changing how some things work it should be easy to get your library to work with both IR libraries.

AnalysIR, 2015-08-12:
The gap between the 3 bursts is approc 21,000 uSecs. However, IRremote can only handle a max of 16383 uSecs for a mark or space. If you enter 21,000 into IRremote you will get some weird value being used for the space. This is related to the use of delayMicroseconds in IRremote, which is a known limitation in the base Arduino code.

This is likely to be your issue. The workaround would be to treat it as 3 signals within IRremote & put a delay (21) between each signal.

AnalysIR, 2015-08-12:
Also, make use of the signal clean feature within AnalysIR & you should get better perfrmance/range.

z3t0, 2015-08-13:
> However, IRremote can only handle a max of 16383 uSecs for a mark or space. 

This seems to be a common problem that makes solving simple problems significantly more involved, I will take a look at figuring out how to remedy this within the library

@baget For now the workaround is your only option, but I will solve this problem very soon :D

z3t0, 2015-08-13:
@baget Can you post the irrecvDumpV2 output of both the original remote and irLib? I am trying to replicate the environment on my side so I can confirm i have solved the problem. 

Note: you will need two different boards to record the output of IRLib (One sending one receiving)

z3t0, 2015-08-13:
@baget I think i have it working on my side, I dont want to commit until we can confirm it. I cannot really check because I dont have that AC and the decode on IRremote is still unchanged, neither do i have Analysir. So waiting on your reply

z3t0, 2015-08-13:
@baget I have possibly fixed the delay problem as of https://github.com/z3t0/Arduino-IRremote/commit/11cb3fe442b2084600ab4bf278d5c727573aced3 please try the latest code.

baget, 2015-08-14:
Latest Code solve the issue!!! Thanks!!!

closed 2015-08-14 by baget

z3t0, 2015-08-14:
@baget Happy it worked!

z3t0, 2015-08-15:
@baget Could you do me a favor and retest this with the latest code, I just merged a PR that improves the delay and would appreciate if you could confirm that this still works for you

baget, 2015-08-27:
@z3t0 sorry for the delay, I took the latest code and it still work fine! :+1:

z3t0, 2015-08-27:
@baget No problem and thanks!