Arduino-IRremote/Arduino-IRremote #192: Compilation error in irISR.cpp

renxinhe, 2015-08-11:
I used this library 3 days ago, and there was no problem. Everything worked fine. But today I suddenly got this error after attempting to compile.

>Energia/libraries/IRremote/irISR.cpp:18:5: error: expected constructor, destructor, or type conversion before '(' token

z3t0, 2015-08-11:
@renxinhe Can you please make sure you are using the latest code from the master branch? Because that file was not changed in the last 10 days.

Can you provide more information? Please provide your full sketch (edit your first post) and also what board.

renxinhe, 2015-08-11:
I've added the full sketch above. I'm using a MSP430G2553 board on a Macbook running OS X Yosemite. I just downloaded the zip from master, so it should be the most updated.

The same error occurs on an online testing demo that used to work on Saturday (8/8) night:
```c
#include <IRremote.h>
int RECV_PIN = 11;
IRrecv irrecv(RECV_PIN);
decode_results results;

void setup()
{
  Serial.begin(9600);
  irrecv.enableIRIn(); // Start the receiver
}

void loop()
{
  if (irrecv.decode(&results))
    {
     Serial.println(results.value, HEX);
     irrecv.resume(); // Receive the next value
    }
}
```

`/Users/Jim/Documents/Energia/libraries/IRremote/irISR.cpp:18:5: error: expected constructor, destructor, or type conversion before '(' token`

z3t0, 2015-08-11:
@renxinhe Please try the latest code from master now, I just committed a minor change where there was a leftover windows ^M. Should work now

z3t0, 2015-08-11:
@renxinhe Do you have the library code that previously worked? If not can you remember when you got that code and from which source? Releases/experimental/master

renxinhe, 2015-08-11:
I tried the latest commit by downloading the zip from repo homepage, and replacing the library files in the Energia library folder, and unfortunately it doesn't fix the issue. I didn't change any code from the version which that worked. Both the pastebin code and the quoted code snippet in my last comment compiled and ran successfully a few days ago.

One of my friend just installed Energia and this library fresh on his mac, and it compiled for him. I'm trying to reinstall Energia to see if such reset can help fixing anything.

z3t0, 2015-08-11:
@renxinhe No problem, happy it worked for you!

closed 2015-08-11 by z3t0