Arduino-IRremote/Arduino-IRremote #194: Help supporting YORK AC remote

renno-bih, 2015-08-12:
Hello all,

I'm glad to see new version of this library. :)

I know that it is difficult to add fully functional AC protocol but I just need to save current state of my AC so I can turn it on, off and make some presets. But strange thing is that this remote produce two codes for one button press. IRrecvDump detects two 32bit NEC codes (74 length) for each button press.

Here are some examples:

Turn ON (Fan:auto, temp:24, mode: cooling):
```
9015260A
Decoded NEC: 9015260A (32 bits)
Raw (74): -7524 9000 -4450 650 -1650 650 -600 650 -550 650 -1600 700 -500 700 -550 650 -550 650 -550 650 -550 650 -550 650 -600 600 -1650 650 -600 650 -1600 700 -550 650 -1600 700 -550 650 -550 650 -1650 650 -550 650 -550 650 -1650 650 -1650 700 -550 650 -550 650 -550 650 -550 650 -550 650 -1650 650 -550 650 -1650 650 -600 600 -600 650 -1600 700 -550 650 
90152606
Decoded NEC: 90152606 (32 bits)
Raw (74): 25536 9000 -4450 650 -1650 650 -600 600 -600 600 -1650 700 -550 650 -550 650 -550 650 -550 650 -550 650 -550 650 -550 650 -1650 650 -550 650 -1650 700 -550 650 -1600 700 -550 650 -550 650 -1650 650 -550 650 -550 650 -1650 650 -1650 650 -550 700 -550 650 -550 650 -550 650 -550 650 -550 650 -1650 650 -1650 650 -550 650 -600 600 -1650 700 -550 650 
```

Turn OFF:
```8015240A
Decoded NEC: 8015240A (32 bits)
Raw (74): 2128 9000 -4450 700 -1600 700 -550 650 -550 650 -550 650 -550 650 -600 600 -600 650 -550 650 -550 650 -550 650 -550 650 -1650 650 -550 650 -1650 650 -550 650 -1650 650 -600 650 -550 650 -1600 700 -550 650 -550 650 -1650 650 -550 650 -550 650 -550 650 -550 650 -600 650 -550 650 -1600 700 -550 650 -1600 700 -550 650 -550 650 -1650 650 -550 650 
80152406
Decoded NEC: 80152406 (32 bits)
Raw (74): 25586 9000 -4450 700 -1600 700 -550 650 -550 650 -550 650 -550 650 -600 600 -600 600 -600 650 -550 650 -550 650 -550 650 -1650 650 -550 650 -1650 650 -550 650 -1650 650 -600 600 -600 650 -1600 700 -550 650 -550 650 -1650 650 -550 650 -550 650 -550 650 -550 650 -550 650 -550 650 -550 700 -1600 700 -1600 700 -550 650 -550 650 -1650 650 -550 650 
```

Change temperature from 24C to 25C:

```9095260A
Decoded NEC: 9095260A (32 bits)
Raw (74): 17628 9000 -4450 650 -1650 700 -550 600 -600 650 -1600 700 -550 650 -550 650 -550 650 -550 650 -1650 650 -550 650 -600 600 -1650 650 -600 650 -1600 700 -550 650 -1600 700 -550 650 -550 650 -1650 650 -550 650 -550 650 -1650 650 -1650 700 -550 650 -550 650 -550 650 -550 650 -550 650 -1650 650 -550 650 -1650 650 -600 600 -600 650 -1600 700 -550 650 
90952606
Decoded NEC: 90952606 (32 bits)
Raw (74): 25536 8950 -4500 650 -1650 650 -600 600 -600 650 -1600 700 -550 650 -550 650 -550 650 -550 650 -1650 650 -550 650 -550 650 -1650 650 -600 600 -1650 650 -600 650 -1600 700 -550 650 -550 650 -1650 650 -550 650 -550 650 -1650 650 -1650 650 -600 600 -600 650 -550 650 -550 650 -550 650 -550 650 -1650 650 -1650 650 -550 650 -550 650 -1650 650 -600 650 ```

Maybe second code is used just for checksum. Any idea how to support this protocol?

AnalysIR, 2015-08-12:
Use the sketch in our blog post and you should get a single signal capture, in the same format..
http://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/

AC signals can be much longer and more complex than standard TV IR signals.

z3t0, 2015-08-13:
I believe I have fixed the send delay, haven't tried anything for receiving as I have no way to test it. However I will still see if I can do something about it.

https://github.com/z3t0/Arduino-IRremote/commit/11cb3fe442b2084600ab4bf278d5c727573aced3

renno-bih, 2015-08-14:
@AnalysIR 
I tried your sketch and here is what I get for one button press:

```
Raw: (279) 9028, -4472, 676, -1632, 668, -568, 644, -560, 648, -1660, 644, -564, 644, -564, 648, -556, 644, -564, 648, -1632, 668, -1640, 672, -1636, 672, -560, 640, -568, 644, -1664, 648, -556, 644, -1640, 672, -1632, 676, -560, 644, -1636, 672, -560, 640, -568, 644, -1664, 644, -1636, 668, -540, 668, -564, 636, -568, 644, -564, 648, -556, 644, -1640, 672, -560, 640, -1640, 668, -564, 648, -560, 640, -1664, 648, -560, 640, -19976, 668, -564, 648, -556, 644, -564, 644, -560, 644, -564, 644, -564, 640, -564, 644, -564, 636, -568, 644, -560, 640, -568, 644, -536, 676, -556, 644, -1636, 676, -556, 644, -564, 648, -560, 640, -564, 644, -564, 648, -556, 644, -564, 648, -560, 640, -564, 648, -556, 644, -564, 648, -560, 640, -564, 644, -564, 648, -1636, 664, -564, 648, -532, 668, -568, 644, -39976, 9008, -4468, 668, -1640, 668, -568, 644, -560, 640, -1640, 672, -564, 648, -532, 668, -564, 648, -556, 644, -1636, 672, -1636, 676, -1636, 664, -568, 644, -560, 640, -1640, 672, -536, 672, -1636, 664, -1644, 668, -540, 672, -1636, 672, -560, 644, -564, 644, -1636, 664, -1644, 668, -564, 648, -560, 640, -564, 648, -560, 640, -564, 648, -560, 648, -1632, 668, -1640, 672, -560, 672, -536, 672, -1608, 704, -532, 668, -19944, 672, -532, 676, -560, 640, -564, 668, -540, 672, -532, 680, -528, 672, -532, 676, -532, 672, -536, 672, -532, 672, -536, 672, -1608, 704, -1604, 696, -536, 676, -1604, 704, -1604, 696, -536, 676, -1608, 692, -1612, 700, -532, 676, -1608, 692, -1616, 696, -508, 704, -528, 672, -1612, 700, -532, 668, -536, 676, -532, 676, -1604, 696, -1612, 700, -1608, 700, -1608, 696, 
```

@z3t0 
I'm not having problem with sending at the moment. I first need to get code. Does this code above shows that this AC remote sends 279 long code that this library cannot handle? What are solutions?

renno-bih, 2015-08-14:
Btw, looks like there is long delay in middle of transmitted code. So I increased _GAP parameter to 26000 and now it detects only one Raw(101) signal. By increasing RAWBUF to 240 it detects Raw(140) signal.

Long gap in middle of transmitted code tricked library. Not it detects only one code. I'll try to retransmit it tomorrow...

AnalysIR, 2015-08-14:
The hex value for your signal is Row: 1
Hex: 243969829000200044872D3032001B6C8F
Binary:
00100100 00111001 01101001 10000010
10010000 00000000 00100000 00000000
01000100 10000111 00101101 00110000
00110010 00000000 00011011 01101100
10001111 

It is 134 bits of data spread over 4 bursts (3 long gaps/spaces in between)

You should be able to send this with IRLib. However, you will need to apply the workaround for longer signals (available in our forum) to the IRLib library.

The long gaps are a know limitations of IRremote. (it wasnt really designed with AC signals in mind). I am not sure if @z3t0 meant ("I believe I have fixed the send delay") that he has now removed this limitation and if so, in what release it is available for IRremote.

renno-bih, 2015-08-14:
Thank you AnalysIR...

My goal is to send codes over WiFi and message that I'm sending needs to be as small as possible. Right now I send four parameters... Type of code, length, value of code (encoded) and bits.
It works pretty fast as I don't need to send array of raw data.

It would be great if someone can help me creating encoding and decoding function for this remote. Currently IRremote library detects it as NEC code and generated hex value contains information of only 68 ticks.

ir_Template.cpp explains it very well but this remote produce more complicated code. 
So, code needs to detect these 4 bursts, and encode value. Any ideas to implement it?

AnalysIR, 2015-08-14:
Try our blog posts here:
http://www.analysir.com/blog/tag/air-conditioner/

you are bound to find a few examples linked to github (with code) that do this for other protocols, which you can use as a starting point.

However, you will have to figure part of  it out for yourself - using above examples & IRremote as a basis.
 .

z3t0, 2015-08-14:
> The long gaps are a know limitations of IRremote. (it wasnt really designed with AC signals in mind). I am not sure if @z3t0 meant ("I believe I have fixed the send delay") that he has now removed this limitation and if so, in what release it is available for IRremote.

The long gaps now work when sending, it will not be in a release yet as I still need to do some fine tuning, https://github.com/z3t0/Arduino-IRremote/commit/11cb3fe442b2084600ab4bf278d5c727573aced3#commitcomment-12712230

renno-bih, 2015-08-17:
I tried two AC remotes so far, but both of them are recognized as NEC protocols, with length of 140 that is not true.
RAWBUF is set to 240.

What could be problem?

renno-bih, 2015-08-18:
Hello @z3t0 

I captured AC remote code using AnalysIR Arduino code. It is 279 length. So I copied these raw values into array that I'm sending using irsend.sendRaw function. 
Here is a code I'm using
```
#include <IRremote.h>

IRsend irsend;
unsigned int Signal_ON_0[] =  {9036, 4464, 672, 1636, 672, 560, 640, 540, 672, 1636, 676, 556, 644, 536, 676, 532, 668, 564, 644, 1636, 676, 556, 644, 1640, 672, 560, 640, 1640, 672, 1640, 668, 560, 640, 1644, 668, 536, 676, 532, 668, 1640, 668, 564, 648, 560, 640, 1640, 672, 1636, 672, 560, 644, 536, 672, 560, 644, 564, 644, 536, 664, 1644, 668, 536, 676, 1632, 668, 540, 676, 556, 640, 1640, 668, 536, 676, 19964, 668, 564, 648, 560, 640, 540, 672, 560, 640, 540, 668, 564, 648, 560, 640, 536, 676, 532, 668, 540, 672, 560, 640, 540, 672, 560, 648, 1632, 668, 540, 672, 560, 640, 540, 672, 560, 640, 568, 644, 564, 644, 532, 668, 568, 648, 556, 640, 568, 644, 560, 652, 532, 668, 564, 644, 536, 664, 568, 644, 1640, 672, 1636, 672, 1636, 668, 39980, 9008, 4472, 676, 1632, 668, 564, 648, 532, 676, 1632, 668, 568, 644, 560, 640, 568, 644, 536, 672, 1632, 672, 536, 672, 1636, 664, 540, 672, 1636, 676, 1632, 676, 560, 640, 1640, 672, 560, 640, 568, 644, 1636, 672, 560, 644, 540, 668, 1636, 676, 1632, 668, 540, 672, 536, 664, 540, 672, 536, 672, 560, 640, 568, 644, 1636, 664, 1644, 668, 536, 676, 560, 640, 1640, 668, 540, 672, 19968, 672, 560, 644, 536, 672, 560, 640, 540, 672, 532, 668, 540, 672, 560, 648, 532, 672, 536, 672, 532, 668, 540, 672, 1636, 664, 1644, 668, 536, 676, 560, 640, 1640, 668, 1640, 672, 532, 668, 1640, 672, 536, 672, 1636, 664, 1644, 668, 536, 676, 532, 668, 1640, 672, 536, 672, 532, 668, 540, 672, 532, 668, 540, 672, 532, 668, 1640, 672};
void setup()
{
  Serial.begin(115200);
}

void loop() {
  Serial.println("Transmitting");
  irsend.sendRaw(Signal_ON_0, sizeof(Signal_ON_0) / sizeof(int), 38); //AnalysIR Batch Export - RAW
  Serial.println(sizeof(Signal_ON_0) / sizeof(int));
  delay(5000);
}

```

Then I use AnalysIR sketch to capture code that Arduino sent. Here is what I get:
```
Raw: (23) 9068, 4452, 692, 1616, 696, 564, 644, 536, 676, 1636, 696, 536, 672, 508, 696, 540, 668, 564, 668, 1616, 696, 540, 668, 
```

I'm using last release of IRremote library. 

Why sendRaw function sends only 23 ticks instead of 279 ticks?

AnalysIR, 2015-08-18:
@renno-bih
279-23 = 256

So sounds like an 8-bit byte limitation. i.e RAWBUF is defined as 8-bit in IRremote. 

I quickly tested the output of your scetch on my oscilloscope & IRremote seems to be sending the full signal (279).

**My guess:** you still have IRremote firmware loaded on the arduino that is receiving the "23". Try to upload the firmware from our blog post again to that arduino....just to be extra sure!

AnalysIR, 2015-08-18:
Oh and just one more thing...

your signal has a space of 19964 in the middle. To date IRremote hasn't supported this, so you would need to split the signal in half and put a delay(20) between each part. 

(I believe there has been an update to remove this limitation in IRremote, which may not be in the official release yet)

renno-bih, 2015-08-18:
z3t0 says that this problem with long gaps is fixed.... 

No, I'm not using IRremote firmware to receive 279 long signal, I'm using your sketch. But IRrecivDump also detects it as 23 long signal.

So, here is what I'm doing...
I'm using your sketch to capture whole signal from AC remote, all 279 ticks. Then I copied these values into int array and then generated IR pulse using sendRaw function as you can see in code above.

I'm using another Arduino with your code to capture IR pulses that first Arduino sent.
It detects it as 23 ticks signal.... So, something must be wrong with current IRremote library.
I'm using the latest release.

AnalysIR, 2015-08-18:
I have an older version of IRremote configured and it seems to work fine (as per my scope), sending the '279' signal..

> z3t0 says that this problem with long gaps is fixed....

..provided you are using the **updated** code?

renno-bih, 2015-08-18:
Yep I'm using the latest build...with custom_delay_usec function.
I tried to delete some ticks from array. Now I'm sending 243 ticks and it works. So it is definitely 8-bit limitation.
@z3t0 
Do you have any news about overcoming this limitation?

AnalysIR, 2015-08-18:
>void IRsend::sendRaw (unsigned int buf[], unsigned char len, unsigned char hz)

it seems that the **len** parameter has changed to unsigned char, instead of **unsigned int**

I also noticed that the **hz** parameter is now changed to 8 bits. In general, this is Ok, but if ever anyone wants to implement a 455kHz carrier, this will become a problem.

@renno-bih : While you are waiting you could change the parameters, as suggested above, test & report back. It is in the irSend.cpp file.

AnalysIR, 2015-08-18:
you will also have to change this line to unsigned int as well

>for (unsigned char i = 0; i < len; i++)

...

z3t0, 2015-08-18:
@renno-bih Try the change mentioned by Analysir, if that works I can push it upstream.

renno-bih, 2015-08-20:
Yep. tried the latest build by z3t0. It works now, it sends all 279 ticks.
Just need to try it in practice. I'll let you know as soon as I test.

z3t0, 2015-08-20:
@renno-bih Cool, the latest build actually just included the changes mentioned by @AnalysIR . His pull request (containing those changes) fixed it.

z3t0, 2015-08-20:
@renno-bih Once you have experimented around, can you share some information about different commands and what effect they have? Thanks! That way we can add support for the YORK AC

renno-bih, 2015-08-20:
IT WORKS :)

My YORK AC accepted code. But here is a problem....
It only works if I use AnalysIR sketch to capture 279 ticks long code and then use sendRaw function to emit code.

But when using IRrecvDump sketch it detects only 74 ticks (RAWBUF is set to 256). As I can see there is long tick (19960) at this position. By changing define _GAP  to 25000 it now detects longer code (140 ticks) but still not whole code.
Is it difficult to overcome this limitation with receiving codes longer than 256?

If you want to analyse IR codes... here are some examples:

AC Off: 
```
Raw: (279) 9008, 4468, 676, 1632, 672, 536, 672, 536, 676, 528, 672, 536, 676, 528, 672, 536, 676, 532, 668, 536, 676, 532, 668, 1640, 672, 532, 676, 1632, 668, 1640, 672, 536, 676, 1628, 672, 1636, 676, 532, 668, 1640, 668, 536, 676, 532, 668, 1640, 672, 532, 668, 540, 672, 532, 680, 528, 672, 536, 676, 528, 672, 1636, 676, 532, 668, 1640, 672, 532, 676, 532, 668, 1640, 672, 532, 680, 19960, 724, 484, 676, 528, 672, 536, 676, 528, 672, 536, 676, 532, 668, 536, 676, 532, 668, 540, 672, 532, 676, 532, 672, 532, 676, 532, 668, 1640, 672, 532, 672, 536, 672, 532, 668, 540, 672, 536, 676, 528, 672, 536, 676, 528, 672, 536, 672, 532, 672, 536, 672, 532, 672, 536, 672, 536, 676, 528, 672, 1636, 676, 1632, 680, 528, 672, 39976, 9012, 4468, 668, 1640, 672, 532, 676, 532, 668, 536, 676, 532, 668, 536, 676, 532, 668, 536, 676, 532, 668, 536, 676, 1632, 676, 532, 672, 1636, 672, 1636, 676, 528, 672, 1636, 676, 1632, 680, 528, 672, 1636, 672, 536, 668, 536, 672, 1636, 676, 532, 668, 536, 676, 532, 668, 536, 676, 532, 668, 540, 672, 532, 668, 1640, 672, 1636, 672, 536, 664, 540, 672, 1636, 676, 532, 668, 19972, 672, 532, 672, 536, 672, 532, 668, 540, 672, 532, 680, 528, 672, 536, 676, 528, 672, 536, 676, 528, 672, 536, 672, 1636, 676, 532, 668, 536, 676, 532, 668, 536, 676, 532, 668, 1640, 672, 532, 676, 1632, 672, 1636, 672, 1636, 676, 532, 668, 536, 676, 1632, 668, 540, 672, 532, 676, 532, 672, 1636, 672, 1632, 680, 1628, 672, 536, 676
```

AC On, Temp:20, Fan:Auto, Mode: cooling:
```
Raw: (279) 9032, 4476, 672, 1636, 676, 556, 644, 564, 644, 1636, 664, 540, 672, 564, 648, 556, 644, 564, 648, 560, 640, 564, 648, 1632, 668, 564, 648, 1636, 672, 1636, 664, 568, 644, 1636, 676, 1632, 668, 564, 648, 1636, 664, 568, 640, 564, 648, 1632, 668, 1640, 672, 560, 640, 568, 644, 564, 648, 556, 644, 564, 644, 1636, 668, 564, 644, 1640, 672, 560, 640, 568, 644, 1636, 676, 556, 644, 19968, 664, 568, 644, 564, 636, 568, 644, 564, 648, 556, 644, 536, 676, 560, 640, 564, 648, 556, 644, 564, 648, 560, 640, 564, 644, 564, 640, 1640, 668, 540, 672, 560, 640, 568, 644, 560, 640, 568, 644, 564, 648, 556, 644, 564, 648, 556, 644, 564, 648, 560, 640, 564, 644, 564, 640, 564, 644, 564, 640, 1640, 668, 1640, 672, 1636, 676, 39972, 9012, 4468, 668, 1640, 672, 560, 640, 568, 644, 1636, 672, 560, 644, 564, 648, 560, 640, 564, 644, 560, 644, 564, 644, 1636, 676, 560, 640, 1640, 672, 1636, 672, 560, 644, 1640, 668, 1640, 672, 560, 640, 1640, 672, 560, 648, 560, 644, 1636, 672, 1636, 676, 556, 644, 564, 648, 560, 640, 564, 648, 560, 640, 564, 648, 1636, 672, 1632, 672, 564, 644, 560, 644, 1636, 672, 564, 648, 19968, 664, 568, 644, 560, 644, 564, 644, 560, 640, 568, 644, 564, 648, 556, 644, 564, 648, 556, 644, 564, 648, 556, 644, 1640, 672, 1632, 676, 1636, 664, 1644, 668, 1640, 668, 1640, 672, 560, 672, 532, 680, 1604, 696, 1608, 700, 1608, 704, 532, 668, 536, 676, 1608, 704, 528, 672, 536, 672, 532, 672, 536, 672, 1608, 692, 1616, 696, 1612, 700
```
AC On, Temp:21, Fan:Auto, Mode: cooling:
```
Raw: (279) 9040, 4464, 672, 1640, 672, 560, 640, 564, 648, 1632, 668, 568, 644, 560, 648, 560, 644, 560, 648, 1632, 668, 568, 644, 1636, 676, 556, 648, 1632, 676, 1632, 676, 560, 640, 1640, 672, 1636, 676, 560, 640, 1640, 672, 560, 640, 568, 644, 1636, 672, 1636, 664, 568, 644, 564, 648, 556, 644, 564, 648, 560, 640, 1640, 672, 560, 640, 1640, 672, 560, 640, 568, 640, 1640, 672, 560, 640, 19972, 672, 560, 644, 564, 644, 564, 640, 564, 644, 564, 636, 576, 636, 564, 636, 568, 644, 564, 648, 556, 644, 564, 648, 556, 644, 564, 648, 1632, 668, 564, 644, 564, 640, 564, 644, 564, 648, 556, 644, 564, 648, 560, 640, 564, 648, 560, 640, 568, 644, 560, 640, 568, 644, 560, 648, 560, 644, 1636, 672, 1636, 676, 1632, 668, 1640, 672, 39976, 9012, 4468, 668, 1640, 668, 564, 648, 560, 640, 1640, 672, 560, 640, 568, 644, 560, 648, 560, 644, 1636, 672, 560, 640, 1644, 668, 564, 648, 1632, 668, 1640, 672, 560, 648, 1632, 668, 1640, 672, 560, 640, 1644, 668, 564, 648, 560, 640, 1640, 672, 1636, 672, 560, 640, 568, 644, 560, 640, 568, 644, 560, 640, 568, 644, 1636, 676, 1632, 668, 540, 700, 532, 668, 1612, 700, 532, 668, 19944, 672, 536, 672, 560, 644, 564, 644, 560, 644, 564, 676, 532, 668, 536, 676, 532, 668, 536, 676, 532, 668, 540, 672, 1608, 672, 560, 668, 1616, 696, 1612, 700, 1608, 704, 1604, 696, 536, 672, 536, 664, 1616, 696, 1612, 700, 1604, 696, 540, 672, 532, 680, 1604, 696, 536, 672, 536, 664, 540, 672, 536, 664, 1616, 696, 1612, 700, 1608, 700
```
AC On, Temp:22, Fan:Auto, Mode: cooling:
```
Raw: (279) 9016, 4464, 672, 1636, 672, 532, 668, 540, 672, 1636, 676, 532, 668, 536, 676, 532, 668, 540, 672, 532, 668, 1640, 672, 1636, 672, 536, 668, 1636, 672, 1636, 676, 560, 640, 1640, 672, 1636, 676, 556, 644, 1640, 668, 564, 648, 532, 668, 1640, 672, 1636, 676, 556, 644, 536, 676, 532, 668, 536, 672, 536, 668, 1640, 668, 564, 648, 1632, 668, 568, 644, 536, 664, 1640, 672, 564, 648, 19964, 668, 564, 6 [... cut ...] , 560, 644, 1636, 672, 1636, 676, 560, 640, 564, 648, 560, 640, 568, 644, 560, 648, 560, 644, 1636, 672, 1636, 676, 560, 640, 564, 652, 1628, 668, 564, 648, 19968, 676, 556, 644, 564, 648, 556, 644, 564, 644, 560, 644, 564, 644, 564, 640, 564, 644, 564, 648, 556, 644, 564, 648, 1632, 668, 564, 648, 1636, 672, 560, 644, 1636, 672, 1636, 676, 556, 644, 564, 648, 1632, 668, 1640, 668, 1640, 672, 560, 640, 568, 644, 1636, 676, 560, 640, 564, 648, 560, 640, 564, 648, 560, 648, 1632, 668, 544, 668
```

AnalysIR, 2015-08-20:
@renno-bih   great, thanks for the update.

I checked the first signal you posted and it comes up on my system as Electrolux134AC (134 bit AC protocol)
Row: 1
Hex: 200B69029000200034016D203200105C8E
Binary:
00100000 00001011 01101001 00000010
10010000 00000000 00100000 00000000
00110100 00000001 01101101 00100000
00110010 00000000 00010000 01011100
10001110 

Often AC (control) units are re-badged so who the original manufacturer is, is always a guess.

Regarding your other comments: IRremote is a library, intended to be integrated into other Arduino sketches and as such is designed to be as lean as possible to allow enough resources for other sketches & libraries. 

The sketch on our blog is a standalone sketch with more accurate timing & because of the size of AC signals, it is not really meant to be part of another sketch, as is. This is mainly due to the limited circa 2K SRAM available on standard Arduinos.

In summary, IRremote is working, as designed & our standalone sketch works fine for longer AC Signals.

z3t0, 2015-08-20:
@renno-bih I am taking a new approach to the new library (IRSmartLib) where the "protocols" are really just sketches and such, this way we wont have a problem with memory since the sketches are meant to show how things work and be the main theme, this means that although they take more memory the user is aware of this and can modify it to suit their needs

daniloqb, 2015-09-21:
Thanks AnalysIR!! I used your sketch and works fine. But only works if I record the data and send using rawbuf. I am analysing the bytes to try how to create a personalized code. But I am stucked. My AC remote control send 136bits (17 bytes) like the code above and I am  mapping the bits to know where they change when I change the configuration. 
What I had noted, is  that the change occurs in the middle of the stream and in the last 2 bytes. I  think that these 16 bits in the end are checksum, I try and try but I have nothing. Could you help me to figure out what is this 16 bits and how to calculate them?
Below is the stream of bits that I capture.

Thanks again!!!

turn off
00001000 00010000 00000111 00000010 
01000000 00001000 00000001 00001000
00000010 01100000 00000000 00000000 
00000000 00000000 11101100 01100101 
11011001

turn on
00001000 00010000 00000111 00000010 
01000000 00001000 00000001 00011000 
00000010 01100000 00000000 00000000
00000000 00000000 11101100 01110100 
00011000

jefersonla, 2015-11-13:
With the old version of IRremote me and a friend have decoded some of the codes used in York AC, but now I don't remeber where I have put the code,  sorry for this :/.
Nowadays I don't use York AC anymore but, I'm working in a project to create an object that represents the remote control of MideaAC, based on this post of @veillard [image]. The Midea AC library is already done (or almost done) so if you send me some IR codes, and test my code in your AC, we could create a similar library to York AC @renno-bih. What do you think?

You can find the MideaAC Library in this repository [image] he uses IRremote to send the signals (thanks a lot @z3t0 and Shirriff).

veillard, 2015-11-13:
> With the old version of IRremote me and a friend have decoded some of the codes used in York AC, but now I don't remeber where I have put the code,  sorry for this :/.
> Nowadays I don't use York AC [...]

  Hi Jeferson,

Always pleased when others reuse my code, however ...
TBH I'm horribly busy with work, familly and other things and didn't had
time to look into this recently. It just work for me now and I'm happy with
it. However over the summer I helped Reyne Jasson Marcelino de Brito
<reyne.jasson@academico.ifrn.edu.br> who was trying to make that example
to work for his needs, I think both of you should probably discuss
maybe Reyne can have a look at your code too :-)

  thanks !

Daniel

> ---
> https://github.com/z3t0/Arduino-IRremote/issues/194#issuecomment-156431606

jefersonla, 2015-11-13:
Thanks @veillard, I'll send an e-mail for Reyne soon. :+1:

renno-bih, 2015-11-17:
@jefersonla 
Sure, I'll be glad to provide you raw codes and then test it.
Let me know what codes do you need to begin with?

knoid, 2015-12-30:
I don't know if this is still ongoing but I just wanted to let you know that I had the same problem with an AC and changing `_GAP` and `RAWBUF` fixed it. Is there a way we could implement this changes? Make this constants configurable? Maybe adding a `#ifndef` before both defines so we can define them before `#include`ing the library?

z3t0, 2015-12-30:
@knoid Do you mean define them in-sketch?

knoid, 2015-12-30:
Exactly, but provide default values if not.

vaibhavkhandelwal23, 2017-04-15:
how to send data from arduino to ac...do anyone has code

z3t0, 2017-04-15:
Check link by analysir

vaibhavkhandelwal23, 2017-04-16:
Here is my code to send data from arduino to ac:But its not working.can anyone sort out my problem?
/***
IR Transmitter.ino
***/
#include<IRremote.h>
IRsend irsend;
unsigned int data[349]={632, 17792, 3020, 8916, 528, 464, 532, 1460, 524, 492, 496, 496, 496, 496, 500, 492, 516, 480, 524, 468, 532, 460, 532, 1436, 548, 464, 528, 464, 532, 1436, 548, 1436, 552, 464, 528, 1452, 532, 1476, 508, 1468, 516, 1472, 532, 1452, 544, 468, 532, 460, 532, 460, 532, 464, 528, 464, 528, 464, 532, 464, 528, 464, 528, 464, 528, 464, 532, 464, 528, 488, 504, 492, 496, 496, 496, 496, 500, 496, 496, 496, 516, 476, 532, 460, 532, 464, 528, 464, 528, 464, 528, 464, 532, 464, 528, 464, 528, 464, 532, 464, 528, 464, 528, 464, 528, 492, 504, 488, 500, 492, 500, 496, 496, 496, 508, 1472, 532, 1436, 552, 2940, 3020, 8892, 552, 1436, 552, 464, 528, 464, 532, 460, 532, 464, 528, 464, 528, 464, 532, 488, 504, 488, 496, 1468, 520, 496, 500, 492, 520, 1444, 552, 464, 532, 1436, 548, 1436, 552, 1436, 548, 1436, 548, 1440, 548, 1444, 540, 492, 504, 488, 500, 492, 500, 496, 496, 496, 504, 488, 524, 472, 528, 464, 528, 464, 528, 464, 532, 464, 528, 464, 528, 464, 528, 464, 532, 464, 528, 464, 528, 464, 532, 488, 504, 488, 504, 492, 496, 496, 496, 496, 500, 492, 516, 480, 524, 468, 532, 460, 532, 460, 532, 464, 528, 464, 528, 464, 532, 464, 528, 464, 528, 464, 528, 464, 532, 464, 528, 480, 524, 2924, 3020, 8928, 516, 1484, 500, 496, 500, 496, 520, 472, 528, 464, 528, 464, 532, 464, 528, 464, 528, 464, 532, 1432, 552, 464, 528, 464, 532, 464, 528, 1436, 548, 1460, 524, 1480, 504, 496, 496, 1468, 544, 1448, 544, 1436, 552, 1436, 548, 1440, 548, 1436, 548, 1436, 552, 1456, 528, 492, 500, 492, 496, 496, 496, 1468, 520, 1472, 540, 1440, 552, 464, 528, 464, 532, 464, 528, 464, 528, 464, 528, 1440, 548, 1432, 552, 464, 532, 464, 528, 1460, 520, 496, 496, 1468, 520, 1464, 544, 472, 528, 464, 532, 464, 528, 464, 528, 464, 532, 464, 528, 464, 528, 464, 528, 464, 528, 468, 528, 1440, 544, 1468, 540};        // Samsung air conditioner ON/OFF received HEX data

int KHz = 38;                    // default frequency
void setup() 
{
Serial.begin(115200);   // begin serial communication with baud rate 9600 
}
void loop()
{
irsend.sendRaw (data, 43, KHz);  // Samsung air conditioner ON/OFF
}

z3t0, 2017-04-16:
Create a new issue.

vaibhavkhandelwal23, 2017-04-16:
what does that mean?

z3t0, 2017-04-16:
https://github.com/z3t0/Arduino-IRremote/issues/new

vaibhavkhandelwal23, 2017-04-16:
#include <IRremote.h>
#include <IRremoteInt.h>

#define PIN_IR 3
#define PIN_DETECT 2
#define PIN_STATUS 13

IRsend irsend;
void setup()
{
  pinMode(PIN_DETECT, INPUT);
  pinMode(PIN_STATUS, OUTPUT);
  irsend.enableIROut(38);
  irsend.mark(0);
}

void loop() {
  digitalWrite(PIN_STATUS, !digitalRead(PIN_DETECT));
}

vaibhavkhandelwal23, 2017-04-16:
error is coming that the IRsend is not a data type

AnalysIR, 2017-04-16:
@vaibhavkhandelwal23

I can see some obvious issues with your code.....and will/might let you know what they are after you follow the instructions of @z3t0 (HINT: open a new issue)

btandiawan, 2017-05-09:
Hi, I am having a trouble.
I recorded my Panasonic AC remote using IRremote.h and this was the data recorded for on/off.

unsigned acres[] ={3550,1550,550,300,550,1200,500,350,500,350,550,300,550,300,550,300,550,300,550,300,550,350,500,350,500,350,550,300,500,1200,550,300,550,300,550,350,500,350,500,350,550,300,550,300,550,1150,550,1200,500,1200,550,300,550,300,550,1150,550,300,550,350,500,350,500,350,550,300,550,300,550,300,550,300,550,300,550,350,500,350,500,350,500,350,550,300,550,300,550,300,550,300,550,350,500,350,500,350,500,350,550,300};

I tried to send it back with these code:

int khz = 38;
irsend.sendRaw(acres, sizeof(acres)/sizeof(acres), khz);

back to my AC but my AC did not respond anything
when i tried changing the temperature on the original remote, my AC responded on/off; and back to its normal function then.

I suspect there is ACK bit for this AC but I'm just guessing.
Any help?

AnalysIR, 2017-05-09:
5 comments:
1. It looks like you are hitting the limit for RAWBUF of 100. You can increase that to say 230 by changing the definition.

2. `unsigned acres[]` =......should be `unsigned int acres[]`

3. `irsend.sendRaw(acres, sizeof(acres)/sizeof(acres), khz);` should be `irsend.sendRaw(acres, sizeof(acres)/sizeof(acres[0]), khz);`

4. To get better timings...try the following sketch from our blog. (also avoids '1' above)
[Air Conditioners: Recording long Infrared Remote control signals with Arduino](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)

5. I am not aware of any ACK bit or signal for AC units. The AC remote usually sends the full config/settings in each signal, which is why they are longer than TV signals.

closed 2020-07-02 by ArminJo