Arduino-IRremote/Arduino-IRremote #197: AIWA RC T501

KyKyIIIKuH, 2015-08-14:
Hello.
I would like to know how you get "POWER 0x7F80"? I'm trying to get code of power button "AIWA RC T501" and getting this:
"
Encoding  : NEC
Code      : 76044FC0 (32 bits)
Timing[88]: 
     -26716
     +9050, -4400     + 650, - 450     + 550, -1700     + 650, -1600
     + 600, -1600     + 550, - 600     + 650, -1550     + 550, -1700
     + 650, - 500     + 500, - 600     + 650, - 450     + 650, - 500
     + 500, - 600     + 650, - 450     + 650, -1600     + 650, - 450
     + 650, - 500     + 550, - 550     + 650, -1600     + 550, - 550
     + 650, - 500     + 650, -1600     + 600, -1600     + 550, -1700
     + 650, -1600     + 600, -1600     + 550, -1700     + 650, - 450
     + 550, - 600     + 650, - 450     + 650, - 450     + 550, - 600
     + 650, - 450     + 650, - 500     + 500, - 600     + 650, -1600
     + 500, -1700     + 650, -1600     + 650, -1600     + 550, -1650
     + 700, -1550     + 650, -1600     + 550, -1700     + 650, 
unsigned int  rawData[89] = {25438, 181,88, 13,9, 11,34, 13,32, 12,32, 11,12, 13,31, 11,34, 13,10, 10,12, 13,9, 13,10, 10,12, 13,9, 13,32, 13,9, 13,10, 11,11, 13,32, 11,11, 13,10, 13,32, 12,32, 11,34, 13,32, 12,32, 11,34, 13,9, 11,12, 13,9, 13,9, 11,12, 13,9, 13,10, 10,12, 13,32, 10,34, 13,32, 13,32, 11,33, 14,31, 13,32, 11,34, 13,0};  // NEC 76044FC0
unsigned int  data = 0x76044FC0;
"
Using IR receiver "sm3374" analog of "TSOP1736"
I would like to know what I'm doing wrong ?

z3t0, 2015-08-15:
I do not know the specifics of that protocol, but will do some research and get back to you tomorrow. 
Just to be clear, you already have the remote and want to figure out how you can replicate the power command?

KyKyIIIKuH, 2015-08-15:
Yeah, you're right. I just want to get same result as yours.

PAq01, 2016-08-25:
Hello. This can help  http://lirc.sourceforge.net/remotes/aiwa/RC-T501

bengtmartensson, 2016-08-26:
The Aiwa protocol is not the same as the NEC1 protocol you try to use. It is 42 bits long while NEC1 is 32 bits.

[IrScrutinizer](http://www.harctoolbox.org/IrScurinizer.html) can generate the IRremote code from the Lirc file given above. [Here you go!](https://gist.github.com/bengtmartensson/de8bbe3722845f78ca6637ff79294f75)

closed 2019-04-19 by KyKyIIIKuH